ros-navigation/navigation2 #4386: Remove old Gazebo stuff once new gazebo lands

tonynajjar, 2024-05-31:
In https://github.com/ros-navigation/navigation2/pull/4298 we commented out stuff from old gazebo to get CI up and running. This task is a reminder to clean those up new gazebo is up and running.

SteveMacenski, 2024-05-31:
Can you enumerate these please!

tonynajjar, 2024-05-31:
Let's put it that way: grep for `gazebo_ros_pkgs` and `turtlebot3_gazebo` and make sure they are removed or replaced. nothing else.

(easier than listing every instance)

SteveMacenski, 2024-05-31:
:+1: Just as long as when we move on and forget the details we know what to do. Sounds good hah

stevedanomodolor, 2024-06-03:
@SteveMacenski  is the main still pointing to rolling? I am having many issues trying to simulate building the docker for ros rolling. Does nav2 still support rolling, when we develop for main, should it be based on rolling?.
The current documentation seem to be only for humble and does not work for rolling. 
```
INFO] [launch]: All log files can be found below /root/.ros/log/2024-06-03-22-01-45-130642-stevedan-145
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught multiple exceptions when trying to load file of format [py]:
 - PackageNotFoundError: "package 'turtlebot3_gazebo' not found, searching: ['/opt/overlay_ws/install/nav2_system_tests', '/opt/overlay_ws/install/nav2_bringup', '/opt/overlay_ws/install/navigation2', '/opt/overlay_ws/install/nav2_smoother', '/opt/overlay_ws/install/nav2_graceful_controller', '/opt/overlay_ws/install/nav2_dwb_controller', '/opt/overlay_ws/install/nav2_controller', '/opt/overlay_ws/install/dwb_plugins', '/opt/overlay_ws/install/dwb_critics', '/opt/overlay_ws/install/dwb_core', '/opt/overlay_ws/install/nav_2d_utils', '/opt/overlay_ws/install/dwb_msgs', '/opt/overlay_ws/install/nav_2d_msgs', '/opt/overlay_ws/install/nav2_waypoint_follower', '/opt/overlay_ws/install/nav2_theta_star_planner', '/opt/overlay_ws/install/nav2_smac_planner', '/opt/overlay_ws/install/nav2_rotation_shim_controller', '/opt/overlay_ws/install/nav2_regulated_pure_pursuit_controller', '/opt/overlay_ws/install/nav2_planner', '/opt/overlay_ws/install/nav2_navfn_planner', '/opt/overlay_ws/install/nav2_mppi_controller', '/opt/overlay_ws/install/nav2_constrained_smoother', '/opt/overlay_ws/install/nav2_bt_navigator', '/opt/overlay_ws/install/nav2_behaviors', '/opt/overlay_ws/install/nav2_core', '/opt/overlay_ws/install/nav2_collision_monitor', '/opt/overlay_ws/install/costmap_queue', '/opt/overlay_ws/install/nav2_costmap_2d', '/opt/overlay_ws/install/nav2_voxel_grid', '/opt/overlay_ws/install/nav2_velocity_smoother', '/opt/overlay_ws/install/nav2_rviz_plugins', '/opt/overlay_ws/install/nav2_map_server', '/opt/overlay_ws/install/nav2_lifecycle_manager', '/opt/overlay_ws/install/nav2_behavior_tree', '/opt/overlay_ws/install/nav2_amcl', '/opt/overlay_ws/install/nav2_util', '/opt/overlay_ws/install/nav2_simple_commander', '/opt/overlay_ws/install/nav2_msgs', '/opt/overlay_ws/install/nav2_common', '/opt/ros/rolling']"
 - InvalidFrontendLaunchFileError: The launch file may have a syntax error, or its format is unknown

```

SteveMacenski, 2024-06-04:
The system tests in rolling are commented out since we do not have a working simulation in new-gazebo yet

stevedanomodolor, 2024-06-04:
Sure, It is a blocking m development also so jajajaja. I would need more context though

SteveMacenski, 2024-06-04:
This gets the ball rolling https://github.com/ros-navigation/navigation2/pull/3634 with a TB3 sim but isn't released in binaries for us to be able to use in Nav2 so I think my current thought process is to:
- Get the TB3 stuff from Azzey's fork over into the nav2 system tests (as discussed in the thread) and port 1+ tests to use it 
- Merge that PR so Nav2 Bringup has something that works with a source build of the TB3 forked gazebo work to get started with 
- Complete + release https://github.com/ros-navigation/nav2_minimal_turtlebot4_simulation into binaries so that there is a TB4 sim in Jazzy/Rolling that generally works 
- Use that for the new nav2_bringup robot + package dependency so we can release binaries 
- Update the simple commander API demos to use the new simulation and/or environments 
- Update the remaining system tests 

So the TB3 stuff in the fork + doing the system tests could be parallelized with me getting the TB4 stuff completed and ready for bringup, if you were open to helping!

stevedanomodolor, 2024-06-04:
For now understood, will probably ask for more info later, but this should be good for a starting point

SteveMacenski, 2024-06-06:
Beyond the simple commander and system tests that need to be ported (see https://github.com/ros-navigation/navigation2/issues/4399) this is done. Closing

closed 2024-06-06 by SteveMacenski