ros-navigation/navigation2 #4387: revisit nav2_behavior_tree workarounds

tonynajjar, 2024-05-31:
@SteveMacenski in https://github.com/ros-navigation/navigation2/pull/4298 you introduced some workarounds to account for issues in BT.CPP. This ticket is a reminder to address them once the fixes are done on BT.CPP's side

SteveMacenski, 2024-05-31:
Are you referring to 

```
add_compile_options(-Wno-shadow) # Delete after https://github.com/BehaviorTree/BehaviorTree.CPP/issues/811 is released
```

Or anything else?

tonynajjar, 2024-05-31:
You added some more things in `nav2_behavior_tree`, not sure if these are workarounds or here to stay. From my stuff yes just this

SteveMacenski, 2024-05-31:
Those are to stay, that's a feature of the new BT.CPP versions. Possibly a regression, but also doesn't really matter since its something that only comes up in the isolated testing environment.

SteveMacenski, 2024-06-27:
https://github.com/ros-navigation/navigation2/pull/4500 Addresses

closed 2024-06-27 by SteveMacenski