ros-navigation/navigation2 #4388: test_decorator_speed_controller complains about not finding template specialization (but succeeds)

tonynajjar, 2024-05-31:
Just something I saw while testing something else, documenting it

```
test 25
      Start 25: test_decorator_speed_controller

25: Test command: /usr/bin/python3 "-u" "/opt/ros/rolling/share/ament_cmake_test/cmake/run_test.py" "/opt/overlay_ws/build/nav2_behavior_tree/test_results/nav2_behavior_tree/test_decorator_speed_controller.gtest.xml" "--package-name" "nav2_behavior_tree" "--output-file" "/opt/overlay_ws/build/nav2_behavior_tree/ament_cmake_gtest/test_decorator_speed_controller.txt" "--command" "/opt/overlay_ws/build/nav2_behavior_tree/test/plugins/decorator/test_decorator_speed_controller" "--gtest_output=xml:/opt/overlay_ws/build/nav2_behavior_tree/test_results/nav2_behavior_tree/test_decorator_speed_controller.gtest.xml"
25: Working Directory: /opt/overlay_ws/build/nav2_behavior_tree/test/plugins/decorator
25: Test timeout computed to be: 60
25: -- run_test.py: invoking following command in '/opt/overlay_ws/build/nav2_behavior_tree/test/plugins/decorator':
25:  - /opt/overlay_ws/build/nav2_behavior_tree/test/plugins/decorator/test_decorator_speed_controller --gtest_output=xml:/opt/overlay_ws/build/nav2_behavior_tree/test_results/nav2_behavior_tree/test_decorator_speed_controller.gtest.xml
25: [==========] Running 1 test from 1 test suite.
25: [----------] Global test environment set-up.
25: [----------] 1 test from SpeedControllerTestFixture
25: [WARN] [1717182653.469035623] []: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
25: [INFO] [1717182653.569213885] [test_behavior_tree_fixture]: Transforms are available now!
25: [ RUN      ] SpeedControllerTestFixture.test_behavior
25: You (maybe indirectly) called BT::convertFromString() for type [std::vector<geometry_msgs::msg::PoseStamped_<std::allocator<void> >, std::allocator<geometry_msgs::msg::PoseStamped_<std::allocator<void> > > >], but I can't find the template specialization.
25: 
25: You (maybe indirectly) called BT::convertFromString() for type [std::vector<geometry_msgs::msg::PoseStamped_<std::allocator<void> >, std::allocator<geometry_msgs::msg::PoseStamped_<std::allocator<void> > > >], but I can't find the template specialization.
25: 
25: You (maybe indirectly) called BT::convertFromString() for type [std::vector<geometry_msgs::msg::PoseStamped_<std::allocator<void> >, std::allocator<geometry_msgs::msg::PoseStamped_<std::allocator<void> > > >], but I can't find the template specialization.
25: 
25: You (maybe indirectly) called BT::convertFromString() for type [std::vector<geometry_msgs::msg::PoseStamped_<std::allocator<void> >, std::allocator<geometry_msgs::msg::PoseStamped_<std::allocator<void> > > >], but I can't find the template specialization.
25: 
25: You (maybe indirectly) called BT::convertFromString() for type [std::vector<geometry_msgs::msg::PoseStamped_<std::allocator<void> >, std::allocator<geometry_msgs::msg::PoseStamped_<std::allocator<void> > > >], but I can't find the template specialization.
25: 
25: You (maybe indirectly) called BT::convertFromString() for type [std::vector<geometry_msgs::msg::PoseStamped_<std::allocator<void> >, std::allocator<geometry_msgs::msg::PoseStamped_<std::allocator<void> > > >], but I can't find the template specialization.
25: 
25: You (maybe indirectly) called BT::convertFromString() for type [std::vector<geometry_msgs::msg::PoseStamped_<std::allocator<void> >, std::allocator<geometry_msgs::msg::PoseStamped_<std::allocator<void> > > >], but I can't find the template specialization.
25: 
25: You (maybe indirectly) called BT::convertFromString() for type [std::vector<geometry_msgs::msg::PoseStamped_<std::allocator<void> >, std::allocator<geometry_msgs::msg::PoseStamped_<std::allocator<void> > > >], but I can't find the template specialization.
25: 
25: You (maybe indirectly) called BT::convertFromString() for type [std::vector<geometry_msgs::msg::PoseStamped_<std::allocator<void> >, std::allocator<geometry_msgs::msg::PoseStamped_<std::allocator<void> > > >], but I can't find the template specialization.
25: 
25: You (maybe indirectly) called BT::convertFromString() for type [std::vector<geometry_msgs::msg::PoseStamped_<std::allocator<void> >, std::allocator<geometry_msgs::msg::PoseStamped_<std::allocator<void> > > >], but I can't find the template specialization.
25: 
25: You (maybe indirectly) called BT::convertFromString() for type [std::vector<geometry_msgs::msg::PoseStamped_<std::allocator<void> >, std::allocator<geometry_msgs::msg::PoseStamped_<std::allocator<void> > > >], but I can't find the template specialization.
25: 
25: You (maybe indirectly) called BT::convertFromString() for type [std::vector<geometry_msgs::msg::PoseStamped_<std::allocator<void> >, std::allocator<geometry_msgs::msg::PoseStamped_<std::allocator<void> > > >], but I can't find the template specialization.
25: 
25: You (maybe indirectly) called BT::convertFromString() for type [std::vector<geometry_msgs::msg::PoseStamped_<std::allocator<void> >, std::allocator<geometry_msgs::msg::PoseStamped_<std::allocator<void> > > >], but I can't find the template specialization.
25: 
25: You (maybe indirectly) called BT::convertFromString() for type [std::vector<geometry_msgs::msg::PoseStamped_<std::allocator<void> >, std::allocator<geometry_msgs::msg::PoseStamped_<std::allocator<void> > > >], but I can't find the template specialization.
25: 
25: You (maybe indirectly) called BT::convertFromString() for type [std::vector<geometry_msgs::msg::PoseStamped_<std::allocator<void> >, std::allocator<geometry_msgs::msg::PoseStamped_<std::allocator<void> > > >], but I can't find the template specialization.
25: 
25: You (maybe indirectly) called BT::convertFromString() for type [std::vector<geometry_msgs::msg::PoseStamped_<std::allocator<void> >, std::allocator<geometry_msgs::msg::PoseStamped_<std::allocator<void> > > >], but I can't find the template specialization.
25: 
25: [       OK ] SpeedControllerTestFixture.test_behavior (13006 ms)
25: [----------] 1 test from SpeedControllerTestFixture (13006 ms total)
25: 
25: [----------] Global test environment tear-down
25: [==========] 1 test from 1 test suite ran. (13121 ms total)
25: [  PASSED  ] 1 test.
25: -- run_test.py: return code 0                  
25: -- run_test.py: inject classname prefix into gtest result file '/opt/overlay_ws/build/nav2_behavior_tree/test_results/nav2_behavior_tree/test_decorator_speed_controller.gtest.xml'
25: -- run_test.py: verify result file '/opt/overlay_ws/build/nav2_behavior_tree/test_results/nav2_behavior_tree/test_decorator_speed_controller.gtest.xml'
25/61 Test #25: test_decorator_speed_controller ........................   Passed   13.29 sec
```

SteveMacenski, 2024-05-31:
I haven't seen this!

tonynajjar, 2024-05-31:
I guess it's missing `inline std::vector<geometry_msgs::msg::PoseStamped> convertFromString(const StringView key)` in `bt_utils.hpp`.

tonynajjar, 2024-05-31:
> I guess it's missing `inline std::vector<geometry_msgs::msg::PoseStamped> convertFromString(const StringView key)` in `bt_utils.hpp`. Not sure how we want to handle parsing a vector from a string, add another delimiter like `-` to differentiate between the PoseStamped? Or no new delimiter and just assume that every 9 elements are for a PoseStamped?

While adding a `inline std::vector<geometry_msgs::msg::PoseStamped> convertFromString(const StringView key)` function does solve the issue, I think that's a fake solution. `getInput` should be able to handle custom classes afaik. It looks like only an issue with `BT::getInputOrBlackboard`. @facontidavide can we borrow your support here?

SteveMacenski, 2024-06-27:
@facontidavide do you have any thoughts on this?

facontidavide, 2024-06-27:
> getInput should be able to handle custom classes afaik

No, it is not supposed to work magically, I am sorry

SteveMacenski, 2024-07-03:
I'm not entirely following the conversation here, @tonynajjar what's the ask for Davide on this or what we need to do in Nav2 so we can close this out?

facontidavide, 2024-07-03:
> While adding a inline std::vector<geometry_msgs::msg::PoseStamped> convertFromString(const StringView key) function does solve the issue, I think that's a fake solution.

No, this IS THE SOLUTION

> getInput should be able to handle custom classes afaik.

No, it should not, unless you did implement the template specialization of `convertFromString`

SteveMacenski, 2024-08-07:
@tonynajjar can you submit your template specialization for the PoseStamped vector?

closed 2024-09-04 by tonynajjar