ros-navigation/navigation2 #4396: Issue on Starting up Nav 2

MinElite, 2024-06-05:
## Bug report

**Required Info:**

- Operating System:
  -Ubuntu 22.04
- ROS2 Version:
  - Humble

Dear all,

I encountered an issue while starting up Nav2. The log information is as follows:

[component_container_isolated-1] [INFO] [1717585672.084058133] [amcl]: Message Filter dropping message: frame 'base_link' at time 1717585670.983 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1717585672.088607526] [lifecycle_manager_navigation]: Server controller_server connected with bond.
[component_container_isolated-1] [INFO] [1717585672.088835300] [lifecycle_manager_navigation]: Activating smoother_server
[component_container_isolated-1] [INFO] [1717585672.090181297] [smoother_server]: Activating
[component_container_isolated-1] [INFO] [1717585672.090358712] [smoother_server]: Creating bond (smoother_server) to lifecycle manager.
[component_container_isolated-1] [INFO] [1717585672.198680525] [lifecycle_manager_navigation]: Server smoother_server connected with bond.
[component_container_isolated-1] [INFO] [1717585672.198752327] [lifecycle_manager_navigation]: Activating planner_server
[component_container_isolated-1] [INFO] [1717585672.199156641] [planner_server]: Activating
[component_container_isolated-1] [INFO] [1717585672.199210961] [global_costmap.global_costmap]: Activating
[component_container_isolated-1] [INFO] [1717585672.199223136] [global_costmap.global_costmap]: Checking transform
[component_container_isolated-1] [INFO] [1717585672.199244264] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Lookup would require extrapolation into the past.  Requested time 1717585672.168612 but the earliest data is at time 1717585672.385210, when looking up transform from frame [base_link] to frame [map]
[rviz2-2] [INFO] [1717585672.696496079] [rviz2]: Message Filter dropping message: frame 'base_link' at time 1717585671.584 for reason 'discarding message because the queue is full'
[component_container_isolated-1] [INFO] [1717585672.699445222] [global_costmap.global_costmap]: start
[rviz2-2] [INFO] [1717585673.209751267] [rviz2]: Trying to create a map of size 60 x 60 using 1 swatches
[component_container_isolated-1] [INFO] [1717585673.753100722] [planner_server]: Activating plugin GridBased of type NavfnPlanner
[rviz2-2] [INFO] [1717585673.755042543] [rviz2]: Trying to create a map of size 1344 x 960 using 1 swatches
[component_container_isolated-1] [INFO] [1717585673.755860582] [planner_server]: Creating bond (planner_server) to lifecycle manager.
[component_container_isolated-1] [INFO] [1717585673.863538394] [lifecycle_manager_navigation]: Server planner_server connected with bond.
[component_container_isolated-1] [INFO] [1717585673.863752806] [lifecycle_manager_navigation]: Activating behavior_server
[component_container_isolated-1] [INFO] [1717585673.864547756] [behavior_server]: Activating
[component_container_isolated-1] [INFO] [1717585673.864597724] [behavior_server]: Activating spin
[component_container_isolated-1] [INFO] [1717585673.864675177] [behavior_server]: Activating backup
[component_container_isolated-1] [INFO] [1717585673.864708880] [behavior_server]: Activating drive_on_heading
[component_container_isolated-1] [INFO] [1717585673.864744580] [behavior_server]: Activating wait
[component_container_isolated-1] [INFO] [1717585673.864796454] [behavior_server]: Creating bond (behavior_server) to lifecycle manager.
[component_container_isolated-1] [INFO] [1717585673.974593474] [lifecycle_manager_navigation]: Server behavior_server connected with bond.
[component_container_isolated-1] [INFO] [1717585673.974652062] [lifecycle_manager_navigation]: Activating bt_navigator
[component_container_isolated-1] [INFO] [1717585673.974985231] [bt_navigator]: Activating
[component_container_isolated-1] Magick: abort due to signal 11 (SIGSEGV) "Segmentation Fault"...
[ERROR] [component_container_isolated-1]: process has died [pid 317908, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_isolated --ros-args --log-level info --ros-args -r __node:=nav2_container --params-file /tmp/tmp7_n6ru0r --params-file /tmp/launch_params_pjgwe921 -r /tf:=tf -r /tf_static:=tf_static'].

After this, the program gets stuck and neither makes a 2D pose estimation nor performs a navigation task. As shown above, there are several error messages, particularly:

[rviz2-2] [INFO] [1717585672.696496079] [rviz2]: Message Filter dropping message: frame 'base_link' at time 1717585671.584 for reason 'discarding message because the queue is full'
[component_container_isolated-1] [INFO] [1717585672.199244264] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Lookup would require extrapolation into the past.  Requested time 1717585672.168612 but the earliest data is at time 1717585672.385210, when looking up transform from frame [base_link] to frame [map]
[ERROR] [component_container_isolated-1]: process has died [pid 317908, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_isolated --ros-args --log-level info --ros-args -r __node:=nav2_container --params-file /tmp/tmp7_n6ru0r --params-file /tmp/launch_params_pjgwe921 -r /tf:=tf -r /tf_static:=tf_static'].

I am unsure of the cause and effects of these issues. Could you please assist in troubleshooting this problem? Any guidance or suggestions would be greatly appreciated.

Thanks a lot.

SteveMacenski, 2024-06-05:
Please fully fill in the PR template. How did you install it? What parameters are you using? How can we reproduce this? Etc.

I can't reproduce using the standard `tb3_simulation.launch.py` on Humble or otherwise.

MinElite, 2024-06-06:
Hi, thanks for the quick reply. Seems like some files were corrupted... I rebuilt the complete workspace and now it works fine.

closed 2024-06-06 by MinElite