ros-navigation/navigation2 #4402: [AMCL] Robot pose's covariance on turns

jesusramondovale, 2024-06-07:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble binaies
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
1. Setup a differential robot with a 180º LiDAR.
2. Configure Nav2 and AMCL for localization.
3. Monitor the `amcl_pose` while the robot moves.

#### Expected behavior
AMCL maintains a stable covariation in `amcl_pose`, indicating consistent confidence in the robot's estimated position.

#### Actual behavior
The covariation in `amcl_pose` increases over time when robot turns on itself, suggesting a growing uncertainty in the robot's position.

Is it possible to configure the number of degrees used by the LiDAR in AMCL to improve the accuracy and reduce the covariation? If so, how can this be achieved? Additionally, are there any recommended parameters or configurations for using AMCL with a 180º LiDAR to mitigate this issue?

---

## Feature request

#### Feature description
Allow configuration of the LiDAR scan angle within the AMCL module to enhance localization accuracy and reduce covariation.

#### Implementation considerations
Provide a parameter in the AMCL configuration to set the LiDAR scan angle. This would help users fine-tune the localization process based on their specific hardware setup, particularly for those using 180º LiDAR systems. Consider documenting the recommended settings for various LiDAR angles to guide users in achieving optimal performance.

SteveMacenski, 2024-06-07:
I don't understand. If the robot rotates and triggers and update, that can definitely adjust the confidence in the positioning. Who's to know how well you were localized before nor if your rotation resulted in slippage that didn't achieve the odometric rotation intended. There's many good reasons for AMCL's confidence to adjust when updating during rotations. 

 > Is it possible to configure the number of degrees used by the LiDAR in AMCL

I don't understand what this means.

jesusramondovale, 2024-06-07:
The robot is almost perfectly located on the map with amcl_pose's covariance matrix (all values) <0.004 before turning on itself. 

I'm using a Realsense Camera IMU and EKF node to get the exact rotation so no odometry from wheels is "needed" on turns with zero linear_x component.

I turn the camera on itself a random degrees amount (the same as if I'd turn the robot with the camera on it)

The /tf published by EKF node is perfect, the base_link axis turns the exact way on rviz as I do with my hand (with the camera or the physical robot) and without any lag. 

Also, the points of /scan topic are positioned over the map walls almost perfectly. 

Covariance of the amcl_pose grows significally on this process (when I turn the camera ot the robot with the camera on it). From <0.004 to >0.02

QUESTION: 

Is there any AMCL parameter that has something to do with the /scan topic angle used by this module? If there is, which one?

SteveMacenski, 2024-06-24:
@jesusramondovale any updates?

jesusramondovale, 2024-06-24:
We're thinking about using two 270° LiDARs on each corner to cover the full 360 degrees.

SteveMacenski, 2024-06-24:
Is there anything more we should address in this ticket?

jesusramondovale, 2024-06-25:
No, you can close this ticket and I'll come back to you if I make any progress.

Thank you

closed 2024-06-25 by SteveMacenski