ros-navigation/navigation2 #4404: Nav2 Docking: Optional Collision Checking on undocking / docking

SteveMacenski, 2024-06-08:
Should be straight forward: check the future trajectory from the graceful controller for collisions, ignoring within some tolerance of the dock itself (since we're making or in contact there). We can use the costmap collision topic object from the costmap package analog to the Behavior Server's use.

ajtudela, 2024-06-11:
Assign me this as it is the same behavior mentioned here: https://github.com/open-navigation/opennav_docking/pull/43. I can do both at the same time.

SteveMacenski, 2024-06-11:
Lets add this in a separate PR containing just this work though, don't want to have franken-PRs that are too large to review / merge. Smaller bite sized pieces are always better. Then that can be used in your work. Might make sense for this to be an independent module used by the controller or something so that it can be unit-tested and deployed elsewhere as needed easily

SteveMacenski, 2024-11-21:
Merging imminent, thanks @ajtudela this is a great feature!

closed 2024-11-21 by SteveMacenski