ros-navigation/navigation2 #4405: Nav2 Docking: Option for Non-Charging Dock Plugin Types

SteveMacenski, 2024-06-08:
If you want to "dock" with a non-charging type, we can easily add in a new plugin type for a non-charging docking maneuver. For example: docking with a convey belt or another robot.

It should be straight forward: have a non-charging dock plugin type and adjust the logic that if we're a non-charging dock to bypass certain parts of the process. We just rely more heavily on user-defined plugin for the specific situation to understand when "done" is _done_ (though the tolerance is a pretty good and reliable metric as we found in docking even without the joint state motor efforts)

redvinaa, 2024-07-01:
Summing up #44:
How to implement a scenario where a robot has to dock both forward and backward (for example, forward to a table / conveyor belt, and backward to actually charge)?
An option is to run two docking_server instances.

SteveMacenski, 2024-07-01:
^ Or have the option for reversing to be a feature of the dock not of the system like we have today. Food for thought.

SteveMacenski, 2024-08-19:
In progress - should have a PR opened this week

SteveMacenski, 2024-08-19:
https://github.com/ros-navigation/navigation2/pull/4627 implements

closed 2024-08-19 by SteveMacenski