ros-navigation/navigation2 #4428: Navigation2 for Jazzy

mattiadutto, 2024-06-13:
Hi! Which version should I use for Jazzy Jalisco? 
Is it enough to install from a source?
Thanks

SteveMacenski, 2024-06-13:
We have no branch for Jazzy yet, we're still working on the migration which has been far more intensive than we initially anticipated. 

You should use `main` for now, which is the branch targeting for release in Jazzy once we finish things up - see https://github.com/ros-navigation/navigation2/issues/4399 to track the progress. Once we're ready and complete for a binary release, we'll branch off `main` for Jazzy and things are good to go! 

The migration to 24.04 and all the bugs that entailed took longer than anyone thought possible and now the GZ migration is underway, but as of yet still incomplete (but probably ~75% done now)

mattiadutto, 2024-06-14:
Thank you very much!

indramal, 2024-06-18:
Waiting for Jazzy.....

SteveMacenski, 2024-06-21:
@indramal you're welcome to help with https://github.com/ros-navigation/navigation2/issues/4455 to make that possible.

SteveMacenski, 2024-06-24:
https://github.com/ros-navigation/navigation2/pull/4480