ros-navigation/navigation2 #4434: Use static map in outdoor environment without fixed intial pose

EnzoGhisoni, 2024-06-14:
We are using nav2 for outdoor navigation with a similar setup that what is used in the tutorial navigation2 with GPS (https://docs.nav2.org/tutorials/docs/navigation2_with_gps.html). 
The main difference in our application is that our robot doesn't have a fixed initial position on the static map, so we can't set a datum to match the origin of the map with that of the robot.
Is there a standard method for placing the origin of the static map or matching the initial position provided by the global EKF with the origin of the static map?

We thought of defining the origin of the static map in UTM coordinates and then using the UTM frame generated by navsat_transform to locate the robot relative to this point.

SteveMacenski, 2024-06-14:
I think this question is better suited for Robotics Stack Exchange or doing some google-fo on best conventions which I'm sure there are reference documents / projects that highlight. I don't want to misspeak since I haven't personally tried to do what you're doing before myself and this feels like a broader question about robotics systems than just Nav2. 

A good question is why you're even using a static map if you're doing GPS localization. Do you have a map at all? It is not needed to have one as that tutorial explains in the 3 methods for setting up GPS localization: (1) space with map to set sizing, (2) space of unknown or insanely large scale that a map is not reasonable and thus using a rolling global costmap, and (3) space of known size without a map to set a static global costmap size within.

Perhaps @pepisg that helped write that tutorial can chime in, but I think broadly this might be a question for another platform if Pedro doesn't have an answer for you.

pepisg, 2024-06-14:
I think you are mistaking the initial robot pose with the datum. The datum does not have to be set to the initial robot pose, you may set any arbitrary lat,lon,yaw coordinates in the `navsat_tranform` params or via the `/datum` service.

> Is there a standard method for placing the origin of the static map or matching the initial position provided by the global EKF with the origin of the static map?

This is what the datum does. What you put on the datum are the GPS coordinates you want the origin of the `map` frame to have. That's independent of the robot's pose.

> We thought of defining the origin of the static map in UTM coordinates and then using the UTM frame generated by navsat_transform to locate the robot relative to this point.

There's no need to, datum expects lat,lon.

@SteveMacenski I think you can close, I agree this should have been asked on stack exchange

closed 2024-06-14 by SteveMacenski