ros-navigation/navigation2 #4436: NullPtr bug during the `computeVelocityCommands()` calculation of `nav2_mppi_controller`

GoesM, 2024-06-14:
## Bug report

**Required Info:**

- Operating System:
  -  ubuntu 22.04
- ROS2 Version:
  -  humble
- Version or commit hash:
  -   the latest
- DDS implementation:
  -  the defaulted

#### Steps to reproduce issue

The bug occurs during normal use of nav2_mppi_controller. By changing the controller plugin to MPPI while keeping the rest of the configuration the same as the official default, start navigation2. The YAML configuration for changing the controller plugin to MPPI is:

```yaml
controller_server:
  ros__parameters:
    FollowPath:
      AckermannConstraints:
        min_turning_r: 0.2
      ConstraintCritic:
        cost_power: 1
        cost_weight: 4.0
        enabled: true
      CostCritic:
        collision_cost: 1000000.0
        consider_footprint: true
        cost_power: 1
        cost_weight: 3.81
        critical_cost: 300.0
        enabled: true
        near_goal_distance: 1.0
        trajectory_point_step: 2
      GoalAngleCritic:
        cost_power: 1
        cost_weight: 3.0
        enabled: true
        threshold_to_consider: 0.5
      GoalCritic:
        cost_power: 1
        cost_weight: 5.0
        enabled: true
        threshold_to_consider: 1.4
      ObstaclesCritic:
        collision_cost: 10000.0
        collision_margin_distance: 0.1
        consider_footprint: false
        cost_power: 1
        cost_scaling_factor: 10.0
        critical_weight: 20.0
        enabled: true
        inflation_radius: 0.55
        near_goal_distance: 0.5
        repulsion_weight: 1.5
      PathAlignCritic:
        cost_power: 1
        cost_weight: 14.0
        enabled: true
        max_path_occupancy_ratio: 0.05
        offset_from_furthest: 20
        threshold_to_consider: 0.5
        trajectory_point_step: 4
        use_path_orientations: false
      PathAngleCritic:
        cost_power: 1
        cost_weight: 2.0
        enabled: true
[... 147 lines cut ...]

Therefore, when the action_server_ receives a task command, it will invoke the function path computeControl() -> computeAndPublishVelocity():

https://github.com/ros-navigation/navigation2/blob/635880d3d6ad1f23eb75d793409189c653797f58/nav2_controller/src/controller_server.cpp#L482

The computeAndPublishVelocity() function directly uses the mppi_controller plugin functions for calculations:

https://github.com/ros-navigation/navigation2/blob/635880d3d6ad1f23eb75d793409189c653797f58/nav2_controller/src/controller_server.cpp#L584-L605

This causes the action_server_ to access the internal memory of mppi_controller during the computeControl() execution. However, it should be noted that although nav2_mppi_controller is a plugin, it is also an independent ROS2-Node that actively responds to exit signals (similar to how nav2_costmap_ros operates). Therefore, like costmap_ros, mppi_controller might respond to signals from rclcpp::ok() and exit earlier than nav2_controller.

If this response sequence occurs, mppi_controller might have already exited and released its memory while nav2_controller's action_server is still executing calculations, leading to access of released memory and thus an error.
Additional Information for Additional Information

##### Additional information for the additional information:

Note: This is a very similar issue occurred with nav2_costmap_ros.

All lifecycle nodes might randomly respond to signals:
https://github.com/ros-navigation/navigation2/blob/6dfe1b002f01257760064871b53846bc48196a03/nav2_planner/src/planner_server.cpp#L239-L243

Our previous discussion and fix for similar bugs:
https://github.com/ros-navigation/navigation2/issues/3971#issue-2001528486

https://github.com/ros-navigation/navigation2/pull/3972#issuecomment-1820538411

GoesM, 2024-06-15:
### futhermore Additional information

> So that, the bug seems to be caused because `action_server_` is still executing but mppi has been freed.

But, I'm confused about that because I notice there's already a `action_server_->deactivate()` to make sure it shutdown before cleanup.

https://github.com/ros-navigation/navigation2/blob/635880d3d6ad1f23eb75d793409189c653797f58/nav2_controller/src/controller_server.cpp#L280

So I check all related functions , and I noticed those lines:

https://github.com/ros-navigation/navigation2/blob/635880d3d6ad1f23eb75d793409189c653797f58/nav2_controller/src/controller_server.cpp#L454-L482

It's found that `computeAndPublishVelocity()` is called followed by `while(rclcpp::ok())` 

But, as our experiments before, we could know that `rclcpp::ok()` might bring some exception errors for waiting threads exit, here's the link to our discussion before: 

https://github.com/SteveMacenski/slam_toolbox/issues/691

##### suggestion
Shall we also replace the `rclcpp::ok()` by a bool flag ?

padhupradheep, 2024-06-18:
same happens while using RPP, see: https://github.com/ros-navigation/navigation2/issues/4438 (duplicate)

SteveMacenski, 2024-06-19:
@GoesM do you see any of the following errors with deactivate? https://github.com/ros-navigation/navigation2/blob/main/nav2_util/include/nav2_util/simple_action_server.hpp#L305-L320

I assume you only see this one shutdown when the system is currently operating, not when the robot is staying still and you shutdown the system (otherwise that thread / action server wouldn't be active in the first place). Thus, I would think you'd see one of these 2 errors / warnings.

If we set the server active state to false in the deactive function, it should be accessible by the [controller server here](https://github.com/ros-navigation/navigation2/blob/1788615b9b45ceec344c09fe882542e96de5e7b9/nav2_controller/src/controller_server.cpp#L457) to be checking it, which runs every cycle. So, at most we should be 1 cycle away from checking that we're not active anymore and thus should exit the control loop. The action server should be waiting 500ms while trying to shut down (https://github.com/ros-navigation/navigation2/blob/main/nav2_util/include/nav2_util/simple_action_server.hpp#L315) which should be plenty for  the controller's loop to end since the control rates should pretty much always be 10hz+. 

So... this seems like we shouldn't be having an issue, regardless of `rclcpp::ok` and you should be seeing error / warning logs about some failures on `deactivate()` that we could use to debug further.

GoesM, 2024-06-24:
Firstly, sorry for my previous misunderstanding regarding this bug. 

To ensure the accuracy of my fix, I conducted several experiments to verify my hypothesis. 

After these experiments, I invalidated my previous analysis, but fortunately, I now have a very definite answer.

## Conclusion
This is still an issue caused by `costmap_ros_` as an independent lifecycle node responding to the `Ctrl+C` signal before `nav2_controller`, resulting in a null pointer bug.

Therefore, this issue and #4438 are actually additional tickets to issue #4437, which will be resolved through a single PR.

***
## Details of the experiments and analysis are as follows:

### Experiment 1: Adding a delay before `rclcpp::ok()` does not trigger the bug.
```cpp
std::this_thread::sleep_for(std::chrono::seconds(2));
rclcpp::WallRate loop_rate(controller_frequency_);
while (rclcpp::ok()) {
```

### Experiment 2: Adding a delay after `rclcpp::ok()` but before `computeVelocityCommands()` almost always triggers the bug.
```cpp
updateGlobalPath();
std::this_thread::sleep_for(std::chrono::seconds(2));
computeAndPublishVelocity();
```

### Experiment 3: Inserting a delay at the same position as in Experiment 2 but replacing `rclcpp::ok()` with a `bool flag` method still almost always triggers the bug.

This indicates that `rclcpp::ok()` is not the cause. My previous speculative analysis was incorrect.

### Detailed Cause Analysis:

Inserting a delay before `computeVelocityCommands()` makes the `deactivate` process of `action_server_` very slow, while `cost_map_`, as an independent ROS2-LifecycleNode, has already responded to the `Ctrl+C` signal;

At this time, `nav2_controller` is attempting to execute `computeVelocityCommands()` and will call different controller plugins. However, plugins like `mppi_controller`, `regulated_pure_pursuit_controller`, and the default `dwb_controller` all call functions of `costmap_ros_->()`, leading to null pointer access.

##### `regulated_pure_pursuit_controller` accesses `costmap_ros_->getCostmap()`:
https://github.com/ros-navigation/navigation2/blob/9fbae3e66ecc3b7315ca7cbb070468a564f5e111/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp#L164-L171

##### `mppi_controller` accesses `costmap_ros_->worldToMap()`:
https://github.com/ros-navigation/navigation2/blob/9fbae3e66ecc3b7315ca7cbb070468a564f5e111/nav2_mppi_controller/src/path_handler.cpp#L91-L95

##### `dwb_controller` accesses `costmap_ros_->getSizeInCellsX()`:
https://github.com/ros-navigation/navigation2/blob/9fbae3e66ecc3b7315ca7cbb070468a564f5e111/nav2_dwb_controller/dwb_core/src/dwb_local_planner.cpp#L484-L485

SteveMacenski, 2024-06-24:
Merged!

closed 2024-06-24 by SteveMacenski