ros-navigation/navigation2 #4455: Port nav2 systems test to new gazebo

stevedanomodolor, 2024-06-20:
## Bug report

**Required Info:**

- Operating System:
  - 
- ROS2 Version:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

* This is to keep track of the nav2 system tests that has been ported. Will be closed as soon as all the tests has been ported. 
 ## To be ported 
- [x] localization
- [x] system
- [x] system_failure
- [x] updown
- [ ] waypoint_follower
- [ ] gps_navigation
- [ ] behaviors/spin
- [ ] behaviors/wait
- [ ] behaviors/backup
- [ ] behaviors/drive_on_heading
- [ ] behaviors/assisted_teleop
- [ ] costmap_filters
- [ ] cleanup old files
#### Implementation considerations

stevedanomodolor, 2024-06-24:
@SteveMacenski, so I am not going crazy, does the wait, backup, drive on heading behavior tests  fail once in a while. It does not seem to be stable.

SteveMacenski, 2024-06-24:
> @SteveMacenski, so I am not going crazy, does the wait, backup, drive on heading behavior tests fail once in a while. It does not seem to be stable.

You're not crazy, they're poorly written and it drives me bananas. I want to rewrite them at some point (#4349) to not try to fake everything out in some crazy way. I'm not really sure why this is the way that it is, I suspect it may have been written before we had any GZ simulations setup in the systems tests to work from. It is my plan to have a new contributor rewrite them as a project, but certainly if anyone else was interested in just deleting them and restarting now with the new GZ and the Simple Commander API, I think it could be done much, much more simply.

stevedanomodolor, 2024-06-24:
> > @SteveMacenski, so I am not going crazy, does the wait, backup, drive on heading behavior tests fail once in a while. It does not seem to be stable.
> 
> You're not crazy, they're poorly written and it drives me bananas. I want to rewrite them at some point (#4349) to not try to fake everything out in some crazy way. I'm not really sure why this is the way that it is, I suspect it may have been written before we had any GZ simulations setup in the systems tests to work from. It is my plan to have a new contributor rewrite them as a project, but certainly if anyone else was interested in just deleting them and restarting now with the new GZ and the Simple Commander API, I think it could be done much, much more simply. 
> 
> But, they should generally speaking work - it shouldn't be like 50%, it would be like ~80% successful, except for the Spin test that I disabled because it was failing ~80% of the time. Wait basically never failed. Backup/Drive on Heading was like 9/10 successful. 
> 
> I see you have a couple in draft (#4471 #4472), is that why?

Yes thats why they are on draft. I couldn't figure out why they fail.

SteveMacenski, 2024-07-03:
I completely rewrite the backup/spin/drive on heading tests to remove the flakiness and work with new GZ from your draft PRs. See https://github.com/ros-navigation/navigation2/pull/4515

stevedanomodolor, 2024-07-15:
@SteveMacenski Working on the GPS tests, I have migrated it, I am just having an issue with robot localization, it crashes. I am trying to figure out why. It happens after we call the fromLL service with a out of range value

```
1: [controller_server-5] [INFO] [1721076228.573393372] [controller_server]: Cancellation was successful. Stopping the robot.
1: [controller_server-5] [WARN] [1721076228.573490911] [controller_server]: [follow_path] [ActionServer] Client requested to cancel the goal. Cancelling.
1: [bt_navigator-9] [INFO] [1721076228.573510760] [bt_navigator]: Goal canceled
1: [bt_navigator-9] [WARN] [1721076228.573635926] [bt_navigator]: [navigate_to_pose] [ActionServer] Client requested to cancel the goal. Cancelling.
1: [controller_server-5] [INFO] [1721076228.575679904] [controller_server]: Optimizer reset
1: [tester_node-18] [INFO] [1721076230.533746634] [nav2_gps_waypoint_tester]: Sending goal request...
1: [waypoint_follower-10] [INFO] [1721076230.534812656] [waypoint_follower]: Converting GPS waypoints to map Frame..
1: [navsat_transform_node-17] terminate called after throwing an instance of 'GeographicLib::GeographicErr'
1: [navsat_transform_node-17]   what():  Latitude 35, longitude -118 out of legal range for UTM zone 30
1: [tester_node-18] [INFO] [1721076230.597235718] [nav2_gps_waypoint_tester]: Goal accepted
1: [tester_node-18] [INFO] [1721076230.598689055] [nav2_gps_waypoint_tester]: Waiting for 'follow_gps_waypoints' action to complete
1: [ERROR] [navsat_transform_node-17]: process has died [pid 98044, exit code -6, cmd '/opt/ros/rolling/lib/robot_localization/navsat_transform_node --ros-args -r __node:=navsat_transform --params-file /opt/overlay_ws/src/navigation2/nav2_system_tests/src/gps_navigation/dual_ekf_navsat_params.yaml --params-file /tmp/launch_params_0y7ot46v -r imu/data:=imu/data -r gps/fix:=gps/fix -r gps/filtered:=gps/filtered -r odometry/gps:=odometry/gps -r odometry/filtered:=odometry/global'].
```

SteveMacenski, 2024-07-15:
```
1: [waypoint_follower-10] [INFO] [1721076230.534812656] [waypoint_follower]: Converting GPS waypoints to map Frame..
1: [navsat_transform_node-17] terminate called after throwing an instance of 'GeographicLib::GeographicErr'
1: [navsat_transform_node-17]   what():  Latitude 35, longitude -118 out of legal range for UTM zone 30
```

I imagine that means that the GZ generated GPS is outside of what RL is expecting for the UTM zone. Did the other test set something for this like a datum?

SteveMacenski, 2024-08-07:
@stevedanomodolor any update on the GPS test :-)

stevedanomodolor, 2024-08-13:
Lots of holidays, Just came back not long ago. I will continue on this.

stevedanomodolor, 2024-09-17:
> ```
> 1: [waypoint_follower-10] [INFO] [1721076230.534812656] [waypoint_follower]: Converting GPS waypoints to map Frame..
> 1: [navsat_transform_node-17] terminate called after throwing an instance of 'GeographicLib::GeographicErr'
> 1: [navsat_transform_node-17]   what():  Latitude 35, longitude -118 out of legal range for UTM zone 30
> ```
> 
> I imagine that means that the GZ generated GPS is outside of what RL is expecting for the UTM zone. Did the other test set something for this like a datum to RL or the SDF for where to say the test is taking place w.r.t. GPS?

No, the goal we send in the test is outside of the UTM zone. The test that fail is this one 
# set waypoint outside of map
    time.sleep(2)
    test.setWaypoints([[35.0, -118.0]])
    result = test.run(True, False)
    assert not result
    result = not result
    assert (
        test.action_result.missed_waypoints[0].error_code
        == ComputePathToPose.Result().GOAL_OUTSIDE_MAP
    )
```
waypoint_tester]: Sending goal request...
23: [waypoint_follower-10] [INFO] [1726598194.053635890] [waypoint_follower]: Converting GPS waypoints to map Frame..
23: [tester_node-18] [INFO] [1726598194.054799709] [nav2_gps_waypoint_tester]: Goal accepted
23: [waypoint_follower-10] [INFO] [1726598194.054883944] [waypoint_follower]: Converted all 3 GPS waypoint to map frame
23: [waypoint_follower-10] [INFO] [1726598194.054917546] [waypoint_follower]: Received follow waypoint request with 3 waypoints.
23: [tester_node-18] [INFO] [1726598194.055347863] [nav2_gps_waypoint_tester]: Waiting for 'follow_gps_waypoints' action to complete
23: [bt_navigator-9] [INFO] [1726598194.055691408] [bt_navigator]: Begin navigating from current location (0.00, -0.00) to (-0.05, -0.00)
23: [controller_server-5] [INFO] [1726598194.076576708] [controller_server]: Received a goal, begin computing control effort.
23: [controller_server-5] [WARN] [1726598194.076614514] [controller_server]: No goal checker was specified in parameter 'current_goal_checker'. Server will use only plugin loaded general_goal_checker . This warning will appear once.
23: [controller_server-5] [WARN] [1726598194.076625956] [controller_server]: No progress checker was specified in parameter 'current_progress_checker'. Server will use only plugin loaded progress_checker . This warning will appear once.
23: [controller_server-5] [INFO] [1726598194.091044267] [controller_server]: Reached the goal!
23: [controller_server-5] [INFO] [1726598194.092883752] [controller_server]: Optimizer reset
23: [bt_navigator-9] [INFO] [1726598194.116375488] [bt_navigator]: Goal succeeded
23: [parameter_bridge-2] [INFO] [1726598194.169744050] [ros_gz_bridge]: Passing message from ROS geometry_msgs/msg/Twist to Gazebo gz.msgs.Twist (showing msg only once per type)
23: [waypoint_follower-10] [INFO] [1726598194.555098693] [waypoint_follower]: Succeeded processing waypoint 0, processing waypoint task execution
23: [waypoint_follower-10] [INFO] [1726598194.555136075] [waypoint_follower]: Arrived at 0'th waypoint, sleeping for 200 milliseconds
23: [waypoint_follower-10] [INFO] [1726598194.755823117] [waypoint_follower]: Task execution at waypoint 0 succeeded
23: [waypoint_follower-10] [INFO] [1726598194.755887136] [waypoint_follower]: Handled task execution on waypoint 0, moving to next.
23: [bt_navigator-9] [INFO] [1726598194.756499342] [bt_navigator]: Begin navigating from current location (0.00, -0.00) to (-4.92, -1.24)
23: [controller_server-5] [INFO] [1726598194.777316325] [controller_server]: Received a goal, begin computing control effort.
23: [controller_server-5] [INFO] [1726598195.827478752] [controller_server]: Passing new path to controller.
23: [controller_server-5] [INFO] [1726598196.827447556] [controller_server]: Passing new path to controller.
23: [controller_server-5] [INFO] [1726598197.877443008] [controller_server]: Passing new path to controller.
23: [controller_server-5] [INFO] [1726598198.877443427] [controller_server]: Passing new path to controller.
23: [controller_server-5] [INFO] [1726598199.927461036] [controller_server]: Passing new path to controller.
23: [controller_server-5] [INFO] [1726598200.927454241] [controller_server]: Passing new path to controller.
23: [controller_server-5] [INFO] [1726598201.977476905] [controller_server]: Passing new path to controller.
23: [controller_server-5] [INFO] [1726598202.977445152] [controller_server]: Passing new path to controller.
23: [controller_server-5] [INFO] [1726598204.027445683] [controller_server]: Passing new path to controller.
23: [controller_server-5] [INFO] [1726598205.027454522] [controller_server]: Passing new path to controller.
23: [controller_server-5] [INFO] [1726598206.077445491] [controller_server]: Passing new path to controller.
23: [controller_server-5] [INFO] [1726598207.077452019] [controller_server]: Passing new path to controller.
23: [controller_server-5] [INFO] [1726598208.127445717] [controller_server]: Passing new path to controller.
23: [controller_server-5] [INFO] [1726598209.127449363] [controller_server]: Passing new path to controller.
23: [controller_server-5] [INFO] [1726598210.177444148] [controller_server]: Passing new path to controller.
23: [controller_server-5] [INFO] [1726598211.177449104] [controller_server]: Passing new path to controller.
23: [controller_server-5] [INFO] [1726598212.227447300] [controller_server]: Passing new path to controller.
[... cut ...]
23: [ERROR] [navsat_transform_node-17]: process has died [pid 54129, exit code -6, cmd '/opt/ros/rolling/lib/robot_localization/navsat_transform_node --ros-args -r __node:=navsat_transform --params-file /opt/overlay_ws/src/navigation2/nav2_system_tests/src/gps_navigation/dual_ekf_navsat_params.yaml --params-file /tmp/launch_params_gf2s71vx -r imu/data:=imu/data -r gps/fix:=gps/fix -r gps/filtered:=gps/filtered -r odometry/gps:=odometry/gps -r odometry/filtered:=odometry/global'].
1/1 Test #23: test_gps_waypoint_follower .......***Timeout 180.13 sec

```

stevedanomodolor, 2024-09-17:
I suspect this might actually be the fix for it
https://github.com/cra-ros-pkg/robot_localization/pull/886. 
Has a new version of robot localization been release yet?

SteveMacenski, 2024-09-17:
No, not recently. If you clone and use `ros2` branch, does that fix the issue? If so, I can prioritize a release.

stevedanomodolor, 2024-09-17:
I am trying it now. I will let you know if it solves it as soon i get a result.

stevedanomodolor, 2024-09-17:
The current solution resolves the original issue, but it introduces a new problem, which I'm unsure is the intended behavior. When the "out of range" error occurs, the system returns the default values from the fromll ROS2 service (used for coordinate conversion). In this case, those default values are zero, causing the robot to navigate to coordinates (0,0). The main issue is that the fromll service doesn't differentiate between a successful conversion and an "out of range" scenario.

The root cause of the "out of range" error stems from the fact that when the fromll service is called, it assumes that the UTM zone is the same as the one initially set when the robot started the test. Since this UTM zone is never updated throughout the test, it leads to issues later on. Now that I reflect on it, this is essentially what you had pointed out earlier.
> ```
> 1: [waypoint_follower-10] [INFO] [1721076230.534812656] [waypoint_follower]: Converting GPS waypoints to map Frame..
> 1: [navsat_transform_node-17] terminate called after throwing an instance of [...]

stevedanomodolor, 2024-09-17:
One possible solution would be to avoid using the default UTM zone that is set when the robot launches. Instead, we could check  the UTM zone each time a fromll request is made. This way, the system would ensure that the correct UTM zone is being used for the conversion.

stevedanomodolor, 2024-09-18:
@SteveMacenski prs are open

stevedanomodolor, 2024-09-24:
It is. Without that it would not have worked hence my previous comment. All i had to do was that, put all the points in the same utm zome.

closed 2024-10-04 by SteveMacenski