ros-navigation/navigation2 #4499: Unable to build from source after PR #4393

chrisl8, 2024-06-26:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 24.04 LTS
- ROS2 Version:
  - ROS 2 Jazzy from Source
- Version or commit hash:
  - 6026754da6d1fb2986b4c6be260ea8f500768a0e 
- DDS implementation:
  - 

#### Steps to reproduce issue

Starting with commit 6026754da6d1fb2986b4c6be260ea8f500768a0e (and all commits after that) I am no longer able to build from source due to receiving this error during build:

```
Starting >>> nav2_common
Starting >>> nav_2d_msgs
Starting >>> nav2_minimal_tb4_description
Starting >>> slam_toolbox
Starting >>> nav2_minimal_tb3_sim
Starting >>> rplidar_ros
Finished <<< nav2_minimal_tb4_description [0.18s]
Starting >>> nav2_minimal_tb4_sim
Finished <<< nav2_minimal_tb3_sim [0.19s]
Finished <<< nav2_minimal_tb4_sim [0.17s]
Finished <<< rplidar_ros [0.35s]
Finished <<< nav2_common [0.57s]
Starting >>> nav2_msgs
Starting >>> nav2_voxel_grid
Finished <<< nav2_voxel_grid [0.22s]
Finished <<< nav_2d_msgs [0.98s]
Starting >>> dwb_msgs
Finished <<< slam_toolbox [1.42s]
Starting >>> arlobot_ros
Finished <<< dwb_msgs [1.06s]
Finished <<< arlobot_ros [1.22s]
Finished <<< nav2_msgs [2.87s]
Starting >>> nav2_util
Starting >>> nav2_simple_commander
--- stderr: nav2_util
CMake Error at src/CMakeLists.txt:17 (target_link_libraries):
  Target "nav2_util_core" links to:

    bondcpp::bondcpp

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

CMake Generate step failed.  Build files cannot be regenerated correctly.
gmake: *** [Makefile:520: cmake_check_build_system] Error 1
---
Failed   <<< nav2_util [1.04s, exited with code 2]
Aborted  <<< nav2_simple_commander [1.10s]

Summary: 11 packages finished [6.59s]
  1 package failed: nav2_util
  1 package aborted: nav2_simple_commander
  1 package had stderr output: nav2_util
  35 packages not processed
```

bondcpp package is installed and simply reverting to commit 23a48154bef1be3b063c7566698a2fdf4f497f66 allows the code to build.

DeccanLin, 2024-06-27:
Refer to https://github.com/ros-navigation/navigation2/pull/4480#issuecomment-2190664979

closed 2024-06-27 by chrisl8

SteveMacenski, 2024-06-27:
My favorite kind of ticket, the one the community answers while I have dinner!