ros-navigation/navigation2 #4513: nav2_mppi_controller - ARM (Jetson NANO) Issue? [component_container_isolated-1]: process has died

alex77272, 2024-07-03:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - ROS2 Humble
- Version or commit hash:
  - ros-humble-navigation2 1.1.15-1jammy.20240530.141947
- DDS implementation:
  - rmw_cyclonedds_cpp

#### Steps to reproduce issue
 When I run my navigation on my laptop (higher performance), everything works great. I am very happy with the way the robot navigates, no problems. 

I wanted to deploy the navigation on my Jetson Nano (On the Robot) running a docker with ros2 humble with noVNC but I can't launch my navigation, everything is exactly the same as on my laptop -> ros2_ws, packages, ...

If I use the standard nav_2 params, the navigation starts. If I use the MPPI controller -> ERROR

```controller_server:
  ros__parameters:
    controller_frequency: 20.0
    FollowPath:
      plugin: "nav2_mppi_controller::MPPIController"
      time_steps: 56
      model_dt: 0.05
      batch_size: 2000
      vx_std: 0.2
      vy_std: 0.2
      wz_std: 0.4
      vx_max: 0.5
      vx_min: -0.35
      vy_max: 0.5
      #vy_min: -0.35
      wz_max: 1.9
      iteration_count: 1
      prune_distance: 1.7
      transform_tolerance: 0.1
      temperature: 0.3
      gamma: 0.015
      motion_model: "Omni"
      visualize: false
      reset_period: 1.0 # (only in Humble)
      regenerate_noises: false
      TrajectoryVisualizer:
        trajectory_step: 5
        time_step: 3
      AckermannConstraints:
        min_turning_r: 0.2
      critics: ["ConstraintCritic", "CostCritic", "GoalCritic", "GoalAngleCritic", "PathAlignCritic", "PathFollowCritic", "PathAngleCritic", "PreferForwardCritic"]
      ConstraintCritic:
        enabled: true
        cost_power: 1
        cost_weight: 4.0
      GoalCritic:
        enabled: true
        cost_power: 1
        cost_weight: 5.0
        threshold_to_consider: 1.4
      GoalAngleCritic:
        enabled: true
        cost_power: 1
        cost_weight: 3.0
        threshold_to_consider: 0.5
      PreferForwardCritic:
        enabled: false
        cost_power: 1
        cost_weight: 5.0
        threshold_to_consider: 0.5
      # ObstaclesCritic:
      #   enabled: true
      #   cost_power: 1
      #   repulsion_weight: 1.5
      #   critical_weight: 20.0
      #   consider_footprint: false
      #   collision_cost: 10000.0
      #   collision_margin_distance: 0.1
      #   near_goal_distance: 0.5
      #   inflation_radius: 0.55 # (only in Humble)
      #   cost_scaling_factor: 10.0 # (only in Humble)
      CostCritic:
        enabled: true
        cost_power: 1
        cost_weight: 3.81
        critical_cost: 300.0
        consider_footprint: true
        collision_cost: 1000000.0
        near_goal_distance: 1.0
        trajectory_point_step: 2
      PathAlignCritic:
        enabled: true
        cost_power: 1
        cost_weight: 14.0
        max_path_occupancy_ratio: 0.05
        trajectory_point_step: 4
        threshold_to_consider: 0.5
        offset_from_furthest: 20
        use_path_orientations: false
      PathFollowCritic:
        enabled: true
        cost_power: 1
        cost_weight: 5.0
        offset_from_furthest: 5
        threshold_to_consider: 1.4
      PathAngleCritic:
        enabled: true
        cost_power: 1
        cost_weight: 2.0
        offset_from_furthest: 4
        threshold_to_consider: 0.5
        max_angle_to_furthest: 1.0
        mode: 0
      # VelocityDeadbandCritic:
      #   enabled: true
      #   cost_power: 1
      #   cost_weight: 35.0
      #   deadband_velocities: [0.05, 0.05, 0.05]
      # TwirlingCritic:
      #   enabled: true
      #   twirling_cost_power: 1
      #   twirling_cost_weight: 10.0
      ```

#### Expected behavior
Start the navigation stack
#### Actual behavior
[component_container_isolated-1] [INFO] [1719995187.199859277] [controller_serve
r]: Controller Server has goal_checker goal checkers available.
[component_container_isolated-1] [INFO] [1719995187.203409310] [controller_serve
r]: Created controller: FollowPath of type nav2_mppi_controller::MPPI Controller
[ERROR] [component_container_isolated-1]: process has died [pid 7429, exit code
-4, cmd /opt/ros/humble/lib/rclcpp_components/component_container_isolated --ro
s-args log-level info --ros-args -r _node:=nav2_container --params-file /tmp/
launch_params_3v080w41 --params-file /tmp/launch_params_14fqz859 -r/tf:=tf -r /
tf_static:=tf_static'].
#### Additional information

SteveMacenski, 2024-07-03:
Three immediate questions:
- Please get a trace https://docs.nav2.org/tutorials/docs/get_backtrace.html from the binaries, you should be able to use the gdb process without compiling from source (though provides more limited information). I'd like to know generally speaking where this crashes for you
- Please compile MPPI on your Nano and try again, does that work for you? 
- When this happens, what's the CPU utilization / memory like?  I haven't used a Nano before, but I imagine its weak and might actually just be too weak for this. Model Predictive Control is heavy.

SteveMacenski, 2024-08-07:
Closing - no response in over a month for follow up on backtrace needed to debug further. I assume they found their issue by now

closed 2024-08-07 by SteveMacenski