ros-navigation/navigation2 #4519: nav2 still not available on jazzy (binaries)

g-arjones, 2024-07-04:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 24.04
- ROS2 Version:
  - Jazzy (binaries)
- Version or commit hash:
  - N/A
- DDS implementation:
  - N/A

#### Steps to reproduce issue
```
apt install -y ros-jazzy-nav2-common
```

#### Expected behavior

Installs nav2 common packages

#### Actual behavior

Doesn't find binary packages

#### Additional information

I've seen a [post on RIS discourse announcing the nav2 jazzy release](https://discourse.ros.org/t/nav2-jazzy-release/38321) and also the [PR on rosdistro](https://github.com/ros/rosdistro/pull/41784) was merged but apparently there are no binary releases yet. I assumed the buildfarm would be triggered automatically when things get merged into ros distro... Am I missing something?

g-arjones, 2024-07-04:
@SteveMacenski Any pointers, please?

mikeferguson, 2024-07-04:
You're waiting on the next sync - http://repo.ros2.org/status_page/ros_jazzy_default.html shows that nav2_* packages have built on testing, but not yet been synced to public. You can switch your deb sources to testing or wait for sync (which is at the discretion/timing of the jazzy ROS boss)

g-arjones, 2024-07-05:
I guess I misunderstood this: https://docs.ros.org/en/rolling/The-ROS2-Project/Contributing/Build-Farms.html#execution

Thank you for clarifying...

closed 2024-07-05 by SteveMacenski