ros-navigation/navigation2 #4531: Call Python API navigator.followPath() , but it will fail after several successful executions

LightRadiance, 2024-07-10:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble
- Version or commit hash:
  - 1.1.14-1jammy.20240407.041637

#### Steps to reproduce issue

This problem occurs both with gazebo and in actual deployments and reprocduction is based on Gazebo. 
Firtst, prepare some waypoints in the map of Gazebo and I'm sure all of these points is not beyond map boundary or in the barrier.
```
"initial_pose":create_pose_stamped(-2.0188, -0.4937, -0.0, 1.0),
"pose1":create_pose_stamped(-0.3688, -1.85, 0.0277, 0.9996),
"pose2":create_pose_stamped(1.3483, -1.6491, 0.4655, 0.8851),
"pose3":create_pose_stamped(1.7077, -0.1337, 0.6892, 0.7246),
"pose4":create_pose_stamped(0.3839, 0.5708, -1.0, 0.0021),
"pose5":create_pose_stamped(-1.103, 0.6509, -0.9997, 0.0225), 
```
Then I use Python API navigator.getPath() to get the path between two adjacent points and these paths will be stored with List. At the same time, I will check whether every path is valid. Therefore, I don't think there are problems with these paths.
```
try:
    path = navigator.getPath(start=cur_pose, goal=next_pose)
except:
    print("No valid path to {}!".format(i))
    exit(0)
if path is None or len(path.poses) == 0 :
    print("The resolved path {} is not valid!".format(i))
    exit(0)      
goal_paths.append(TargetPath(path, path_poses[i]))
```
Next, I traverse goal_paths, and when the distance between the robot's current pose and the end of the path is less than 0.5, I will actively call navigator.followPath() to follow the next path. In fact, I traverse goal_paths by pop(0). The goal_pose represents the last pose of the target path. The dist_thre represents the distance thresh of checking path.
```
if dist_to_goal < dist_thre: 
    goal_paths.pop(0)
    path_goal = goal_paths[0].path_goal
    print(len(goal_paths[0].path.poses))
    navigator.followPath(goal_paths[0].path, "FollowPath", "general_goal_checker")                
    goal_pose = path_goal.goal_pose 
    dist_thre = path_goal.dist_thre
    print("Check Path!")
    break
```
Execute the following two commands in sequence to start the program.
```
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False
python3 loop.py
```
Initial pose estimate is published first, then each path is traversed in turn, switching paths when the threshold is reached.

#### Expected behavior
The turtlebot goes through each path in turn.

#### Actual behavior
The turtlebot only goes through the first two paths. When the target path check to the third, the controller reports "Resulting plan has 0 poses in it." and "Received plan with zero length.", then reports "Controller patience exceeded".
```
[component_container_isolated-5] [INFO] [1720597966.370959668] [controller_server]: Passing new path to controller.
[component_container_isolated-5] [INFO] [1720597973.720869886] [controller_server]: Passing new path to controller.
[component_container_isolated-5] [WARN] [1720597973.721201711] [controller_server]: Resulting plan has 0 poses in it.
[component_container_isolated-5] [INFO] [1720597979.570923423] [controller_server]: Passing new path to controller.
[component_container_isolated-5] [WARN] [1720597979.571522055] [controller_server]: Resulting plan has 0 poses in it.
[component_container_isolated-5] [WARN] [1720597979.621067509] [controller_server]: Resulting plan has 0 poses in it.
[component_container_isolated-5] [WARN] [1720597979.670926663] [controller_server]: Received plan with zero length
[component_container_isolated-5] [WARN] [1720597979.720939636] [controller_server]: Received plan with zero length
[component_container_isolated-5] [WARN] [1720597979.771096691] [controller_server]: Received plan with zero length
[component_container_isolated-5] [WARN] [1720597979.820937232] [controller_server]: Received plan with zero length
[component_container_isolated-5] [ERROR] [1720597979.820994034] [controller_server]: Controller patience exceeded
[component_container_isolated-5] [WARN] [1720597979.821038459] [controller_server]: [follow_path] [ActionServer] Aborting handle.
```

#### Additional information
The log for loop.py is as follows.
```
[INFO] [1720597952.680234468] [basic_navigator]: Publishing Initial Pose
[INFO] [1720597954.685248320] [basic_navigator]: Setting initial pose
[INFO] [1720597954.685634238] [basic_navigator]: Publishing Initial Pose
[INFO] [1720597954.685970973] [basic_navigator]: Waiting for amcl_pose to be received
[INFO] [1720597954.686330525] [basic_navigator]: Setting initial pose
[INFO] [1720597954.686619254] [basic_navigator]: Publishing Initial Pose
[INFO] [1720597954.686909031] [basic_navigator]: Waiting for amcl_pose to be received
[INFO] [1720597954.687189395] [basic_navigator]: Setting initial pose
[INFO] [1720597954.687467795] [basic_navigator]: Publishing Initial Pose
[INFO] [1720597954.687757359] [basic_navigator]: Waiting for amcl_pose to be received
[INFO] [1720597958.693125055] [basic_navigator]: Nav2 is ready for use!
[INFO] [1720597958.694007474] [basic_navigator]: Getting path...
96
[INFO] [1720597958.702412622] [basic_navigator]: Getting path...
166
[INFO] [1720597958.707745418] [basic_navigator]: Getting path...
159
[INFO] [1720597958.713013392] [basic_navigator]: Getting path...
129
[INFO] [1720597958.717099465] [basic_navigator]: Getting path...
69
[INFO] [1720597958.719816063] [basic_navigator]: Executing path...
166
[INFO] [1720597966.367932945] [basic_navigator]: Executing path...
Check Path!
159
[INFO] [1720597973.716439123] [basic_navigator]: Executing path...
Check Path!
129
[INFO] [1720597979.553752125] [basic_navigator]: Executing path...
Check Path!
```
The number in the output is printed by me to check if the path is empty when published. But that doesn't seem to be any problem

I read the souce code related to controller server and find  this issue might be caused by the part of action_server_->is_preempt_requested() and action_server_->accept_pending_goal(). I'm not sure.

SteveMacenski, 2024-07-10:
Are the paths in the local costmap or transformable into the odom frame for the controller? Its possible that your paths your send have the start point so far from the robot that there are no points in the local costmap and thus have its transformation rejected for attemping to follow. That could be due to spatially the points being too far or that the frame requested the path within isn't transformable to odom. The error `Received plan with zero length` is thrown by the algorithms when trying to transform the paths into the local frame, but you didn't provide me information about the algorithms you're using or attempting to use.

Overall, this is most probably an application issue, since this is the same API that we use for navigating in the system tests and in folks' deployed applications -- unless you can find an error in https://github.com/ros-navigation/navigation2/blob/main/nav2_simple_commander/nav2_simple_commander/robot_navigator.py#L344-L367. I expect that you can show by a print statement that the path poses are sent properly by printing the length of `goal_msg.path.poses` before sending. If that's the case, then its an issue with the inputs from the application script you mention whether its frames, timestamping, or actual distance from the robot's current pose and transformable by the set path handling parameters in your controller algorithm.

LightRadiance, 2024-07-11:
@SteveMacenski 
In this example, I use the DWB controller. But in actual deployments , I try RPP controller, and this problem occurs with both controllers. 
Today I discovered that the key problem seems to be getPath. Because when I send getPath request with different start pose, it gives the same path.
Actually, I have solved this problem by setting the getPath Parameter "use_start" to true, which default is set to false. I'm sorry to bother you. Thank you so much!

closed 2024-07-11 by LightRadiance