ros-navigation/navigation2 #4533: Extensibility of MPPI Critic

quantumxt, 2024-07-11:
Hello, we are currently experimenting with different MPPI critics in our project, which is achieved by cloning & modifying the source file (`nav2_mppi_controller`) directly.

With that in mind, we are wondering if there's a method to incorporate externally developed MPPI critics into the MPPI controller instead? This would involve creating a separate package to host these custom critics.

SteveMacenski, 2024-07-11:
Yes, of course! I don't have an example of this off hand right now, but it shouldn't be any different from other types of plugins (DWB, Controller, Planner, etc). You should be able to make a package, create a plugin, register it, and then be able to load it in your MPPI controller using provided binaries. You should look into Pluginlib in more general ROS documentation for instructions how to do that but I'll outline a bit here:
- https://github.com/ros-navigation/navigation2/blob/main/nav2_mppi_controller/src/critics/constraint_critic.cpp#L118-L120 --> register the class as a plugin of a plugin base class type 
- https://github.com/ros-navigation/navigation2/blob/main/nav2_mppi_controller/include/nav2_mppi_controller/critics/constraint_critic.hpp#L29 --> plugin should be derived from the base class type
- https://github.com/ros-navigation/navigation2/blob/main/nav2_mppi_controller/package.xml#L36-L40 --> you should export knowledge that this package contains a plugin of a base class type
- https://github.com/ros-navigation/navigation2/blob/main/nav2_mppi_controller/critics.xml --> the package should describe which libraries compiled in CMake contain plugins, and which plugins each of those libraries contain in particular for lookup
- https://github.com/ros-navigation/navigation2/blob/main/nav2_mppi_controller/CMakeLists.txt#L125 --> more exporting (might replace the package.xml export? not 100% sure myself so I do both to be safe, I know something is redundant here)

quantumxt, 2024-07-12:
Thank you for the detailed explanation and resources! We will review the provided links and have a go at it.

SteveMacenski, 2024-07-15:
Great! You can find a bunch of examples for plugins around Nav2 and MoveIt as well if you need more

closed 2024-07-15 by SteveMacenski