ros-navigation/navigation2 #4534: AMCL localization transform stability issues

lukasnoeller, 2024-07-11:
## Bug report

**Required Info:**

- Operating System:
  -

SteveMacenski, 2024-07-11:
Please ask on robot stack exchange. We have examples of multirobot launch files in Nav2, but we don't help fix custom end-user launch/configuration issues unless there's a bug report. Please reference existing files, documentation, and resources!

closed 2024-07-11 by SteveMacenski

lukasnoeller, 2024-07-11:
Hi @SteveMacenski , I have asked there before, but my post sadly did not get any responses. I would like to my post as clear and detailed as possible so that my issue may get resolved as I think it could help other people trying to do multi robot navigation. I can make any adjustments that are needed increase the chances of you understanding my issue.

SteveMacenski, 2024-07-11:
I understand you are having an issue configuring the stack and that always sucks, but you have examples that you can work / learn from below, addition to the wealth of documentation on docs.nav2.org and other community resources: 

- https://github.com/ros-navigation/navigation2/blob/main/nav2_bringup/launch/cloned_multi_tb3_simulation_launch.py
- https://github.com/ros-navigation/navigation2/blob/main/nav2_bringup/launch/unique_multi_tb3_simulation_launch.py 

Unfortunately, we don't have the bandwidth to find issues with end-users' configurations and setups when they customize for their application.

lukasnoeller, 2024-07-11:
I understand, thanks for the links. 

But based on a surface level impression of my post, do you have any clues or gut feeling as to where my issue could be coming from? 

What drives me nuts is the inconsistency. I would rather it not work all the time than work some of the time. Could the rate at which the odom->base_footprint transform is calculated and published affect how the tfs get connected?