ros-navigation/navigation2 #4538: [nav2_smoother] nullptr bug during `smooth_plan()`

GoesM, 2024-07-16:
## Bug report

**Required Info:**

- Operating System:
  -  ubuntu 22.04
- ROS2 Version:
  -  humble
- Version or commit hash:
  -   the latest
- DDS implementation:
  -  the defaulted

#### Steps to reproduce issue

```sh
#!/bin/bash
export ASAN_OPTIONS=halt_on_error=0:new_delete_type_mismatch=0:detect_leaks=0:log_pah=asan
source install/setup.bash
export TURTLEBOT3_MODEL=waffle
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/humble/share/turtlebot3_gazebo/models
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=True use_rviz:=False use_composition:=False 
```

#### Expected behavior

no NullPtr bug occured. 

#### Actual behavior

the Asan report of this NullPtr bug is as following:
```
 =================================================================
==166818==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x7187e491e19a bp 0x7187d75fc2d0 sp 0x7187d75f86a0 T13)
==166818==The signal is caused by a READ memory access.
==166818==Hint: address points to the zero page.
    #0 0x7187e491e19a in nav2_smoother::SmootherServer::smoothPlan() (/home/***/nav2_ws/install/nav2_smoother/lib/libsmoother_server_core.so+0x11e19a) (BuildId: 80891eed01337a00457782ceae8a233ce28b113c)
    #1 0x7187e49f0a93 in nav2_util::SimpleActionServer<nav2_msgs::action::SmoothPath>::work() (/home/***/nav2_ws/install/nav2_smoother/lib/libsmoother_server_core.so+0x1f0a93) (BuildId: 80891eed01337a00457782ceae8a233ce28b113c)
    #2 0x7187e49efe04 in std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::SmoothPath>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::SmoothPath> >)::'lambda'()> >, void>::operator()() const (/home/***/nav2_ws/install/nav2_smoother/lib/libsmoother_server_core.so+0x1efe04) (BuildId: 80891eed01337a00457782ceae8a233ce28b113c)
    #3 0x7187e49efb17 in std::enable_if<is_invocable_r_v<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter>, std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::SmoothPath>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::SmoothPath> >)::'lambda'()> >, void>&>, std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> >::type std::__invoke_r<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter>, std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::SmoothPath>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::SmoothPath> >)::'lambda'()> >, void>&>(std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::SmoothPath>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::SmoothPath> >)::'lambda'()> >, void>&) (/home/***/nav2_ws/install/nav2_smoother/lib/libsmoother_server_core.so+0x1efb17) (BuildId: 80891eed01337a00457782ceae8a233ce28b113c)
    #4 0x7187e49ef958 in std::_Function_handler<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> (), std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::SmoothPath>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::SmoothPath> >)::'lambda'()> >, void> >::_M_invoke(std::_Any_data const&) (/home/***/nav2_ws/install/nav2_smoother/lib/libsmoother_server_core.so+0x1ef958) (BuildId: 80891eed01337a00457782ceae8a233ce28b113c)
    #5 0x7187e5cf5b96 in std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*) (/home/***/nav2_ws/install/nav2_util/lib/libnav2_util_core.so+0x7fb96) (BuildId: 1d6c0d5217ab4e048e9a7919e1f2eabd3bac7da1)
    #6 0x7187e3e99ee7 in __pthread_once_slow nptl/./nptl/pthread_once.c:116:7
    #7 0x7187e49ed521 in std::__future_base::_Async_state_impl<std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::SmoothPath>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::SmoothPath> >)::'lambda'()> >, void>::_M_run() (/home/***/nav2_ws/install/nav2_smoother/lib/libsmoother_server_core.so+0x1ed521) (BuildId: 80891eed01337a00457782ceae8a233ce28b113c)
    #8 0x7187e42dc252  (/lib/x86_64-linux-gnu/libstdc++.so.6+0xdc252) (BuildId: e37fe1a879783838de78cbc8c80621fa685d58a2)
    #9 0x7187e3e94ac2 in start_thread nptl/./nptl/pthread_create.c:442:8
    #10 0x7187e3f2684f  misc/../sysdeps/unix/sysv/linux/x86_64/clone3.S:81

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/home/***/nav2_ws/install/nav2_smoother/lib/libsmoother_server_core.so+0x11e19a) (BuildId: 80891eed01337a00457782ceae8a233ce28b113c) in nav2_smoother::SmootherServer::smoothPlan()
Thread T13 created by T12 here:
    #0 0x5ce319b8a87c in __interceptor_pthread_create (/home/***/nav2_ws/install/nav2_smoother/lib/nav2_smoother/smoother_server+0x9387c) (BuildId: 5647db61db99952a548142c13b8b84d860f54c3b)
    #1 0x7187e42dc328 in std::thread::_M_start_thread(std::unique_ptr<std::thread::_State, std::default_delete<std::thread::_State> >, void (*)()) (/lib/x86_64-linux-gnu/libstdc++.so.6+0xdc328) (BuildId: e37fe1a879783838de78cbc8c80621fa685d58a2)
    #2 0x7187e49ece82 in std::__future_base::_Async_state_impl<std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::SmoothPath>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::SmoothPath> >)::'lambda'()> >, void>::_Async_state_impl<nav2_util::SimpleActionServer<nav2_msgs::action::SmoothPath>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::SmoothPath> >)::'lambda'()>(nav2_util::SimpleActionServer<nav2_msgs::action::SmoothPath>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::SmoothPath> >)::'lambda'()&&) (/home/***/nav2_ws/install/nav2_smoother/lib/libsmoother_server_core.so+0x1ece82) (BuildId: 80891eed01337a00457782ceae8a233ce28b113c)
    #3 0x7187e49ec628 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<std::__future_base::_Async_state_impl<std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::SmoothPath>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::SmoothPath> >)::'lambda'()> >, void>, std::allocator<void>, nav2_util::SimpleActionServer<nav2_msgs::action::SmoothPath>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::SmoothPath> >)::'lambda'()>(std::__future_base::_Async_state_impl<std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::SmoothPath>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::SmoothPath> >)::'lambda'()> >, void>*&, std::_Sp_alloc_shared_tag<std::allocator<void> >, nav2_util::SimpleActionServer<nav2_msgs::action::SmoothPath>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::SmoothPath> >)::'lambda'()&&) (/home/***/nav2_ws/install/nav2_smoother/lib/libsmoother_server_core.so+0x1ec628) (BuildId: 80891eed01337a00457782ceae8a233ce28b113c)
    #4 0x7187e49ea833 in std::future<std::__invoke_result<std::decay<nav2_util::SimpleActionServer<nav2_msgs::action::SmoothPath>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::SmoothPath> >)::'lambda'()>::type>::type> std::async<nav2_util::SimpleActionServer<nav2_msgs::action::SmoothPath>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::SmoothPath> >)::'lambda'()>(std::launch, nav2_util::SimpleActionServer<nav2_msgs::action::SmoothPath>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<na [... cut ...]
          z: 0.0
    - header:
        frame_id: map
        stamp:
          nanosec: 690214053
          sec: 1711026163
      pose:
        orientation:
          w: 0.5592876984531523
          x: 0.0
          y: 0.0
          z: -0.8289736246461499
        position:
          x: 0.6187454462051392
          y: -1.2281272411346436
          z: 0.0
smoother_id: simple_smoother" 
```

step 3.  Then press Ctrl+C to let the system automatically execute deactivate() -> cleanup(), and then ASAN will appear.

GoesM, 2024-07-16:
### additional information [analysis]

after our reproduction steps , we could see the log as following:
```

[smoother_server-10] [INFO] [1721160387.372465263] [smoother_server]:  smoothPlan() start------------------------------------------------
...
...
[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
...
...
[smoother_server-10] [INFO] [1721160389.730205529] [smoother_server]: [smooth_path] [ActionServer] Waiting for async process to finish.
[smoother_server-10] [INFO] [1721160389.830371276] [smoother_server]: [smooth_path] [ActionServer] Waiting for async process to finish.
[smoother_server-10] [WARN] [1721160389.830535857] [smoother_server]: [smooth_path] [ActionServer] Aborting handle.
[smoother_server-10] [ERROR] [1721160389.830736971] [smoother_server]: [smooth_path] [ActionServer] Action callback is still running and missed deadline to stop
[INFO] [gzclient-2]: process has finished cleanly [pid 7243]
[smoother_server-10] [INFO] [1721160389.930855536] [smoother_server]: [smooth_path] [ActionServer] Waiting for async process to finish.
[smoother_server-10] [ERROR] [1721160389.930947484] [smoother_server]: [smooth_path] [ActionServer] Action callback is still running and missed deadline to stop
[smoother_server-10] [INFO] [1721160390.031099435] [smoother_server]: [smooth_path] [ActionServer] Waiting for async process to finish.
[smoother_server-10] [ERROR] [1721160390.031270669] [smoother_server]: [smooth_path] [ActionServer] Action callback is still running and missed deadline to stop
[smoother_server-10] [INFO] [1721160390.131388746] [smoother_server]: [smooth_path] [ActionServer] Waiting for async process to finish.
[smoother_server-10] [ERROR] [1721160390.131473725] [smoother_server]: [smooth_path] [ActionServer] Action callback is still running and missed deadline to stop
[smoother_server-10] [INFO] [1721160390.231640617] [smoother_server]: [smooth_path] [ActionServer] Waiting for async process to finish.
[smoother_server-10] [ERROR] [1721160390.231849635] [smoother_server]: [smooth_path] [ActionServer] Action callback is still running and missed deadline to stop
[smoother_server-10] [INFO] [1721160390.332016420] [smoother_server]: [smooth_path] [ActionServer] Waiting for async process to finish.
[smoother_server-10] [ERROR] [1721160390.332118330] [smoother_server]: [smooth_path] [ActionServer] Action callback is still running and missed deadline to stop
[smoother_server-10] [INFO] [1721160390.372714123] [smoother_server]:  smoothPlan() still working------------------------------------------------
[smoother_server-10] [INFO] [1721160390.372880189] [smoother_server]: Received a path to smooth.
[smoother_server-10] [ERROR] [1721160390.372948935] [smoother_server]: [smooth_path] [ActionServer] A goal is not available or has reached a final state
[smoother_server-10] AddressSanitizer:DEADLYSIGNAL
[ERROR] [smoother_server-10]: process has died [pid 7260, exit code 1, cmd '/home/ubuntu2204-humble/ROS_Workstation/nav2_humble/install/nav2_smoother/lib/nav2_smoother/smoother_server --ros-args --log-level info --ros-args -r __node:=smoother_server --params-file /tmp/launch_params_ep7hihaf -r /tf:=tf -r /tf_static:=tf_static'].
```

Notice the log 
`[smooth_path] [ActionServer] A goal is not available or has reached a final state`
and the code line : 

https://github.com/ros-navigation/navigation2/blob/12a9c1d805847709e3b82f8dcfbb43c67b5b2937/nav2_smoother/src/nav2_smoother.cpp#L268

here's the position where the null-pointer is accessed.

BECAUSE following machnasim:

https://github.com/ros-navigation/navigation2/blob/12a9c1d805847709e3b82f8dcfbb43c67b5b2937/nav2_util/include/nav2_util/simple_action_server.hpp#L405-L415

if the action_server_ is not activated, this function will return a shared pointer pointing to a null pointer.
SO THAT,  `c_name` would try to read `nullPtr->smoother_id`, which lead to the NullPtr bug. 

#### suggestiion
shall we add a check for it ?

by the way, the following code lines could also met the similar situation and cause the bug: 
https://github.com/ros-navigation/navigation2/blob/12a9c1d805847709e3b82f8dcfbb43c67b5b2937/nav2_smoother/src/nav2_smoother.cpp#L277-L278

GoesM, 2024-07-16:
I suggest to change the `smoothPlan()` as following : 

```cpp
...
...
 auto result = std::make_shared<Action::Result>();
  try {
+  auto goal = action_server_->get_current_goal();
+  if (goal == nulPtr) return ; 
-   std::string c_name = action_server_->get_current_goal()->smoother_id;
+  std::string c_name = goal->smoother_id;

    std::string current_smoother;
    if (findSmootherId(c_name, current_smoother)) {
      current_smoother_ = current_smoother;
    } else {
      throw nav2_core::InvalidSmoother("Invalid Smoother: " + c_name);
    }

    // Perform smoothing
-   auto goal = action_server_->get_current_goal();
+  // auto goal = action_server_->get_current_goal();
    result->path = goal->path;
...
...

```

SteveMacenski, 2024-07-17:
Seems reasonable to me!

closed 2024-07-19 by SteveMacenski