ros-navigation/navigation2 #4539: Confusion about MPPI Path Angle Critic

nikkijhart, 2024-07-16:
**Required Info:**
Branch: main
Commit: 5a8cbd3

**Description**
We are considering use of the PathAngleCritic to align the angle of the robot with the orientation information produced by kinematically feasible planners. Our understanding is that mode 2 in the PathAngleCritic is suited for this purpose, but there is some confusion about the intent and implementation:

1. Is the intended outcome of using the PathAngleCritic in mode 2 that the robot _heading_ aligns with the orientation of the lookahead point or that the robot _direction of movement_ aligns with the orientation of the lookahead point?
2. Based on [PathAngleCritic code, lines 138-152](https://github.com/ros-navigation/navigation2/blob/12a9c1d805847709e3b82f8dcfbb43c67b5b2937/nav2_mppi_controller/src/critics/path_angle_critic.cpp#L138), higher penalties occur when the trajectories' yaws do not align with the direction of travel (calculated using an arctan between the goal position and each trajectory position). The goal_yaw is only used to determine whether or not to normalize yaws_between_points; therefore, the cost is not associated with the goal_yaw. If this is correct, how does mode 2 of the PathAngleCritic work wrt its stated purpose? 

Thank you.

SteveMacenski, 2024-07-16:
Please ask on robotics stack exchange, this is not the right place for Q&A

MishaSavchenko, 2024-07-16:
https://robotics.stackexchange.com/questions/112254/confusion-about-mppi-path-angle-critic