ros-navigation/navigation2 #4551: Getting this error while navigation : Extrapolation Error looking up target frame: Lookup would require extrapolation into the past

vatsal21770, 2024-07-22:
## Bug report

**Required Info:**

- Operating System: 
  - Ubuntu 20.04
- ROS2 Version:
  -  Humble source
- Version or commit hash: 
  - ros-humble-navigation2	1.1.12-1focal.20231230.114340
- DDS implementation: 
  - eprosima Fast DDS

#### Steps to reproduce issue

```
- Use Isaac ROS vslam and nvblox for SLAM (release 2.1)
- Give A to B navigation goal to robot via Rviz2
```

#### Expected behavior
The Robot should move to given location.

#### Actual behavior
**Instead of following path created robot goes into recovery and throws these warnings and error.**

[planner_server-5] [WARN] [1721202120.203820830] [planner_server]: Planner loop missed its desired rate of 20.0000 Hz. Current loop rate is 13.6258 Hz
[controller_server-3] [INFO] [1721202120.228779368] [controller_server]: Passing new path to controller.
[planner_server-5] [WARN] [1721202121.298797265] [planner_server]: Planner loop missed its desired rate of 20.0000 Hz. Current loop rate is 14.6459 Hz
[controller_server-3] [INFO] [1721202121.328778346] [controller_server]: Passing new path to controller.

[planner_server-5] [WARN] [1721202050.197822187] [planner_server]: Could not transform the start or goal pose in the costmap frame
[planner_server-5] [WARN] [1721202050.197953773] [planner_server]: [compute_path_to_pose] [ActionServer] Aborting handle.
[behavior_server-6] [INFO] [1721202050.211050888] [behavior_server]: Running backup

**[planner_server-5] [ERROR] [1721202164.347566704] [transformPoseInTargetFrame]: Extrapolation Error looking up target frame: Lookup would require extrapolation into the past.  Requested time 1721202114.598340 but the earliest data is at time 1721202154.328349, when looking up transform from frame [map] to frame [odom]**
[planner_server-5] 
[planner_server-5] [WARN] [1721202164.347646962] [planner_server]: Could not transform the start or goal pose in the costmap frame
[planner_server-5] [WARN] [1721202164.347890549] [planner_server]: [compute_path_to_pose] [ActionServer] Aborting handle.
[planner_server-5] [INFO] [1721202164.518482817] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
**[planner_server-5] [ERROR] [1721202164.826386461] [transformPoseInTargetFrame]: Extrapolation Error looking up target frame: Lookup would require extrapolation into the past.  Requested time 1721202114.598340 but the earliest data is at time 1721202154.813274, when looking up transform from frame [map] to frame [odom]**
[planner_server-5] 
[planner_server-5] [WARN] [1721202164.826458046] [planner_server]: Could not transform the start or goal pose in the costmap frame
[planner_server-5] [WARN] [1721202164.826623392] [planner_server]: [compute_path_to_pose] [ActionServer] Aborting handle.
**[bt_navigator-7] [ERROR] [1721202164.848153897] [bt_navigator]: Goal failed**
[bt_navigator-7] [WARN] [1721202164.848368204] [bt_navigator]: [navigate_to_pose] [ActionServer] Aborting handle.

----
#### Environments
- Using nvidia's isaac ros docker container to run launch file of SLAM (using ROS distribution : humble (installed from source)). 
- Running rviz config file from galactic installed in base OS.
- Have taken the SSH and controlling all the things from there.

#### Steps taken
Searched for the errors similar to this and tried the available solutions as well.

[planner_server-5] [ERROR] [1721202164.826386461] [transformPoseInTargetFrame]: Extrapolation Error looking up target frame: Lookup would require extrapolation into the past. Requested time 1721202114.598340 but the earliest data is at time 1721202154.813274, when looking up transform from frame [map] to frame [odom]

- Most of them talks about **use_sim_time** parameter. That I have kept it **False** as I am running all the nodes from same machine not a simulation.
- Tried with increasing transform tolerance value for costmap as well. But It didn't work for me.

Want to know more about this error if anyone have idea. what could be possible reasons for it? Is there any solution available other than I have mentioned?

SteveMacenski, 2024-07-22:
Are you requesting goals in the `map` frame or the `odom` frame? The error implies that the fixed frame of your rviz config might be set to `odom` causing this issue perhaps. 

You clipped your logs, but I assume the timestamp in 

> Requested time 1721202114.598340

Is the time that your goal was sent, yes?  Then `1721202154.813274` is now. That would point to the TF transformation buffer size just being exceeded because you requested in `odom` at time `..14` so its trying to transform data in planning to `map` at time `..54`. If you requested in `map` to begin with, no transformation would be required.

vatsal21770, 2024-08-05:
All the functionality is running from **same computer** (Jetson device). We have taken the SSH for visualization.

If we talk about another **map->odom** source, then yes, we have seen same error with **cartographer** as well some time ago.

SteveMacenski, 2024-08-05:
How's your CPU usage on the Jetson, is it thrashing so computationally intensive things aren't keeping up or callbacks not being executed in a timely manner? Can you reproduce this on another machine? Is the rviz on an external computer connected over wifi (and what happens if you don't have rviz up)? 

I haven't run into this and I've been working on a project based on Humble on some AMD CPUs on a Clearpath robot recently and everything's been working swimmingly. The only situations I've run into anything similar is when I have rviz2 up on an external machine over a wifi connection broadcasting large data using untuned Fast-DDS configs -- which if its all on 1 computer wouldn't be your issue. 

My thoughts are:
- Try Cyclone (even if you can't use it). If that fixes things, its likely a configuration issue that could be resolved in DDS tuning of Fast-DDS or using Cyclone
- Your computer's CPU is thrashing so there's simply not enough compute resources to do what you're trying to do
- You're sending commands in a non-map frame so the Planner Server is trying to transform goals at some older timestamp into current map frame for planning purposes -- in which you should adjust to use the right frame or set the time to `0` to use `now`'s transform.

SteveMacenski, 2024-09-04:
Hi, any updates?

SteveMacenski, 2024-09-09:
Closing due to lack of response for 1 month and I think plausible solutions / problem analysis has been provided that the user likely resolved their problem on their own with no changes needed to Nav2.

closed 2024-09-09 by SteveMacenski