ros-navigation/navigation2 #4552: Nav2 Fails on Second TurtleBot 4

bioturtles, 2024-07-22:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 Jammy Jellyfish
- ROS2 Version:
  - ROS2 Humble
- Version or commit hash:
  - apt binaries: nav2 version 1.1.15
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

```
1) Use Clearpath's new Discovery Servers for the TurtleBot 4
2) Use Clearpath's TurtleBot 4 launch files for localization and rviz2 (launch through a ssh on the TurtleBot)
3) Use Clearpath's TurtleBot 4 launch file for nav2 (launch through a ssh on the TurtleBot)
4) See the errors
```

#### Expected behavior

Nav2 starts and allows the TurtleBot 4 to navigate to a pose.

#### Actual behavior
I am currently using a discovery server configuration provided by Clearpath for the TurtleBot 4. This is an error I have seen in other discovery methods, albeit less frequent. This happens specifically when I have a second robot on the network. In addition, the first robot that I connect to works fine with nav2, but the second one always seems to do this. The discovery server setup means that robot2 can't see robot1's topics, nodes, etc, so I don't think there are any conflicts with topics or services. I get the following errors on the second robot:

```
ubuntu@ubuntu:~$ ros2 launch turtlebot4_navigation nav2.launch.py namespace:=robot1
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-07-22-15-29-02-837515-ubuntu-2814
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [controller_server-1]: process started with pid [2815]
[INFO] [smoother_server-2]: process started with pid [2817]
[INFO] [planner_server-3]: process started with pid [2819]
[INFO] [behavior_server-4]: process started with pid [2821]
[INFO] [bt_navigator-5]: process started with pid [2823]
[INFO] [waypoint_follower-6]: process started with pid [2825]
[INFO] [velocity_smoother-7]: process started with pid [2827]
[INFO] [lifecycle_manager-8]: process started with pid [2829]
[controller_server-1] [INFO] [1721662145.146748478] [robot1.controller_server]: 
[controller_server-1] 	controller_server lifecycle node launched. 
[controller_server-1] 	Waiting on external lifecycle transitions to activate
[controller_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[lifecycle_manager-8] [INFO] [1721662145.158561280] [robot1.lifecycle_manager_navigation]: Creating
[lifecycle_manager-8] [INFO] [1721662145.184529178] [robot1.lifecycle_manager_navigation]: Creating and initializing lifecycle service clients
[controller_server-1] [INFO] [1721662145.213989148] [robot1.controller_server]: Creating controller server
[controller_server-1] [INFO] [1721662145.314477132] [robot1.local_costmap.local_costmap]: 
[controller_server-1] 	local_costmap lifecycle node launched. 
[controller_server-1] 	Waiting on external lifecycle transitions to activate
[controller_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-1] [INFO] [1721662145.321133095] [robot1.local_costmap.local_costmap]: Creating Costmap
[waypoint_follower-6] [INFO] [1721662145.361790062] [robot1.waypoint_follower]: 
[waypoint_follower-6] 	waypoint_follower lifecycle node launched. 
[waypoint_follower-6] 	Waiting on external lifecycle transitions to activate
[waypoint_follower-6] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-6] [INFO] [1721662145.368436951] [robot1.waypoint_follower]: Creating
[planner_server-3] [INFO] [1721662145.424770100] [robot1.planner_server]: 
[planner_server-3] 	planner_server lifecycle node launched. 
[planner_server-3] 	Waiting on external lifecycle transitions to activate
[planner_server-3] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-3] [INFO] [1721662145.443060609] [robot1.planner_server]: Creating
[behavior_server-4] [INFO] [1721662145.463750133] [robot1.behavior_server]: 
[behavior_server-4] 	behavior_server lifecycle node launched. 
[behavior_server-4] 	Waiting on external lifecycle transitions to activate
[behavior_server-4] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-3] [INFO] [1721662145.525078161] [robot1.global_costmap.global_costmap]: 
[planner_server-3] 	global_costmap lifecycle node launched. 
[planner_server-3] 	Waiting on external lifecycle transitions to activate
[... 74 lines cut ...]
[behavior_server-4] [INFO] [1721662158.185255918] [robot1.behavior_server]: Creating behavior plugin spin of type nav2_behaviors/Spin
[behavior_server-4] [INFO] [1721662158.202689774] [robot1.behavior_server]: Configuring spin
[behavior_server-4] [INFO] [1721662158.363310641] [robot1.behavior_server]: Creating behavior plugin backup of type nav2_behaviors/BackUp
[behavior_server-4] [INFO] [1721662158.387278665] [robot1.behavior_server]: Configuring backup
[behavior_server-4] [INFO] [1721662158.607593587] [robot1.behavior_server]: Creating behavior plugin drive_on_heading of type nav2_behaviors/DriveOnHeading
[behavior_server-4] [INFO] [1721662158.617433126] [robot1.behavior_server]: Configuring drive_on_heading
[behavior_server-4] [INFO] [1721662158.747160213] [robot1.behavior_server]: Creating behavior plugin assisted_teleop of type nav2_behaviors/AssistedTeleop
[behavior_server-4] [INFO] [1721662158.770155995] [robot1.behavior_server]: Configuring assisted_teleop
[behavior_server-4] [INFO] [1721662158.862326970] [robot1.behavior_server]: Creating behavior plugin wait of type nav2_behaviors/Wait
[behavior_server-4] [INFO] [1721662158.868871101] [robot1.behavior_server]: Configuring wait
[lifecycle_manager-8] [INFO] [1721662158.918335199] [robot1.lifecycle_manager_navigation]: Configuring bt_navigator
[bt_navigator-5] [INFO] [1721662158.926386064] [robot1.bt_navigator]: Configuring
[lifecycle_manager-8] [ERROR] [1721662163.823181109] [robot1.lifecycle_manager_navigation]: Failed to change state for node: bt_navigator. Exception: bt_navigator/get_state service client: async_send_request failed.
[lifecycle_manager-8] [ERROR] [1721662163.823367254] [robot1.lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
```

This happens with bt_navigator, controller_server, and planner_server, so it is not specifically one of those. If I keep trying, it finally works but it immediately aborts whenever I try navigating; take a look at the video below to see what I mean.

[Screencast from 07-22-2024 11:02:39 AM.webm](https://github.com/user-attachments/assets/09eddeb7-d2bb-435e-9757-30b56b74a965)

I am unsure what the problem is here, but I don't believe it is a nav2 bug. I am just hoping there might be an idea of what the problem is... Thanks for the help!

#### Additional information

If I put the robots on separate Domain ID's, this behavior doesn't seem to occur.

SteveMacenski, 2024-07-22:
Please bring up with robot vendor. We dont maintain / support TB4 here

closed 2024-07-22 by SteveMacenski

daisukes, 2025-02-02:
I bumped into similar issues with this.

> [lifecycle_manager-8] [ERROR] [1721662163.823181109] [robot1.lifecycle_manager_navigation]: Failed to change state for node: bt_navigator. Exception: bt_navigator/get_state service client: async_send_request failed.

I built a custom robot with Jetson Orin and sometimes got this kind of issue while bringing up lifecycle nodes.
The message is generated here and this is called when lifecycle_manager is changing and getting state of the managed noes.
The problem here might be the default timeout, 2 seconds.

- https://github.com/ros-navigation/navigation2/blob/78ea525b3b2eb1ce863937fb12f2cbb0abe7821d/nav2_util/include/nav2_util/service_client.hpp#L86
- https://github.com/ros-navigation/navigation2/blob/78ea525b3b2eb1ce863937fb12f2cbb0abe7821d/nav2_util/include/nav2_util/lifecycle_service_client.hpp#L54

I could reproduce the issue when I use simulator/docker env with restricted CPUs like

```
  navigation-dev:
    extends:
      file: docker-compose-common.yaml
      service: navigation-dev
    profiles:
      - dev
    deploy:
      resources:
        limits:
          cpus: '1'  # or less such as 0.5
```

I could fix the issue by copying `nav2_util` package into workspace, and extend the timeout from 2s to 10s.

@SteveMacenski 

Do you think, it is good to add another lifecycle_manager parameter to deal with this timeout?