ros-navigation/navigation2 #4555: Collision Monitor in Nav2 Not Displaying Polygons Correctly or Following Robot - Possible DDS Implementation Issue

marcusvinicius178, 2024-07-23:
Required Info:

**Operating System:** Ubuntu 22.04
**ROS2 Version:** ROS2 Iron binaries
**Version or commit hash:** dpkg-query --show "ros-iron-navigation2"
**DDS implementation:** Fast-RTPS (rmw_fastrtps_cpp)

1. **Steps to reproduce the issue**

- Follow the [Nav2 collision monitor tutorial](https://docs.nav2.org/tutorials/docs/using_collision_monitor.html#demo-execution).

- Watch and follow the setup as demonstrated in the [collision monitor's creator video](https://www.youtube.com/watch?v=bWliK0PC5Ms&t=3407s) without cherry-picking or checking out the specified hash (since using ROS2 Iron binaries).

- Change the remapping topics of controller_server and nav2_smoother from cmd_vel_**nav** to cmd_vel_**raw.**
- Launch the system with the same launch files and parameter configurations as shown in the video.

- Observe the behavior of the polygon_stop and polygon_slowdown in RViz2. 

2. **Expected behavior**

- The polygon_stop box **should appear continuously and follow the base_footprint** of the robot.
- The **polygon_slowdown box should appear** on rviz2 plugins add button and function as intended.

3. **Actual behavior**

- The polygon_stop **box appears intermittently only when toggling the polygon checkbox in RViz2** but does not follow the base_footprint of the robot.
- The polygon_**slowdown box never appears**.

4. **Additional information**
I have attached a video demonstrating the observed behavior:

[Collision Monitor Fails to Follow the robot base_footprint](https://youtu.be/35oaBAp4srI)

**Similar issues reported by other users:**

[Collision Monitor polygons are frozen in place](https://github.com/ros-navigation/navigation2/issues/3132)
[Collision Monitor in Nav2 not working](https://robotics.stackexchange.com/questions/108474/collision-monitor-in-nav2-not-work)

**My detailed issue description:**

[Link to detailed issue description](https://robotics.stackexchange.com/questions/112280/collision-monitor-in-nav2-not-working-properly-polygons-not-displayed-or-follow)

6. Question:

Am I using the wrong DDS implementation? The author of the video [Collision Monitor – Software-Based Safety Module](https://www.youtube.com/watch?v=bWliK0PC5Ms&t=3407s), Alexey Merzlyakov, mentions that Collision Monitor is **currently using Cyclone DDS** (the context is ROS HUMBLE) due to an issue with Fast DDS where the lifecycle manager occasionally hangs. **Should I switch to Cyclone DDS** for better compatibility and functionality? Or checkout to a specific commit, in which collision monitor is working correctly?  I am using ROS2 Iron.

Thanks for further assistance on this issue.

SteveMacenski, 2024-07-24:
It looks like your QoS settings are wrong on that display? https://github.com/ros-navigation/navigation2/blob/f34ca63020feff92fd8d7d4382a273ef1dabc2bd/nav2_collision_monitor/src/polygon.cpp#L94-L96 that could be contributing I suppose, though I don't have an immediately deep answer as to 'why' it would appear once in that case. 

Is it working properly minus the visualization? Just making sure I shouldn't scrutinize the code or your configuration for possible issues and its just a transport on the visualizations.

marcusvinicius178, 2024-07-25:
Hi Steve thanks for your attention on this question. 

I'm encountering an issue with visualizing polygons **using the default QoS** settings.

When I launch the collision monitor node first, the polygons always appear without issues, provided I set the Durability QoS to **transient_local.** This setting ensures that the data is retained and available to late-joining subscribers. Here's a screenshot of the working setup:

[image]

There is one **interesting detail ** that I have noticed: If I launch the collision monitor node **first** then the polygons always appear with no issues, (although must be mandatory to set the Durability QoS to transient_local). 

When I set both to transient_local it works:
[image]

However, if I launch the simulation first and then the collision monitor node, the behavior becomes unstable. The polygons do not always appear; sometimes only polygon_stop appears, and other times neither of them show up. Despite this, the messages seem to be sent correctly, as echoed in the terminal, but they are not visualized properly:

[image]

Chuck commented on my issue on [robotics stack exchange](https://robotics.stackexchange.com/questions/112280/collision-monitor-in-nav2-not-working-properly-polygons-not-displayed-or-follow?noredirect=1#comment48420_112280)

`"When I followed the link to the tutorial you were using, I saw this statement: Both polygon shapes in the tutorial were set statically. However, there is an ability to dynamically adjust them over time using topic messages containing vertices points for polygons or footprints. For more information, please refer to the configuration guide. which sounds to me like it exactly describes what you're seeing - you get a single static polygon that does not adjust over time. Have you tried testing with dynamic polygons?"`

I believe this is the last behavior I need to fix. I want the polygons **to dynamically follow my robot's base_footprint**. Unfortunately, I did not understand how to set the parameters and launch file for dynamic behavior from the [documentation](https://docs.nav2.org/configuration/packages/collision_monitor/configuring-collision-monitor-node.html) how to set the parameters and launch file to get the dynamic behavior (In Alexey tutorial video I have not checked any additional steps required to set on configuration file). Could you help me configure this to achieve dynamic polygon behavior?

SteveMacenski, 2024-07-25:
The polygons are all published on the `process` rate: https://github.com/ros-navigation/navigation2/blob/9f417c5cf40e149a81c11e552648eaa3dc2a7835/nav2_collision_monitor/src/collision_monitor_node.cpp#L502 - which you should be able to see in your terminal while `echo`ing the topic a stream of data. This again uses [system default QoS](https://github.com/ros-navigation/navigation2/blob/iron/nav2_collision_monitor/src/polygon.cpp#L104) so some of this might be your own issue of introducing transient local into the mix.

I just tested and see no issues, I would again however point to making sure that your cmd_vel is actually configured to get to the collision monitor since you mentioned changing some of those settings. [Process is only called on the cmd_vel trigger.](https://github.com/ros-navigation/navigation2/blob/iron/nav2_collision_monitor/src/collision_monitor_node.cpp#L183) which would explain all of this behavior since publish the polygon at the start with `activate`. So you'd see only the single polygon from the initial state and not updated. If you echo and it appears once and not a live stream, it means the collision monitor itself is not receiving command velocity data and you should review the configuration guide or `main` examples for setting up the collision monitor: https://docs.nav2.org/configuration/packages/configuring-collision-monitor.html

marcusvinicius178, 2024-07-25:
Thanks Steve! 

You were correct the issue relied on using the cmd_vel_raw instead cmd_vel_nav in [navigation.launch](https://github.com/ros-navigation/navigation2/blob/9f417c5cf40e149a81c11e552648eaa3dc2a7835/nav2_bringup/launch/navigation_launch.py#L136) because I was following the [Alexey tutorial](https://www.youtube.com/watch?v=bWliK0PC5Ms) but without following the context...

> So you'd see only the single polygon from the initial state, which was not updated. It gets updated by a re-subscribe since its in the base_footprint frame, so when it gets the new data, rviz automatically converts it into your visualizer's fixed frame.
> This subscription relation to cmd_vel also explains why the polygon boxes appear on rviz2 after the robot starts moving! 

> If you echo and it appears once and not a live stream, it means the collision monitor itself is not receiving command velocity data and you should review the configuration guide or `main` examples for setting up the collision monitor: https://docs.nav2.org/configuration/packages/configuring-collision-monitor.html
>  Finally the data flow is dynamic: https://youtu.be/vnL7_CqBup8

closed 2024-07-25 by marcusvinicius178