ros-navigation/navigation2 #4559: In the running of goToPose(), cancelTask() does not work

atinfinity, 2024-07-25:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble binaries
- Version
  - ros-humble-navigation2  1.1.15-1jammy.20240627.172212
  - ros-humble-nav2-amcl    1.1.15-1jammy.20240530.085224
  - ros-humble-nav2-behavior-tree   1.1.15-1jammy.20240626.151221
  - ros-humble-nav2-behaviors       1.1.15-1jammy.20240627.171322
  - ros-humble-nav2-bringup 1.1.15-1jammy.20240627.172242
  - ros-humble-nav2-bt-navigator    1.1.15-1jammy.20240627.171011
  - ros-humble-nav2-collision-monitor       1.1.15-1jammy.20240627.170404
  - ros-humble-nav2-common  1.1.15-1jammy.20240524.012550
  - ros-humble-nav2-constrained-smoother    1.1.15-1jammy.20240627.171203
  - ros-humble-nav2-controller      1.1.15-1jammy.20240627.171012
  - ros-humble-nav2-core    1.1.15-1jammy.20240627.170419
  - ros-humble-nav2-costmap-2d      1.1.15-1jammy.20240627.164242
  - ros-humble-nav2-dwb-controller  1.1.15-1jammy.20240627.171715
  - ros-humble-nav2-lifecycle-manager       1.1.15-1jammy.20240627.145323
  - ros-humble-nav2-map-server      1.1.15-1jammy.20240530.085325
  - ros-humble-nav2-mppi-controller 1.1.15-1jammy.20240627.170630
  - ros-humble-nav2-msgs    1.1.15-1jammy.20240524.013914
  - ros-humble-nav2-navfn-planner   1.1.15-1jammy.20240627.170727
  - ros-humble-nav2-planner 1.1.15-1jammy.20240627.170730
  - ros-humble-nav2-regulated-pure-pursuit-controller       1.1.15-1jammy.20240627.170814
  - ros-humble-nav2-rotation-shim-controller        1.1.15-1jammy.20240627.171721
  - ros-humble-nav2-rviz-plugins    1.1.15-1jammy.20240627.164256
  - ros-humble-nav2-simple-commander        1.1.15-1jammy.20240524.014726
  - ros-humble-nav2-smac-planner    1.1.15-1jammy.20240627.170819
  - ros-humble-nav2-smoother        1.1.15-1jammy.20240627.171035
  - ros-humble-nav2-theta-star-planner      1.1.15-1jammy.20240627.170840
  - ros-humble-nav2-util    1.1.15-1jammy.20240530.084407
  - ros-humble-nav2-velocity-smoother       1.1.15-1jammy.20240530.094814
  - ros-humble-nav2-voxel-grid      1.1.15-1jammy.20240524.013953
  - ros-humble-nav2-waypoint-follower       1.1.15-1jammy.20240627.171044
- DDS implementation:
  - RMW_IMPLEMENTATION=rmw_cyclonedds_cpp

#### Steps to reproduce issue

I used <https://github.com/atinfinity/megarover_samples_ros2>.

```shell
sudo apt install ros-humble-rmw-cyclonedds-cpp
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
mkdir -p ~/dev_ws/src
cd ~/dev_ws/src
git clone https://github.com/atinfinity/megarover_samples_ros2.git
cd ..
rosdep install -y -i --from-paths src/megarover_samples_ros2
colcon build --symlink-install
source ~/dev_ws/install/setup.bash
```

And, I launched Gazebo and navigation2.

```shell
ros2 launch megarover_samples_ros2 vmegarover_with_sample_world.launch.py
ros2 launch megarover_samples_ros2 vmegarover_navigation.launch.py
```

I launched the following node using BasicNavigator(<https://docs.nav2.org/commander_api/index.html>).

```python
import rclpy
from rclpy.node import Node
from geometry_msgs.msg import PoseStamped
from std_msgs.msg import Empty
from nav2_simple_commander.robot_navigator import BasicNavigator

class ExampleNavigationNode(Node):

    def __init__(self):
        super().__init__('example_navigation')
        self.navigation_cancel_subscriber = self.create_subscription(
            Empty,
            '/navigation/cancel',
            self.sub_navigation_cancel_callback,
            10)
        timer_period = 1.0
        self.timer = self.create_timer(timer_period, self.timer_callback)

        self.goal_pose = PoseStamped()
        self.goal_pose.header.frame_id = 'map'
        self.goal_pose.header.stamp = self.get_clock().now().to_msg()
        self.goal_pose.pose.position.x = 4.0
        self.goal_pose.pose.position.y = 0.0
        self.goal_pose.pose.orientation.w = 1.0
        self.goal_pose.pose.orientation.z = 0.0

        self.navigator = BasicNavigator()
        self.navigator.waitUntilNav2Active()
        self.navigator.goToPose(self.goal_pose)

    def sub_navigation_cancel_callback(self, msg):
        self.get_logger().info("sub_navigation_cancel_callback")
        self.navigator.cancelTask()

    def timer_callback(self):
        # get path to goal
        path = self.navigator.getPath(PoseStamped(), self.goal_pose, use_start=False)

def main():
    rclpy.init()
    node = ExampleNavigationNode()
    rclpy.spin(node)
    node.destroy_node()
    rclpy.shutdown()

if __name__ == "__main__":
    main()
```

And, I run the following command.

```
ros2 topic pub --once /navigation/cancel std_msgs/msg/Empty "{}"
```

I called `getPath()` to debug in timer callback.
In this case, `cancelTask()` does not work.

#### Expected behavior

In the running of `goToPose()`, `cancelTask()` works.

#### Actual behavior

When I called `getPath()` in timer callback.
In this case, `cancelTask()` does not work.

SteveMacenski, 2024-07-25:
`getPath` probably already completed, so there's no action to cancel when your cancel subscription is triggered. `cancelTask` cancels on-going tasks, but getting a path is a sub-200ms (usually) task so it doesn't do anything. `goToPose` is a long-running task the length of your navigation run so that would still be on-going in order to be cancelable.

atinfinity, 2024-07-25:
Thank you for your comment. In this case, I think that the `goToPose()` cancell.

SteveMacenski, 2024-07-25:
The _simple_ navigator assumes one action is called at a time and stores the future for that to use in cancellations / getting feedback. It looks like your application calls multiple actions, so you may need to handle your own futures / cancelations. The Simple Navigator is not meant to be a catch all for advanced applications, but a simple to use and understand baseline for students, prototypers, researchers, or creating proof of concepts.

I wouldn't be opposed to a PR that would maintain a dict of different running actions statuses that you could `cancelTask(taskid='my_curr_task')` with the particular current running action you'd like to cancel, but that was explicitly a non-requirement for the simple navigator. Most applications do not require calling multiple actions simultaneously from the python script.

You can most easily get around this by creating your own client / handling for `getPath` yourself and leave the `goToPose` to the `navigator` object so that you don't call multiple actions using the navigator to override the navigation task's processing state (`getPath` is also a trivial and short action, so that's easy enough).

atinfinity, 2024-07-26:
Thank you for your advice. I understood. I'll consider it.
So, I closed this issue.

closed 2024-07-26 by atinfinity