ros-navigation/navigation2 #4567: Revert Double Spin in #4558 once rclcpp #2589 fixed

SteveMacenski, 2024-07-25:
https://github.com/ros-navigation/navigation2/pull/4558 fixed our CI issues caused by https://github.com/ros2/rclcpp/issues/2589. Revert once resolved and for the test case

SteveMacenski, 2024-09-17:
Has been merged into Jazzy and Rolling, we can revert once new binaries are cut

tonynajjar, 2025-01-10:
> Has been merged into Jazzy and Rolling, we can revert once new binaries are cut (latest as of writing 9/17)

Not really, there was some [fix](https://github.com/ros2/rclcpp/pull/2591) merged for the EventsExecutor but we don't use that; the general [fix](https://github.com/ros2/rclcpp/pull/2595) is still pending. Meanwhile https://github.com/ros-navigation/navigation2/pull/4784 is a backport that fixes the tests in Jazzy

SteveMacenski, 2025-01-27:
New version has been merged for the other executors. We'll wait until the next sync to verify, but this is then unblocked to revert

mini-1235, 2025-08-21:
I ran into a similar issue in my own code, so I did some experiments with this issue

I reverted #4558 and https://github.com/ros-navigation/navigation2/pull/5055/files, running colcon test gives me the following result
```
[  FAILED  ] IsBatteryChargingConditionTestFixture.test_behavior_power_supply_status
[  FAILED  ] IsBatteryLowConditionTestFixture.test_behavior_percentage
[  FAILED  ] IsBatteryLowConditionTestFixture.test_behavior_voltage
```

If I change the spin_all to spin_some in goal_updater, then it will also report a failing test
```
[  FAILED  ] GoalUpdaterTestFixture.test_get_latest_goal_update
```

Specifically, it fails when the subscriber is failing to get the latest data from the publisher, it gets goal_to_update_1 instead of goal_to_update_2

https://github.com/ros-navigation/navigation2/blob/69a60dfacb80eb3c9c2ad47f7590eea8defbdaec/nav2_behavior_tree/test/plugins/decorator/test_goal_updater_node.cpp#L214-L228

Reverting all double spin and make spin_some to spin_all pass all cases, I also think spin_all makes more sense here

closed 2025-08-27 by SteveMacenski