ros-navigation/navigation2 #4570: Nav2 BasicNavigator raises ValueError exception

lucasl84, 2024-07-26:
Hi everyone,

I'm facing an error from `BasicNavigator` from the `ros-galactic-nav2-simple-commander` package. I'm following the examples provided by the package. I'm on ROS Galactic with Ubuntu 22.04. If I understand correctly from the examples, to use `Basicnavigator` it is just a matter of doing:

```python
navigator = BasicNavigator()
navigator.goToPose(goal_pose)  # Where `goal_pose` is an instance of `PoseStamped` properly filled.
```

But then, when simply calling `.goToPose(goal_pose)` it blows up spitting this error:

```bash
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.8/dist-packages/paho/mqtt/client.py", line 3591, in _thread_main
    self.loop_forever(retry_first_connection=True)
  File "/usr/local/lib/python3.8/dist-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.8/dist-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.8/dist-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.8/dist-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.8/dist-packages/paho/mqtt/client.py", line 3033, in _packet_handle
    return self._handle_publish()
  File "/usr/local/lib/python3.8/dist-packages/paho/mqtt/client.py", line 3327, in _handle_publish
    self._handle_on_message(message)
  File "/usr/local/lib/python3.8/dist-packages/paho/mqtt/client.py", line 3570, in _handle_on_message
    on_message(self, self._userdata, message)
  File "/home/ros2/install/ros_bridge/lib/python3.8/site-packages/ros_bridge/__init__.py", line 653, in _on_mqtt_message
    self._process_goal(client, userdata, deserialized_message)
  File "/home/ros2/install/ros_bridge/lib/python3.8/site-packages/ros_bridge/__init__.py", line 635, in _process_goal
    self._navigate_to_goal(message['pose'])
  File "/home/ros2/install/ros_bridge/lib/python3.8/site-packages/ros_bridge/__init__.py", line 626, in _navigate_to_goal
    self._navigator.goToPose(goal_pose)
  File "/opt/ros/galactic/lib/python3.8/site-packages/nav2_simple_commander/robot_navigator.py", line 148, in goToPose
    rclpy.spin_until_future_complete(self, send_goal_future)
  File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/__init__.py", line 222, in spin_until_future_complete
    executor.spin_until_future_complete(future, timeout_sec)
  File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/executors.py", line 288, in spin_until_future_complete
    self.spin_once_until_future_complete(future, timeout_sec)
  File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/executors.py", line 716, in spin_once_until_future_complete
    self.spin_once(timeout_sec)
  File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/executors.py", line 705, in spin_once
    handler, entity, node = self.wait_for_ready_callbacks(timeout_sec=timeout_sec)
  File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/executors.py", line 691, in wait_for_ready_callbacks
    return next(self._cb_iter)
ValueError: generator already executing
```

The weird part is that if I catch this exception and don't do anything, the Nav2 stack will properly start moving the robot until the goal is reached, so I'm wondering what should I be doing in this situation. Should I only catch `ValueError` and ignore the exception? Is this exception maybe normal to be raised by that thread?

Another issue that I spotted is that I'm unable to cancel a goal. When the robot is following its path to reach a goal and call `navigator.cancelTask()` I get:

```bash
[controller_server-8] [INFO] [1721984056.695628675] [mir_1.controller_server]: Passing new path to controller.
[controller_server-8] [INFO] [1721984057.695650003] [mir_1.controller_server]: Passing new path to controller.
[controller_server-8] [INFO] [1721984058.695656721] [mir_1.controller_server]: Passing new path to controller.
[controller_server-8] [INFO] [1721984059.695638605] [mir_1.controller_server]: Passing new path to controller.
[controller_server-8] [INFO] [1721984060.695634357] [mir_1.controller_server]: Passing new path to controller.
[controller_server-8] [INFO] [1721984061.695713383] [mir_1.controller_server]: Passing new path to controller.
[controller_server-8] [INFO] [1721984062.695675648] [mir_1.controller_server]: Passing new path to controller.
[mir_100_bridge-2] [INFO] [1721984063.422832713] [mir_1.mir_1]: Canceling current task.
[controller_server-8] [INFO] [1721984063.695630734] [mir_1.controller_server]: Passing new path to controller.
[controller_server-8] [INFO] [1721984064.695641657] [mir_1.controller_server]: Passing new path to controller.
[controller_server-8] [INFO] [1721984065.695635392] [mir_1.controller_server]: Passing new path to controller.
[mir_100_bridge-2] [INFO] [1721984066.457500050] [mir_1.mir_1]: Canceling current task.
[controller_server-8] [INFO] [1721984066.695661966] [mir_1.controller_server]: Passing new path to controller.
[controller_server-8] [INFO] [1721984067.695665462] [mir_1.controller_server]: Passing new path to controller.
[controller_server-8] [INFO] [1721984068.695673984] [mir_1.controller_server]: Passing new path to controller.
```

In this example I called it twice, as evidenced by the `Cancelling current task` messages, but nothing else happens and the robot continues following its path (is as if the `.cancelTask()` call has no effect at all), no errors at all. I believe I'm not doing anything wrong as no further errors are being displayed on the console.

Does anyone have an insight as why I'm experiencing these issues?

Thanks for everyone's time,
Lucas.

SteveMacenski, 2024-07-26:
Galactic is EOL and doesn't receive support, so I'm closing this as `wontfix` since its explicitly not under support or development anymore. 

I'll say though that since this goes into the path `ros_bridge/__init__.py` and `paho/mqtt/client.py`, this seems unlikely to be a problem with Nav2 and instead whatever is going on with MQTT which I'm not sure why is even in the mix. I assume you introduced MQTT which means this is something in your application glue you should look into.

closed 2024-07-26 by SteveMacenski

lucasl84, 2024-07-26:
Hi Steve,

Thanks for your reply. Yes, I'm currently using MQTT to receive and process a user pose sent to the node and also to communicate data to outside of the node. I haven't got any issues with MQTT + ROS so far (for what I understand MQTT spawns and handles its own thread as well) before implementing this particular feature in the app, that's why I raised the issue as I didn't think MQTT could be the source of the problem. Anyways I'll give this another shot with a more up to date ROS version.

Thanks for your time,
Lucas.