ros-navigation/navigation2 #4571: Exception: According to the loaded plugin descriptions the class nav2_smac_planner::SmacPlannerHybrid with base class type nav2_core::GlobalPlanner does not exist.

atinfinity, 2024-07-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble binaries
- Version
  - ros-humble-navigation2  1.1.15-1jammy.20240627.172212
  - ros-humble-nav2-amcl    1.1.15-1jammy.20240530.085224
  - ros-humble-nav2-behavior-tree   1.1.15-1jammy.20240626.151221
  - ros-humble-nav2-behaviors       1.1.15-1jammy.20240627.171322
  - ros-humble-nav2-bringup 1.1.15-1jammy.20240627.172242
  - ros-humble-nav2-bt-navigator    1.1.15-1jammy.20240627.171011
  - ros-humble-nav2-collision-monitor       1.1.15-1jammy.20240627.170404
  - ros-humble-nav2-common  1.1.15-1jammy.20240524.012550
  - ros-humble-nav2-constrained-smoother    1.1.15-1jammy.20240627.171203
  - ros-humble-nav2-controller      1.1.15-1jammy.20240627.171012
  - ros-humble-nav2-core    1.1.15-1jammy.20240627.170419
  - ros-humble-nav2-costmap-2d      1.1.15-1jammy.20240627.164242
  - ros-humble-nav2-dwb-controller  1.1.15-1jammy.20240627.171715
  - ros-humble-nav2-lifecycle-manager       1.1.15-1jammy.20240627.145323
  - ros-humble-nav2-map-server      1.1.15-1jammy.20240530.085325
  - ros-humble-nav2-mppi-controller 1.1.15-1jammy.20240627.170630
  - ros-humble-nav2-msgs    1.1.15-1jammy.20240524.013914
  - ros-humble-nav2-navfn-planner   1.1.15-1jammy.20240627.170727
  - ros-humble-nav2-planner 1.1.15-1jammy.20240627.170730
  - ros-humble-nav2-regulated-pure-pursuit-controller       1.1.15-1jammy.20240627.170814
  - ros-humble-nav2-rotation-shim-controller        1.1.15-1jammy.20240627.171721
  - ros-humble-nav2-rviz-plugins    1.1.15-1jammy.20240627.164256
  - ros-humble-nav2-simple-commander        1.1.15-1jammy.20240524.014726
  - ros-humble-nav2-smac-planner    1.1.15-1jammy.20240627.170819
  - ros-humble-nav2-smoother        1.1.15-1jammy.20240627.171035
  - ros-humble-nav2-theta-star-planner      1.1.15-1jammy.20240627.170840
  - ros-humble-nav2-util    1.1.15-1jammy.20240530.084407
  - ros-humble-nav2-velocity-smoother       1.1.15-1jammy.20240530.094814
  - ros-humble-nav2-voxel-grid      1.1.15-1jammy.20240524.013953
  - ros-humble-nav2-waypoint-follower       1.1.15-1jammy.20240627.171044
- DDS implementation:
  - RMW_IMPLEMENTATION=rmw_cyclonedds_cpp

#### Steps to reproduce issue

I set parameter to use Smac Hybrid-A* Planner using <https://docs.nav2.org/configuration/packages/smac/configuring-smac-hybrid.html> as a reference.

```yaml
planner_server:
  ros__parameters:
    planner_plugins: ["GridBased"]
    use_sim_time: True

    GridBased:
      plugin: "nav2_smac_planner::SmacPlannerHybrid" # In Iron and older versions, "/" was used instead of "::"
      downsample_costmap: false           # whether or not to downsample the map
      downsampling_factor: 1              # multiplier for the resolution of the costmap layer (e.g. 2 on a 5cm costmap would be 10cm)
      tolerance: 0.25                     # dist-to-goal heuristic cost (distance) for valid tolerance endpoints if exact goal cannot be found.
      allow_unknown: true                 # allow traveling in unknown space
      max_iterations: 1000000             # maximum total iterations to search for before failing (in case unreachable), set to -1 to disable
      max_on_approach_iterations: 1000    # Maximum number of iterations after within tolerances to continue to try to find exact solution
      max_planning_time: 5.0              # max time in s for planner to plan, smooth
      motion_model_for_search: "DUBIN"    # Hybrid-A* Dubin, Redds-Shepp
      angle_quantization_bins: 72         # Number of angle bins for search
      analytic_expansion_ratio: 3.5       # The ratio to attempt analytic expansions during search for final approach.
      analytic_expansion_max_length: 3.0  # For Hybrid/Lattice nodes: The maximum length of the analytic expansion to be considered valid to prevent unsafe shortcutting
      analytic_expansion_max_cost: 200.0  # The maximum single cost for any part of an analytic expansion to contain and be valid, except when necessary on approach to goal
      analytic_expansion_max_cost_override: false  #  Whether or not to override the maximum cost setting if within critical distance to goal (ie probably required)
      minimum_turning_radius: 0.40        # minimum turning radius in m of path / vehicle
      reverse_penalty: 2.0                # Penalty to apply if motion is reversing, must be => 1
      change_penalty: 0.0                 # Penalty to apply if motion is changing directions (L to R), must be >= 0
      non_straight_penalty: 1.2           # Penalty to apply if motion is non-straight, must be => 1
      cost_penalty: 2.0                   # Penalty to apply to higher cost areas when adding into the obstacle map dynamic programming distance expansion heuristic. This drives the robot more towards the center of passages. A value between 1.3 - 3.5 is reasonable.
      retrospective_penalty: 0.015
      lookup_table_size: 20.0             # Size of the dubin/reeds-sheep distance window to cache, in meters.
      cache_obstacle_heuristic: false     # Cache the obstacle map dynamic programming distance expansion heuristic between subsiquent replannings of the same goal location. Dramatically speeds up replanning performance (40x) if costmap is largely static.
      debug_visualizations: false         # For Hybrid nodes: Whether to publish expansions on the /expansions topic as an array of poses (the orientation has no meaning) and the path's footprints on the /planned_footprints topic. WARNING: heavy to compute and to display, for debug only as it degrades the performance.
      use_quadratic_cost_penalty: False
      downsample_obstacle_heuristic: True
      allow_primitive_interpolation: False
      smooth_path: True                   # If true, does a simple and quick smoothing post-processing to the path

      smoother:
        max_iterations: 1000
        w_smooth: 0.3
        w_data: 0.2
        tolerance: 1.0e-10
        do_refinement: true
        refinement_num: 2
```

#### Expected behavior

I could use Smac Hybrid-A* Planner as planner.

#### Actual behavior

But, it occurs the following error.

```
[component_container_isolated-1] [FATAL] [1722174094.878101603] [planner_server]: Failed to create global planner. Exception: According to the loaded plugin descriptions the class nav2_smac_planner::SmacPlannerHybrid with base class type nav2_core::GlobalPlanner does not exist. Declared types are  nav2_navfn_planner/NavfnPlanner nav2_smac_planner/SmacPlanner2D nav2_smac_planner/SmacPlannerHybrid nav2_smac_planner/SmacPlannerLattice nav2_theta_star_planner/ThetaStarPlanner
[component_container_isolated-1] [INFO] [1722174094.878229752] [global_costmap.global_costmap]: StaticLayer: Resizing costmap to 169 X 188 at 0.050000 m/pix
[component_container_isolated-1] [ERROR] [1722174094.878351698] [lifecycle_manager_navigation]: Failed to change state for node: planner_server
[component_container_isolated-1] [ERROR] [1722174094.878376186] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
```

#### Additional information

I fixed nav2 configuration.

```diff
-      plugin: "nav2_smac_planner::SmacPlannerHybrid"
+      plugin: "nav2_smac_planner/SmacPlannerHybrid"
```

As a result, I could use Smac Hybrid-A* Planner.
So, I think that the documentation is old. And, other planners have the same problem.

SteveMacenski, 2024-07-28:
The documentation mentions this in the blob you posted 🙂

> In Iron and older versions, "/" was used instead of "::"

atinfinity, 2024-07-29:
Thank you for your reply. I understood.
Sorry. I overlooked this explanation.

closed 2024-07-29 by atinfinity