ros-navigation/navigation2 #4573: setCostmap function in Footprint Collision Checker

anath93, 2024-07-29:
- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - ROS2 Iron
  
- DDS implementation:
  -  rmw_implementation 

Question/Issue:

Under Python API for FootprintCollisionChecker, sorry I did not find much documentation on this and could not get my question engaged on Robotics Stack.
setCostmap function, can this be fed from topic or getGlobalCostmap ? As I do not get why are we recreating occupancy grid message to pass it in when we already have costmap published with that info ?
Any navigators on additional info/small example would be appreciated for this.

SteveMacenski, 2024-07-29:
Please ask on robot stack exchange. This is not the appropriate place for Q&A

closed 2024-07-29 by SteveMacenski

anath93, 2024-07-29:
@SteveMacenski I thought so, but did not had anyone engage it when I previously posted a few days ago but reposted again. I know you have a lot on your plate, but if you get a chance one of these days please do reply on Robotics forum page. 

https://robotics.stackexchange.com/questions/112436/footprint-collision-checker-nav2

SteveMacenski, 2024-07-29:
It was only up for 2 days over the weekend. You need to be patient.