ros-navigation/navigation2 #4577: About Nav2 CycloneDDS and Micro XRCE-DDS

frkngltk, 2024-07-30:
## About Nav2 CycloneDDS and Micro XRCE-DDS

**System Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble
- DDS implementation:
  - Nav2 -> CycloneDDS
  - Control Card -> Micro XRCE-DDS

Hello;

I have a robot that I actively use with Nav2 (CycloneDDS). The speed data coming from the control card used on this robot comes with the Micro XRCE-DDS communication protocol. In addition, the speed data (cmd_vel) broadcasted from the Nav2 system goes to the control card in real time and allows the robot to move. (Diff drive control)

I am encountering a problem here. The cmd_vel data coming from the Nav2 system comes as 0 (only comes once) linearly and angularly when the robot reaches the relevant point. However, this data is not transferred to the control card. My main question here is:
- Is there a problem in the communication infrastructure of the CycloneDDS architecture used in the system with the Micro XRCE-DDS? (Communication is made via UART.) If there is such a problem, what can I do to solve this problem?
- How can I ensure that the 0 data broadcasted with cmd_vel is broadcasted more than once when the relevant point is reached in the Nav2 system?

I think the questions mentioned above are related to DDS. I am waiting for your suggestions regarding this issue.

SteveMacenski, 2024-07-30:
As you point out, this is probably an issue with your system / XRCE-DDS and not something w.r.t. Nav2. I don't think the maintainers here will have a silver bullet for you since it is going to be pretty dependent on your hardware, DDS vendor, and so forth. Unless they've run into the exact same problem, I don't think we can offer suggestions in this case. Try reaching out on the XRCE github about data coming through not being populated.

closed 2024-07-30 by SteveMacenski

frkngltk, 2024-07-31:
Thanks for the feedback.