ros-navigation/navigation2 #4584: AMCL is unable to correctly publish the map->odom coordinate transformation.

suixing108, 2024-08-01:
Hello, I am encountering some issues and it seems that AMCL is not working properly.I would greatly appreciate any assistance you can provide in resolving this issue.
### Environment Description:
* Physical car
* Ubuntu 24.04
* ROS2-Jazzy
### Problem Description:
When using Nav2, AMCL is not publishing the map->odom transformation correctly, resulting in the following log message:
```bash
[component_container_isolated-1] [INFO] [1722479581.777346684] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
```
### Specific Details and Context:
The car's LiDAR and odometer are functioning normally.
The constructed grid map information can be loaded correctly (as shown in Figure 1).
AMCL can manage the lifecycle normally (as shown in Figure 2).
AMCL shows it can subscribe to the map topic correctly (as shown in Figure 3).
The AMCL lifecycle indicates it is in an active state (as shown in Figure 4).
The static layer subscribes to the /map topic (as shown in Figure 5).
Specific error information (INFO level) (as shown in Figure 6).
TF tree (does not publish the map->odom transformation information), partial topic communications shown in Figures 7 and 8.
Launch and specific params information are as follows:
```
def generate_launch_description():
    robot_navigation2_dir = get_package_share_directory('robot_nav')
    nav2_bringup_dir = get_package_share_directory('nav2_bringup')

    use_sim_time = LaunchConfiguration('use_sim_time', default='false')
    map_yaml_path = LaunchConfiguration('map', default=os.path.join(robot_navigation2_dir, 'maps', 'map.yaml'))
    nav2_param_path = LaunchConfiguration('params_file', default=os.path.join(robot_navigation2_dir, 'param', 'robot_params.yaml'))

    nav2_bringup_launch = IncludeLaunchDescription(
        PythonLaunchDescriptionSource([nav2_bringup_dir, '/launch', '/bringup_launch.py']),
        launch_arguments={
            'map': map_yaml_path,
            'use_sim_time': use_sim_time,
            'params_file': nav2_param_path
        }.items(),
    )
```
```yaml
amcl:
  ros__parameters:
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_link"
#    base_frame_id: "odom"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
[... 379 lines cut ...]
    # The following example illustrates configuring dock instances.
    # docks: ['home_dock']  # Input your docks here
    # home_dock:
    #   type: 'simple_charging_dock'
    #   frame: map
    #   pose: [0.0, 0.0, 0.0]

    controller:
      k_phi: 3.0
      k_delta: 2.0
      v_linear_min: 0.15
      v_linear_max: 0.15

```

[image]
[image]
[image]
[image]
[image]
[image]

### Attempted Solutions:
Replaced AMCL with Gmapping for the map->odom transformation. Navigation works correctly without any errors, but the inputted prior map cannot be displayed.
Started the map_server first to independently manage the lifecycle of the prior map, but the error "cannot find map->odom" still occurs.

SteveMacenski, 2024-08-01:
Is the scan actually on the `scan` topic? This seems like a classic: everything's setup but you didn't give it data yet. Or, you didn't set the robot's initial pose so AMCL isn't initialized yet. Your clipped logs make this difficult to analyze. 

I'm pretty convinced that this works since I use it every day, there's just something AMCL's not being given - I think - that it needs to get going. That's either a map, initial pose, or scan. 

https://github.com/ros-navigation/navigation2/blob/main/nav2_amcl/src/amcl_node.cpp#L1562 --> this means the subscription was created, not that it was received. You'd see this log: https://github.com/ros-navigation/navigation2/blob/main/nav2_amcl/src/amcl_node.cpp#L1419 if a map was received. Maybe your map topic is different or not set up using the correct QoS?

suixing108, 2024-08-02:
Thank you very much for your reply, it has been very inspiring. To more accurately identify the problem, I will provide more details and look forward to your further response.

### Problem Description

Before I start the robot, my ROS graph and topic list are as follows:
[image]

```bash
r@r:~$ ros2 topic list 
/cmd_vel
/diagnostics
/imu
/joint_states
/lslidar_driver_node/transition_event
/lslidar_order
/odom
/parameter_events
/robot_description
/rosout
/scan
/tf
/tf_static
```
### Launch File and Configuration File
Here is my launch file. In fact, I have only specified the path to the configuration file. The static transform is for my testing purposes, and I have already commented it out.

```bash
import os
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.actions import IncludeLaunchDescription, RegisterEventHandler
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import Node
from launch.event_handlers import OnProcessExit

def generate_launch_description():
    # 1. 定位到包的地址
    robot_navigation2_dir = get_package_share_directory('robot_nav')
    nav2_bringup_dir = get_package_share_directory('nav2_bringup')

    # 2. 声明参数，获取配置文件路径
    use_sim_time = LaunchConfiguration('use_sim_time', default='false')
    map_yaml_path = LaunchConfiguration('map', default=os.path.join(robot_navigation2_dir, 'maps', 'map.yaml'))
    nav2_param_path = LaunchConfiguration('params_file', default=os.path.join(robot_navigation2_dir, 'param', 'robot_params.yaml'))

    # # 3. 添加静态变换
    # static_transform_publisher = Node(
    #     package='tf2_ros',
    #     executable='static_transform_publisher',
    #     name='static_transform_publisher',
    #     output='screen',
    #     arguments=[
    #         '0.0', '0.0', '0.0',  # translation
    #         '0.0', '0.0', '0.0', '1.0',  # quaternion (w, x, y, z)
    #         'map', 'odom',  # source_frame, target_frame
    #     ]
    # )

    # 4. 声明启动launch文件，传入：地图路径、是否使用仿真时间以及nav2参数文件
    nav2_bringup_launch = IncludeLaunchDescription(
        PythonLaunchDescriptionSource([nav2_bringup_dir, '/launch/bringup_launch.py']),
        launch_arguments={
            'map': map_yaml_path,
            'use_sim_time': use_sim_time,
            'params_file': nav2_param_path
        }.items(),
    )
    #
    # # 5. 创建事件处理程序，确保静态变换发布后再启动nav2_bringup
[... 878 lines cut ...]
[component_container_isolated-1] [WARN] [1722568940.151821625] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[component_container_isolated-1] [WARN] [1722568942.198193911] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[component_container_isolated-1] [WARN] [1722568944.199508955] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
```
Here is the ROS graph at this time:
[image]

Since the `map` coordinate frame cannot be selected in rviz2's Fixes Frame, I cannot set the initial pose at this time. It can be seen that `scan` has data. What puzzles me is that the map cannot be displayed in` rviz2` at this time. However, as can be seen from the log file, the map information has already been read. Below is `map.yaml`:

```
image: map.pgm
mode: trinary
resolution: 0.05
origin: [-10, -10, 0]
negate: 0
occupied_thresh: 0.65
free_thresh: 0.25
```
[image]
When I check the TF tree, I find that there is still no transformation relationship from map->odom.

[image]
​	I think it might be that my amcl is not correctly publishing the map->odom coordinate transformation. Could it be that the selection of coordinate frames in my configuration file has some issues? I am a bit confused about whether the configuration file should include the` /`.

​	The above is a more detailed description of my problem. I would be honored if you could reply to me again. Thank you once again!

SteveMacenski, 2024-08-02:
> [component_container_isolated-1] [INFO] [1722568867.496972710] [amcl]: Received a 384 X 384 map @ 0.050 m/pix

You have a map :+1: 

> [component_container_isolated-1] [INFO] [1722568867.617009788] [amcl]: createLaserObject

You're getting laser scan data :+1: 

> [component_container_isolated-1] [WARN] [1722568869.460232746] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...

This is your issue. You need to set the initial pose for AMCL to use. Then it'll start publishing. Otherwise, it has no idea where it starts and would just publish a garbage initial transformation. This can be done for testing in rviz using the default "2D pose estimate" tool, or via a publisher to the `/initialpose` topic (or from parameters, or global localization service to have it try to 'guess' where it is with some motion). 

This is all covered in our getting started tutorial https://docs.nav2.org/getting_started/index.html

SteveMacenski, 2024-08-07:
Closing - I believe this is the clear answer given the logs and there hasn't been a response in a week. Happy to engage still if there's anything more.

closed 2024-08-07 by SteveMacenski

donymorph, 2024-11-05:
at the beginning rviz was throwing the same error messages for a while simply added static transform in order to ged rid of map->odom transform errors

    `map2odomlink=Node(`
            package='tf2_ros',
          executable='static_transform_publisher',
            name='static_transform_publisher',
            arguments=['0', '0', '0.15', '0', '0', '0', 'map', 'odom'],
        )

[https://github.com/donymorph/racecar-ROS2-foxy/blob/foxy/src/racecar_bringup/launch/racecar.launch.py](url)

BengIslam7, 2025-04-15:
Same problem : [planner_server-3] [INFO] [1744717285.000978969] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
Any solution?