ros-navigation/navigation2 #4587: Create BT action node RemoveInCollisionGoals

tonynajjar, 2024-08-01:
## Feature request

#### Feature description

Similarly to the existing `RemovePassedGoals`, I propose creating `RemoveInCollisionGoals` that will make use of the collision checkers to remove goals that are in collision. 
@SteveMacenski anything against that? If not I can work on it soon

#### Implementation considerations

SteveMacenski, 2024-08-01:
Sure thing! The only implementation consideration is where is it done, since working on the `global_costmap` is quite the data structure and opening up more subscriptions to it should be avoided, if possible and clean. 

Maybe planner server? Maybe actually just in the costmap itself as a service option to validate points? We have that new `GetCost` service, maybe that can be used under the hood for the BT node ... 

https://github.com/ros-navigation/navigation2/pull/4546

tonynajjar, 2024-08-01:
> opening up more subscriptions to it should be avoided, if possible and clean.

Ah, I thought I could use `local_collision_checker_` and `global_collision_checker_` from the behavior server but I guess it doesn't make sense for the `RemoveInCollisionGoals` server/service to be a behavior like `wait`, `spin`, etc... I'm just wondering why not? Right now it would feel out of place because these are behaviors that generate movement. Any other reason?

SteveMacenski, 2024-08-01:
> I'm just wondering why not? Right now it would feel out of place because these are behaviors that generate movement. Any other reason?

Those are mostly actions that move the robot to create a behavior. That does seems a little weird, but not out of the question. Seems like something in the planner server might be more natural in that case. 

Also, if there's already a `GetCost` service in the costmaps, can't the BT node just call that to get the cost at the goal and gauge if its occupied to remove it? It seems like everything you need is already implemented, just need the BT node to call it. Not sure why it would need to be any more complicated.

SteveMacenski, 2024-08-02:
We could also modify it to get costs (plural) so you don’t need multiple calls

closed 2024-08-08 by SteveMacenski