ros-navigation/navigation2 #4596: [smac_planner] On Hybrid A*, any reason to use floor to get the quantizied angles?

corot, 2024-08-06:
This is a question (with a potential follow up PR) so someone please tag it accordingly.

Hybrid A* operates with quantizied angles (72 bins by default) so it converts all angles (including start and goal poses) to a bin.
It does that by dividing the angles by angle_bin_size (2pi / 72) and getting the integer part using floor. This introduces quantization errors of up to 0.087266463 radians, which can be a problem in some cases, e.g if the goal yaw tolerance is smaller than that number.

On [this PR](https://github.com/rapyuta-robotics/smac_planner/pull/2) over our backport of smac_planner to ROS 1, I'm replacing floor with round, halving the worst case error to 0,043633231.
I can open a similar PR to this repo, if the maintainers consider it a positive change.

SteveMacenski, 2024-08-06:
Mostly to handle the situation that we don't overflow and because the motion primitives are exact increments of angular bins, so there's no chance that during the planning that we have to round the angles, they are exactly searched. 

A port back here would be good I think! Just make sure to update the lattice planner with the same changes

SteveMacenski, 2024-08-21:
@corot https://github.com/ros-navigation/navigation2/pull/4636 implements for Nav2, I'd appreciate a review, but its a minor riff on yours

closed 2024-08-21 by SteveMacenski

corot, 2024-08-22:
> @corot #4636 implements for Nav2, I'd appreciate a review, but its a minor riff on yours

Ah,,, thanks and sorry for not doing the PR myself

SteveMacenski, 2024-08-23:
@corot I think I found a potential issue with this approach. See https://github.com/ros-navigation/navigation2/blob/3233b39eeb20bf6fe5503dc76dc48d23a67cea85/nav2_smac_planner/src/analytic_expansion.cpp#L225

This will implicitly `floor` from the `unsigned int` casting. Should we `round` and check the `% bin_size` when converting into the index?

corot, 2024-08-26:
> This will implicitly `floor` from the `unsigned int` casting. Should we `round` and check the `% bin_size` when converting into the index?

I think it's fine.  `angle` is already a bin, so it should be always be an integer (I made a fast check and indeed it is)