ros-navigation/navigation2 #4612: Collision Monitor parameters on ROS2 Humble

EnzoGhisoni, 2024-08-09:
**Required Info:**

- Operating System:
Ubuntu 22.04

- ROS2 Version:
ROS2 Humble

- Version or commit hash:
nav2 1.1.15

## Feature request

We are working with a collision_monitor on our robot. According to the [nav2](https://docs.nav2.org/configuration/packages/collision_monitor/configuring-collision-monitor-node.html) documentation the parameters <source name>.enabled and <polygon_name name>.enabled are supposed to be dynamic but they are not available on ROS2 humble (but are implemented for Iron and Jazzy). We would like to avoid the ROS2 version to Iron (EOL) and Jazzy (Ubuntu 22.04).
Is a port of this feature planned for ROS2 Humble?

SteveMacenski, 2024-08-09:
Can you find the commit that it was introduced? It may be API/ABI incompatible and thus not backportable. If it is backportable, then I'd be happy to have a PR that backports to Humble so you can use it!

EnzoGhisoni, 2024-08-10:
Here is the pull request where the parameter has been introduced: https://github.com/ros-navigation/navigation2/pull/3825 and it seems to be the commit [c2d84dfe671077bd76781ea356141b6d7ebb86f4](https://github.com/ros-navigation/navigation2/commit/c2d84dfe671077bd76781ea356141b6d7ebb86f4)

EnzoGhisoni, 2024-08-12:
I will try to send a pull request to add these features, at first look the changes seem to be backportable.

SteveMacenski, 2024-08-12:
Great! I took a look and it looks good to me. Lets continue the discussion in that PR!

closed 2024-08-12 by SteveMacenski