ros-navigation/navigation2 #4618: The default_server_timeout parameter in bt_navigator is always overridden by the server_timeout port in each BT action node.

alanxuefei, 2024-08-14:
## Bug report

The `default_server_timeout` parameter in `bt_navigator` is always overridden by the `server_timeout` port in each BT action node. 

As a result, the `default_server_timeout` never takes effect because the `server_timeout` in each BT action node is mandatory and supersedes the default value.

Please refer to
https://github.com/ros-navigation/navigation2/blob/main/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_server_impl.hpp#L185

The code

```
    server_timeout_ = config().blackboard->template get<std::chrono::milliseconds>("server_timeout");
    getInput<std::chrono::milliseconds>("server_timeout", server_timeout_);
```

may be changed as follows:

```cpp
if (!getInput<std::chrono::milliseconds>("server_timeout", server_timeout_)) {
    server_timeout_ = config().blackboard->template get<std::chrono::milliseconds>("server_timeout");
}
```
   

**Required Info:**

- **Operating System:**
  - Ubuntu 22.04
- **ROS2 Version:**
  - Humble binaries
- **Version or commit hash:**
  - Main
- **DDS implementation:**
  - Fast-RTPS

#### Steps to reproduce issue

1. Create a Behavior Tree (BT) action node in a Nav2 configuration file.
2. Set a default `server_timeout` value in the corresponding YAML file.
3. Execute the BT with the action node.
4. Observe that the default value of `server_timeout` from the YAML file is not applied.

```cpp
// Code in bt_action_node.hpp:
server_timeout_ = config().blackboard->template get<std::chrono::milliseconds>("server_timeout");
getInput<std::chrono::milliseconds>("server_timeout", server_timeout_);
```

#### Expected behavior

The default value of `server_timeout` specified in the YAML file should be applied unless overridden by a specific BT action node configuration.

#### Actual behavior

The default value of `server_timeout` from the YAML file is never applied because `server_timeout_` is required to be explicitly set for each BT action node, which overrides the default.

#### Additional information

This bug prevents the default `server_timeout` value specified in the YAML configuration from being used in BT action nodes. As a result, users must manually set `server_timeout` for each node, leading to potential inconsistencies and additional configuration overhead.
```

SteveMacenski, 2024-08-14:
Makes sense. Seems like another potential use case for BT Utils: https://github.com/ros-navigation/navigation2/blob/main/nav2_behavior_tree/include/nav2_behavior_tree/bt_utils.hpp#L184C6-L201 to get either the input (when set) or the blackboard. 

We'd have the change all the BT nodes not to have defaults in that case then so that `getInput` would return false when unset to use the global timeout. 

That sounds like a good, straightforward contribution @alanxuefei !

alanxuefei, 2024-08-15:
I will create a pull request using getInputPortOrBlackboard() this week.

SteveMacenski, 2024-08-31:
To be merged imminently

closed 2024-08-31 by SteveMacenski