ros-navigation/navigation2 #4628: Planner for uneven outdoor surfaces

ShyamKrishna33, 2024-08-20:
I am designing an outdoor robot, which needs to autonomously navigate in uneven terrain with stones and rocks. The robot could dodge these MINOR obstacles in the best-case scenario and go on top of them in the worst-case scenario. But currently NAV2 does not support this kind of implementation as in any obstacle cannot be passed on top of by the robot. Any suggestions for the changes that needs to be done on the source code to meet the expected behavior?

BCKSELFDRIVEWORLD, 2024-08-20:
Maybe you might find this approach too basic. You can modify by defining different cost_values within the obstacle_layer. Assign lower cost values to minor obstacles like small stones, allowing the robot to go over them if no better path is available. Additionally, you might implement a dynamic threshold system to evaluate the obstacle's size, determining if the robot can safely pass over it.

SteveMacenski, 2024-08-20:
Please look online, Robotics Stack Exchange, etc. If you still cannot find your answer, file a new question on Robotics Stack Exchange.  This is not a Q&A platform.

closed 2024-08-20 by SteveMacenski