ros-navigation/navigation2 #4650: Building nav2 from source fails, ubuntu 22.04 ROS2 Humble

ming0071, 2024-08-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04.4 LTS
- ROS2 Version:
  - ROS_DISTRO=humble
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```
mkdir -p ~/nav2_ws/src 
cd ~/nav2_ws/src 
git clone https://github.com/ros-navigation/navigation2.git -b humble
cd ~/nav2_ws 
rosdep install -y -r -q --from-paths src --ignore-src --rosdistro humble
colcon build --symlink-install

```

#### Expected behavior
Crash when compiling to costmap queue
#### Actual behavior
[image]

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

ming0071, 2024-08-27:
> ## Bug report
> **Required Info:**
> 
> * Operating System:
>   
>   * Ubuntu 22.04.4 LTS
> * ROS2 Version:
>   
>   * ROS_DISTRO=humble
> * ## Version or commit hash:
> * ## DDS implementation:
> 
> #### Steps to reproduce issue
> ```
> mkdir -p ~/nav2_ws/src 
> cd ~/nav2_ws/src 
> git clone https://github.com/ros-navigation/navigation2.git -b humble
> cd ~/nav2_ws 
> rosdep install -y -r -q --from-paths src --ignore-src --rosdistro humble
> colcon build --symlink-install
> ```
> 
> #### Expected behavior
> Crash when compiling to costmap queue at 54%
> 
> #### Actual behavior
> ![S__24076290](https://private-user-images.githubusercontent.com/112502009/361799760-4bf3060c-00f4-48c4-85cc-51ce510d9ff6.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ3NjE5MzgsIm5iZiI6MTcyNDc2MTYzOCwicGF0aCI6Ii8xMTI1MDIwMDkvMzYxNzk5NzYwLTRiZjMwNjBjLTAwZjQtNDhjNC04NWNjLTUxY2U1MTBkOWZmNi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgyN1QxMjI3MThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hOTU3NDdmYjQ4ZDYzYmRiOGE4NTUwNGE2NGE4MDBlZTI5MmE2N2FhMTBmZmNmMjgwZjI1NjFhNWU2NTc2ZjE5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.fA06yJMlRrlzCxq_mGbqEkTmyeHfqun_9GUv3zkdOCA)
> 
> #### Additional information
> ## Feature request
> #### Feature description
> #### Implementation considerations

SteveMacenski, 2024-08-27:
Do you not have enough memory? Try compiling with only 1 package in parallel. You have 15 ongoing, you’re probably just running out of memory.

ming0071, 2024-08-28:
Yes, I only have 32GB of memory. After monitoring the system, I found that the memory was indeed saturated. I successfully compiled it using the following method. Since the issue has been resolved, I will close the comment.
```
mkdir -p ~/nav2_ws/src 
cd ~/nav2_ws/src 
git clone https://github.com/ros-navigation/navigation2.git -b humble
cd ~/nav2_ws 
# The successful method is to not pre-install related dependencies, 
# but simply install what is missing as needed.
# rosdep install -y -r -q --from-paths src --ignore-src --rosdistro humble
colcon build
```

closed 2024-08-28 by ming0071