ros-navigation/navigation2 #4651: Keepout filter mask reload on the fly?

lucasl84, 2024-08-27:
Hi everyone,

This is just a question related to the keepout filter mask functionality. I'm using the keepout filter mask and everything is working as expected. The thing I would like to do now is on the fly change the mask for a completely different one. I've manually tested this from the console by:

1 - Updating the PGM filter mask file.
2 - Pull the current YAML configuration from the node with `ros2 param dump /filter_mask_server > filter_mask_server.yaml`.
3 - Attempting to reload the filter mask by calling: `ros2 param load /filter_mask_server ./filter_mask_server.yaml`.

The console output for the last command (the `ros2 param load`) is:
```
Set parameter /bond_disable_heartbeat_timeout successful
Set parameter bond_heartbeat_period successful
Set parameter frame_id successful
Set parameter start_type_description_service failed: parameter 'start_type_description_service' cannot be set because it is read-only
Set parameter topic_name successful
Set parameter use_sim_time successful
Set parameter yaml_filename successful
```

However if I take a look at RViz the old mask is still in place. So I wanted to know if there's a way to achieve this. Preferably programmatically as I'm trying to integrate this feature with Python (just tried the command line to see if possible at all).

Thanks everyone for their time,
Lucas.

SteveMacenski, 2024-08-27:
You can simply publish the new keepout filter mask and it should be immediately used in place of the old one https://github.com/ros-navigation/navigation2/blob/main/nav2_costmap_2d/plugins/costmap_filters/keepout_filter.cpp#L123.

I don't think you need to do anything more complicated. You can load a new map in map server from a service call https://github.com/ros-navigation/navigation2/blob/main/nav2_map_server/src/map_server/map_server.cpp#L121-L123 without reinitializing the node.

lucasl84, 2024-08-28:
Hi Steve,

Once again thanks for clarifying this point!

Lucas.

closed 2024-08-28 by lucasl84