ros-navigation/navigation2 #4655: NavfnPlanner randomly fails for a clearly reachable goal saying "Failed to create a plan from potential when a legal .... found"

saad-waqar-robotics, 2024-08-31:
## Bug report

**Required Info:**

- Operating System:
  UBUNTU 22
- ROS2 Version:
  Humble

- Version or commit hash:
  dpkg-query --show "ros-$ROS_DISTRO-navigation2"
  ros-humble-navigation2	1.1.16-1jammy.20240824.191514

- DDS implementation:
 Cyclone

#### Steps to reproduce issue

```
I am using a diff_drive robot with ROS2 Humble, NAV2 & slam_toolbox. 
I am currently simulating the robot in Gazebo classic, House map. 
I have started using RegulatedPurePursuit with NavfnPlanner.
My params file is:

amcl:
  ros__parameters:
    use_sim_time: True
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan

bt_navigator:
  ros__parameters:
    use_sim_time: True
    global_frame: map
    robot_base_frame: base_link
[... 386 lines cut ...]
Global planning fails randomly on the way while re-planning. At the beginning, it perfectly finds a plan for a clearly reachable goal, starts driving on it, but randomly fails during re-planning saying:
**[planner_server]: Failed to create a plan from potential when a legal potential was found. This shouldn't happen.**
[planner_server]: GridBased: failed to create plan with tolerance 0.50.
 [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-6.30, -2.68)

#### Additional information
I am using a diff_drive robot with ROS2 Humble, NAV2 & slam_toolbox. Everything seems to be in place.

I am currently simulating the robot in Gazebo classic, House map. 

I have started using RegulatedPurePursuit with NavfnPlanner. Usually it works nicely, but randomly I am getting this error from planner:

**[planner_server]: Failed to create a plan from potential when a legal potential was found. This shouldn't happen.**
[planner_server]: GridBased: failed to create plan with tolerance 0.50.
 [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-6.30, -2.68)

This issue comes purely RANDOMLY. It usually does not occur when I send a new goal request. Usually, it takes a goal request, plans the path and starts moving. While moving on its way to a clearly reachable goal, it randomly starts giving this error while re-planning the global path. Due to this error, it falls into recovery mode, sometimes it recovers, sometimes it gives a failure after multiple tries.
I have tried to look on google, I could not find anything helplful.

Please let me know if I am doing something wrong. :) :)

****
[image]

In this screenshot, I was travelling from position 3 to position 1, that is clearly reachable. It started on the path perfectly fine, but in the middle it starting giving this error and fell down to recovery mode.

saad-waqar-robotics, 2024-08-31:
Additional info:
I enabled debug mode, and then reproduced the issue on a long path:

[planner_server-8] [DEBUG] [1725088039.676608922] [planner_server.rclcpp_action]: Accepted goal 7f8e7e847727a3bebe82e1ea16f1576
[planner_server-8] [DEBUG] [1725088039.676846811] [planner_server]: [compute_path_to_pose] [ActionServer] Receiving a new goal
[planner_server-8] [DEBUG] [1725088039.676868748] [planner_server]: [compute_path_to_pose] [ActionServer] Executing goal asynchronously.
[planner_server-8] [DEBUG] [1725088039.676953543] [planner_server]: [compute_path_to_pose] [ActionServer] Executing the goal...
[planner_server-8] [DEBUG] [1725088039.676959452] [rcl]: Service server taking service request
[bt_navigator-10] [DEBUG] [1725088039.676618873] [rcl_action]: Taking action goal response
[bt_navigator-10] [DEBUG] [1725088039.676627021] [rcl]: Client taking service response
[bt_navigator-10] [DEBUG] [1725088039.676634695] [rcl]: Client take response succeeded: true
[bt_navigator-10] [DEBUG] [1725088039.676638454] [rcl_action]: Action goal response taken
[bt_navigator-10] [DEBUG] [1725088039.676648860] [rcl_action]: Sending action result request
[bt_navigator-10] [DEBUG] [1725088039.676652736] [rcl]: Client sending service request
[bt_navigator-10] [DEBUG] [1725088039.676673816] [rcl_action]: Action result request sent
[bt_navigator-10] [DEBUG] [1725088039.676942051] [rcl_action]: Taking action feedback
[bt_navigator-10] [DEBUG] [1725088039.676970917] [rcl]: Subscription taking message
[bt_navigator-10] [DEBUG] [1725088039.676992406] [rcl]: Subscription take succeeded: true
[bt_navigator-10] [DEBUG] [1725088039.677002877] [rcl_action]: Action feedback taken
[bt_navigator-10] [DEBUG] [1725088039.677015764] [bt_navigator.rclcpp_action]: Received feedback for unknown goal. Ignoring...
[planner_server-8] [DEBUG] [1725088039.677073986] [rcl]: Service take request succeeded: true
[planner_server-8] [DEBUG] [1725088039.677099029] [planner_server]: Attempting to a find path from (2.20, 0.38) to (-6.45, -2.10).
[planner_server-8] [DEBUG] [1725088039.677158458] [planner_server]: Making plan from (2.20,0.38) to (-6.45,-2.10)
[planner_server-8] [DEBUG] [1725088039.677204235] [rclcpp]: [NavFn] Array is 299 x 212
[planner_server-8] 
[planner_server-8] [DEBUG] [1725088039.677447142] [rclcpp]: [NavFn] Setting start to 21,65
[planner_server-8] 
[planner_server-8] [DEBUG] [1725088039.677504076] [rclcpp]: [NavFn] Setting goal to 194,115
[planner_server-8] 
[lifecycle_manager-13] [DEBUG] [1725088039.677942484] [rcl]: Calling timer
[lifecycle_manager-13] [DEBUG] [1725088039.678152287] [rcl]: Subscription taking message
[lifecycle_manager-13] [DEBUG] [1725088039.678171900] [rcl]: Subscription take succeeded: true
[lifecycle_manager-13] [DEBUG] [1725088039.678186451] [rcl]: Subscription taking message
[lifecycle_manager-13] [DEBUG] [1725088039.678192139] [rcl]: Subscription take succeeded: true
[lifecycle_manager-13] [DEBUG] [1725088039.678199954] [rcl]: Subscription taking message
[lifecycle_manager-13] [DEBUG] [1725088039.678205543] [rcl]: Subscription take succeeded: true
[lifecycle_manager-13] [DEBUG] [1725088039.678213775] [rcl]: Subscription taking message
[lifecycle_manager-13] [DEBUG] [1725088039.678219796] [rcl]: Subscription take succeeded: true
[lifecycle_manager-13] [DEBUG] [1725088039.678227202] [rcl]: Subscription taking message
[lifecycle_manager-13] [DEBUG] [1725088039.678232494] [rcl]: Subscription take succeeded: true
[lifecycle_manager-13] [DEBUG] [1725088039.678240141] [rcl]: Subscription taking message
[lifecycle_manager-13] [DEBUG] [1725088039.678245729] [rcl]: Subscription take succeeded: true
[lifecycle_manager-13] [DEBUG] [1725088039.678252797] [rcl]: Subscription taking message
[lifecycle_manager-13] [DEBUG] [1725088039.678258418] [rcl]: Subscription take succeeded: true
[velocity_smoother-12] [DEBUG] [1725088039.678229050] [rcl]: Subscription taking message
[velocity_smoother-12] [DEBUG] [1725088039.678282073] [rcl]: Subscription take succeeded: true
[bt_navigator-10] [DEBUG] [1725088039.678222613] [rcl]: Subscription taking message
[bt_navigator-10] [DEBUG] [1725088039.678283295] [rcl]: Subscription take succeeded: true
[behavior_server-9] [DEBUG] [1725088039.678230539] [rcl]: Subscription taking message
[behavior_server-9] [DEBUG] [1725088039.678286470] [rcl]: Subscription take succeeded: true
[smoother_server-7] [DEBUG] [1725088039.678239359] [rcl]: Subscription taking message
[smoother_server-7] [DEBUG] [1725088039.678281619] [rcl]: Subscription take succeeded: true
[smoother_server-7] [DEBUG] [1725088039.678322052] [rcl]: Initializing timer with period: 4000000000ns
[smoother_server-7] [DEBUG] [1725088039.678373467] [rcl]: Timer canceled
[smoother_server-7] [DEBUG] [1725088039.678382177] [rcl]: Timer canceled
[waypoint_follower-11] [DEBUG] [1725088039.678383800] [rcl]: Subscription taking message
[waypoint_follower-11] [DEBUG] [1725088039.678456380] [rcl]: Subscription take succeeded: true
[planner_server-8] [DEBUG] [1725088039.678409545] [rcl]: Subscription taking message
[planner_server-8] [DEBUG] [1725088039.678451321] [rcl]: Subscription take succeeded: true
[controller_server-6] [DEBUG] [1725088039.678410547] [rcl]: Subscription taking message
[controller_server-6] [DEBUG] [1725088039.678464369] [rcl]: Subscription take succeeded: true
[planner_server-8] [DEBUG] [1725088039.679474291] [rclcpp]: [NavFn] Used 467 cycles, 20807 cells visited (35%), priority buf max 800
[planner_server-8] 
[... cut ...]
[planner_server-8] [ERROR] [1725088039.680184300] [planner_server]: Failed to create a plan from potential when a legal potential was found. This shouldn't happen.
[planner_server-8] [WARN] [1725088039.680211369] [planner_server]: GridBased: failed to create plan with tolerance 0.50.
[planner_server-8] [WARN] [1725088039.680226135] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-6.45, -2.10)
[planner_server-8] [WARN] [1725088039.680242217] [planner_server]: [compute_path_to_pose] [ActionServer] Aborting handle.

SteveMacenski, 2024-08-31:
Please provide a reproducable example. You might find https://github.com/botsandus/planner_playground useful to create that example if you fork + add your map and start/goal locations

saad-waqar-robotics, 2024-09-01:
Thank you for your kind response @SteveMacenski 

Here is the robot footprint, seems valid.
[image]

The cost-map looks like this:
[image]

Map png file:
[image]

map yaml file:
image: bug_wala_map.png
mode: trinary
resolution: 0.05
origin: [-7.47, -5.27, 0]
negate: 0
occupied_thresh: 0.65
free_thresh: 0.25

Issue was reproduced multiple-times for the goal:
[image]

I am also trying to reproduce everything on the playground ( https://github.com/botsandus/planner_playground ), but due to some reasons, it does not load my custom map (without showing any warning or error, although it reads the map parameters fine). I am working on that.

SteveMacenski, 2024-09-04:
I will need a way to reproduce in order to help. There should be no difference in replanning with NavFn, but you could adjust the planner playground to plans 2x in a row if the replan is what causes the issue to be reproducable (should I be wrong about that!).

saad-waqar-robotics, 2024-09-08:
I am creating a package so you will be able to reproduce the issue. I have tried changing the inflation radius etc, but still the problem persists. 
I have noticed that this issue usually occurs when the robot is passing through a narrow area (still enough distance to go). It creates plan correctly, but when enters the narrow areas it throws the error and retries. Sometimes it moves through after some tries, sometimes it declares failure. I will try to get a package and steps to reproduce in a while.

Thanks for the help.

SteveMacenski, 2024-09-09:
Still need something preproducible. 

I wonder if this relates to the fact that you have a very low robot radius since you have the pivot point so close to the front of the robot. As you can see, the blue inscribed band around obstacles is incredibly small - and very unusual. It could be that the neighbor expansion in some situations isn't hitting that cell so it by passes it and then causes issues when doing the gradient descent its creating a problem.

What happens if you add `footprint_padding` on the radius so that the inscribed 253 cost has a non-single-pixel wide band? i.e. 5-10cm?

saad-waqar-robotics, 2024-09-15:
Thanks @SteveMacenski. I'll definitely check that and report the results. The pivot point is actually on the rear side of the robot (here in the image we are looking at the rear side, its a rear side differential drive, forward 2 wheels are castor wheels. Here in the image, it was going straight on the path, but when the failure occurred, then it went into recovery behavior and spun towards right side.)
Recently, I increased the inflation radius and tinkered with cost-scaling factor, now I have not seen this issue in a while. I will keep on testing, and I will also check out footprint_padding as well. 
Note: My robot radius is not small, I am using robot_footprint through a polygon (as you can see in the image in green). But yes, I will try adding padding and will report the results.

THanks!

saad-waqar-robotics, 2024-09-15:
I checked with footprint padding of 7.5cm. The blue inscribed band definitely increased, but I still got the error around the corners. 
[image]

I tried with robot radius (instead of robot_footprint polygon), then it inflated the inscribed blue band appropriately). 
[image]

(I have my baselink at the center of rotation, so the circle is also on offset but the obstacles are inflated correctly. Although the error is very less frequent which this, but still I saw it once. May be due to this invalid circle position, I will confirm it by fixing the center of the circle.)

My question is, isnt costmap supposed to inflate the inscribed region according to robot footprint as well (just like it does with robot_radius)?
If not, is addition of padding the only way to increase the inscribed region?
May be its a silly question, but I could not get it from the docs.

Thanks a lot.

saad-waqar-robotics, 2024-09-15:
After setting the circle to the center (by changing robot_base_frame for costmaps to chassis_link {center link in my robot}), the error is very rare. But I still got it 3 times, even in open spaces. 
[image]

Here is my latest .yaml file:

amcl:
  ros__parameters:
    use_sim_time: True
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan

bt_navigator:
  ros__parameters:
    use_sim_time: True
    global_frame: map
    robot_base_frame: base_link
    odom_topic: /odom
    bt_loop_duration: 10
    default_server_timeout: 20
    default_nav_to_pose_bt_xml: $(find-pkg-share carrier_bot_sim)/config/navigate_w_replanning_and_recovery.xml
    navigators: ['navigate_to_pose']
    navigate_to_pose:
      plugin: "nav2_bt_navigator::NavigateToPoseNavigator" # In Iron and older versions, "/" was used instead of "::"
    # 'default_nav_through_poses_bt_xml' and 'default_nav_to_pose_bt_xml' are use defaults:
    # nav2_bt_navigator/navigate_to_pose_w_replanning_and_recovery.xml
    # nav2_bt_navigator/navigate_through_poses_w_replanning_and_recovery.xml
    # They can be set here or via a RewrittenYaml remap from a parent launch file to Nav2.
    plugin_lib_names:
      - nav2_compute_path_to_pose_action_bt_node
      - nav2_compute_path_through_poses_action_bt_node
      - nav2_smooth_path_action_bt_node
      - nav2_follow_path_action_bt_node
      - nav2_spin_action_bt_node
      - nav2_wait_action_bt_node
      - nav2_assisted_teleop_action_bt_node
[... 337 lines cut ...]
waypoint_follower:
  ros__parameters:
    use_sim_time: True
    loop_rate: 20
    stop_on_failure: false
    waypoint_task_executor_plugin: "wait_at_waypoint"
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
      enabled: True
      waypoint_pause_duration: 200

velocity_smoother:
  ros__parameters:
    use_sim_time: True
    smoothing_frequency: 20.0
    scale_velocities: False
    feedback: "OPEN_LOOP"
    max_velocity: [0.5, 0.0, 1.0]
    min_velocity: [-0.5, 0.0, -1.0]
    max_accel: [2.5, 0.0, 3.2]
    max_decel: [-2.5, 0.0, -3.2]
    odom_topic: "odom"
    odom_duration: 0.1
    deadband_velocity: [0.0, 0.0, 0.0]
    velocity_timeout: 1.0

saad-waqar-robotics, 2024-09-15:
`amcl:
  ros__parameters:
    use_sim_time: True
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan

bt_navigator:
  ros__parameters:
    use_sim_time: True
    global_frame: map
    robot_base_frame: base_link
    odom_topic: /odom
    bt_loop_duration: 10
    default_server_timeout: 20
    default_nav_to_pose_bt_xml: $(find-pkg-share carrier_bot_sim)/config/navigate_w_replanning_and_recovery.xml
    navigators: ['navigate_to_pose']
    navigate_to_pose:
      plugin: "nav2_bt_navigator::NavigateToPoseNavigator" # In Iron and older versions, "/" was used instead of "::"
    # 'default_nav_through_poses_bt_xml' and 'default_nav_to_pose_bt_xml' are use defaults:
    # nav2_bt_navigator/navigate_to_pose_w_replanning_and_recovery.xml
    # nav2_bt_navigator/navigate_through_poses_w_replanning_and_recovery.xml
    # They can be set here or via a RewrittenYaml remap from a parent launch file to Nav2.
    plugin_lib_names:
      - nav2_compute_path_to_pose_action_bt_node
      - nav2_compute_path_through_poses_action_bt_node
      - nav2_smooth_path_action_bt_node
      - nav2_follow_path_action_bt_node
      - nav2_spin_action_bt_node
      - nav2_wait_action_bt_node
      - nav2_assisted_teleop_action_bt_node
      - nav2_back_up_action_bt_node
      - nav2_drive_on_heading_bt_node
      - nav2_clear_costmap_service_bt_node
      - nav2_is_stuck_condition_bt_node
      - nav2_goal_reached_condition_bt_node
[... 334 lines cut ...]
    use_sim_time: True
    loop_rate: 20
    stop_on_failure: false
    waypoint_task_executor_plugin: "wait_at_waypoint"
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
      enabled: True
      waypoint_pause_duration: 200

velocity_smoother:
  ros__parameters:
    use_sim_time: True
    smoothing_frequency: 20.0
    scale_velocities: False
    feedback: "OPEN_LOOP"
    max_velocity: [0.5, 0.0, 1.0]
    min_velocity: [-0.5, 0.0, -1.0]
    max_accel: [2.5, 0.0, 3.2]
    max_decel: [-2.5, 0.0, -3.2]
    odom_topic: "odom"
    odom_duration: 0.1
    deadband_velocity: [0.0, 0.0, 0.0]
    velocity_timeout: 1.0

`

SteveMacenski, 2024-09-15:
> I tried with robot radius (instead of robot_footprint polygon), then it inflated the inscribed blue band appropriately).

Did that not work? 

> isnt costmap supposed to inflate the inscribed region according to robot footprint as well (just like it does with robot_radius)?

It uses the footprint if given, else the radius. 

---

This is very strange. I've never seen this & I've never received a report about this in all the years of working with ROS 1 Navigation and now Nav2. Please provide a reproducable example and I can toy around with it!

saad-waqar-robotics, 2024-09-16:
> Did that not work?

I still get errors (although very less frequently). 

Thank you for your assistance. I am trying to create a reproducable example, and I'll share that ASAP.

patham9, 2024-10-30:
The easiest way to reproduce is to install Ubuntu 24 from scratch and to install ROS Jazzy with Nav2 from the official packages there, then to run turtlebot4 simulation and send Nav2 goals. I think it is safe to say that Nav2 is not in a good shape, very unreliable compared to what ROS1 featured.

SteveMacenski, 2024-11-04:
@patham9 I'm not sure what makes you say this - but if you had specific advice for @saad-waqar-robotics or information on this ticket to share, we'd love to hear it :-)

patham9, 2025-01-15:
@saad-waqar-robotics: Which error do you get still?  Is it "[planner_server]: Failed to create a plan from potential when a legal potential was found. This shouldn't happen."?
I will try the costmap parameters you have posted, inflation radius seems to be the key factor.

The-Bloop, 2025-05-06:
@saad-waqar-robotics, does this issue specifically occur at turning locations?