ros-navigation/navigation2 #4656: [MPPI

RobinHeitz, 2024-08-31:
## Bug report

**Required Info:**

- Operating System:
  - 
- ROS2 Version:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2024-09-03:
This is a pretty niche problem that is going to likely necessitate that you create a few of your own custom plugin critics to score the MPPI samples w.r.t. only the XY to create the kinds of trajectories you seek. I can't say with any certainty for this situation what you precisely need to do or settings to configure. This is something I myself would need to think through, experiment with, see how it goes, and iterate from there looking into the various modules to make sure I covered my bases. I don't think in abstract I can give you a roadmap for this specialized situation without myself simply _doing it_ and having a robot to experiment with. 

Sorry :frowning:

SteveMacenski, 2024-12-10:
Closing due to lack of actionability at this moment. If a user would like to implement and contribute this, we would be happy to merge it, however this is not something maintainers plan to do given the niche nature

closed 2024-12-10 by SteveMacenski