ros-navigation/navigation2 #4659: follow_path action keeps on running after having canceled the goal

milidam, 2024-09-03:
## Bug report

**Required Info:**

- Operating System:
  - Custom Linux distribution
- ROS2 Version:
  - Iron 2024-07-12
- Version or commit hash:
  - navigation2 1.2.9
- DDS implementation:
  - CycloneDDS

#### Steps to reproduce issue
In some cases, the `follow_path` action keeps on running after having canceled a `navigate_to_pose` goal.

#### Expected behavior

After canceling the `navigate_to_pose` goal, we would expect the `controller_server` to properly interrupt its `follow_path` action.
```
[controller_server-8] [INFO]: Goal was canceled. Stopping the robot.
[controller_server-8] [WARN]: [follow_path] [ActionServer] Client requested to cancel the goal. Cancelling.
```

#### Actual behavior

The `bt_navigator` gets a `Failed to get result for compute_path_to_pose in node halt!` or a `Failed to get result for follow_path in node halt!` error, and the `controller_server` keeps on trying to follow the last received path.

```
[bt_navigator-12] [INFO]: Starting path following
[controller_server-8] [INFO]: Received a goal, begin computing control effort.
[planner_server-9] [WARN]: Map update loop missed its desired rate of 5.0000Hz. Current loop rate is 0.9686Hz
[bt_navigator-12] [WARN]: Behavior tree engine cycle is slower than expected: 114 ms (timeout is 50 ms)
[bt_navigator-12] [WARN]: Time required to tick the tree is 114 ms
[bt_navigator-12] [WARN]: Timed out while waiting for action server to acknowledge goal request for follow_path
[bt_navigator-12] [INFO]: Pausing path following due to slow path computation (>500ms)
[planner_server-9] [INFO]: Attempting to find a path from (-12.62, 3.38) to (-54.58, 11.29).
[planner_server-9] [INFO]: SmacPlannerHybrid: Start pose in world: (-12.624103, 3.381263), in costmap (452, 103)
[planner_server-9] [INFO]: SmacPlannerHybrid: Goal pose in world: (-54.581326, 11.286437), in costmap (32, 182)
[app-21] [INFO]: Request to cancel goals on /navigate_to_pose was accepted by server, waiting for result
[controller_server-8] [WARN]: Control loop missed its desired rate of 10.0000 Hz. Current loop rate is 9.2498 Hz.
[bt_navigator-12] [ERROR]: Failed to get result for compute_path_to_pose in node halt!
[bt_navigator-12] [INFO]: Goal canceled
[bt_navigator-12] [WARN]: [navigate_to_pose] [ActionServer] Client requested to cancel the goal. Cancelling.
[controller_server-8] [WARN]: Control loop missed its desired rate of 10.0000 Hz. Current loop rate is 9.5395 Hz.
[controller_server-8] [WARN]: Control loop missed its desired rate of 10.0000 Hz. Current loop rate is 10.9403 Hz.
```

#### Additional information

In the case reported above, before that new instance of the `Starting path following`, in a previous attempt in the same instance of the `navigate_to_pose` behavior tree, the `bt_navigator` already encountered the following error
```
[bt_navigator-12] [ERROR]: Failed to get result for follow_path in node halt!
```

what could indicate that the `bt_navigator` might somehow be desynchronized with the `controller_server`?

Also, the map is quite big, and the `planner_server` takes some time to compute a path. The goal canceling occurred while the `compute_path_to_pose` action was still running.

SteveMacenski, 2024-09-03:
That occurs here [1], so the server timeout doesn't cancel the action in time. If you were to set this to say - 1 second - does that remove your issue? Also, what so you have your `server_timeout` set to - that could be unrealistically short.  

It looks like this condition was added in [2] and may need to (generally speaking) have a longer timeout than the server timeout proportional to steady-state BT node ticking. If we're halting / exiting out a BT, we may find it better to have the rate exceeded a bit in exchange for a more realistic chance of those actions being canceled successfully. 

> Also, the map is quite big, and the planner_server takes some time to compute a path. The goal canceling occurred while the compute_path_to_pose action was still running.

This is definitely why. If you have a server timeout of 10ms and you've just started a planning request that is 100ms in length, then it won't always cancel in time. We added a cancel checker [3] in the Planner API so that we query in the planning loop if a cancel is requested to be more responsive, but that is not backported to Iron and you'll need to upgrade to Jazzy to receive that. That actually by itself might completely fix your issue. 

[1] https://github.com/ros-navigation/navigation2/blob/34d41ed32745e3978a0c0e30e9795c32ea184e8a/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L319

[2] https://github.com/ros-navigation/navigation2/commit/bfb4506c9b6beaebfb8532b221b517a18e40ac3d

[3] https://github.com/ros-navigation/navigation2/blob/main/nav2_navfn_planner/src/navfn_planner.cpp#L137

SteveMacenski, 2024-09-09:
@milidam have you had a chance to look into these items? I'd love to get feedback on the precise nature of it to write up a solution for you to test

milidam, 2024-09-10:
Hi @SteveMacenski,
Our `server_timeout` is currently set to 100ms.
I haven't had time yet to try to reproduce the issue with an increased timeout.

Our planner's `max_planning_time` is 30.0s.
Does that also increase the risk of not being able to actually cancel it?

Note that migration to Jazzy is planned, but not yet started.

SteveMacenski, 2024-12-11:
@milidam any updates?

mamariomiamo, 2024-12-20:
From the terminal output shared by the OP, **follow_path** BT action node never cancel the current goal. This may be because of the following sequence of events:
1. We see the ```Timed out while waiting for action server to acknowledge goal request for follow_path``` in the console logging, which means server already timed out, ```future_goal_handle_``` gets ```reset()``` https://github.com/ros-navigation/navigation2/blob/34d41ed32745e3978a0c0e30e9795c32ea184e8a/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L405-L413
and ```goal_handle_``` was not assigned value
https://github.com/ros-navigation/navigation2/blob/34d41ed32745e3978a0c0e30e9795c32ea184e8a/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L425-L432

2. When ```halt()``` function is called, ```should_cancel_goal()``` returns ```false``` since ```goal_handle_``` is still a nullptr and thus the goal was never canceled.
https://github.com/ros-navigation/navigation2/blob/34d41ed32745e3978a0c0e30e9795c32ea184e8a/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L340-L350

Should we add a call to cancel goal before we return failure here? Since we never get a valid ```goal_handle_```, we can use ```async_cancel_all_goals()```
https://github.com/ros-navigation/navigation2/blob/34d41ed32745e3978a0c0e30e9795c32ea184e8a/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L209-L228

If we do not properly cancel the follow_path action here and if we have any other recovery mechanism for example DriveOnHeading, we would have two nodes publishing to the ```cmd_vel``` at the same time.

SteveMacenski, 2024-12-20:
> We see the Timed out while waiting for action server to acknowledge goal request for follow_path in the console logging, which means server already timed out, future_goal_handle_ gets reset() 

That's not the only `false` return condition from `is_future_goal_handle_complete()`, there's also the following which is caused when `result == rclcpp::FutureReturnCode::TIMEOUT` -- but yes, another form of the timeout. However `future_goal_handle_` is not reset in this case. 

https://github.com/ros-navigation/navigation2/blob/34d41ed32745e3978a0c0e30e9795c32ea184e8a/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L434

>  and goal_handle_ was not assigned value

This is true either way, yes!

> When halt() function is called, should_cancel_goal() returns false since goal_handle_ is still a nullptr and thus the goal was never canceled.

True, but we use the `goal_handle_` to cancel things, so if `nullptr`, we couldn't use it away. We'd either have to set the goal handle (which I don't think is possible) or switch the API to use `[async_cancel_all_goals](https://docs.ros2.org/latest/api/rclcpp_action/classrclcpp__action_1_1Client.html#a51cfb2f4267c89bd1d08e15df249c7d6) or [async_cancel_goals_before](https://docs.ros2.org/latest/api/rclcpp_action/classrclcpp__action_1_1Client.html#ad0516a6a80c4033e4e9c0741aa5ed517). Neither of which are very precise, but technically possible. 

> Should we add a call to cancel goal before we return failure here? Since we never get a valid goal_handle_, we can use async_cancel_all_goals()

Definitely if we do something like that using the other 2 cancel goal APIs would be discouraged because it is a sludge hammer when we want something more targeted. 

In this situation is `future_goal_handle_.get() != nullptr` in order to retrieve a goal handle to use for cancellation? If so, that seems technically possible. However isn't this block focused on the goal sending, not the actual goal execution? This is a timeout for getting an ACK that the goal was receieved. I don't think canceling is appropriate if the goal was never received because there's no goal to handle. From the action server's side, we'd have to see how a cancel to a goal that doesn't exist is handled (and if that's cleanly done that we could possibly send excessive requests and not cause a problem). 

https://github.com/ros-navigation/navigation2/blob/34d41ed32745e3978a0c0e30e9795c32ea184e8a/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L209-L228

adivardi, 2025-01-22:
> Do you know when you would be able to? If you increased that to 500ms or 1s and it went away for Follow Path, that could be something we could replace that spin with since that cancel should only happen when halting an entire tree on task exit (which keeping the 10ms tick rate is unnecessary)

I have a similar issue. I use the `navigate_w_basic_complete_coverage` behavior tree. When canceling `navigate_complete_coverage`, I often get `[bt_navigator_navigate_complete_coverage_rclcpp_node]: Failed to get result for follow_path in node halt!`

Increasing `default_server_timeout` from 30 to even ~50ms solves this issue.

Example of full output (this repeats for a while):
```
[ERROR] [bt_navigator_navigate_complete_coverage_rclcpp_node]: Failed to get result for follow_path in node halt!
[WARN]  [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 14.2857 Hz.
[INFO]  [controller_server]: Cancellation was successful. Stopping the robot.
[INFO]  [controller_server]: [follow_path] [ActionServer] Client requested to cancel the goal. Cancelling.
[INFO]  [controller_server]: Optimizer reset
[INFO]  [controller_server]: Received a goal, begin computing control effort.
[INFO]  [controller_server]: Cancellation was successful. Stopping the robot.
[INFO]  [controller_server]: [follow_path] [ActionServer] Client requested to cancel the goal. Cancelling.
[INFO]  [controller_server]: Optimizer reset
[INFO]  [controller_server]: Received a goal, begin computing control effort.
```

---------------

I also have a similar issue with `IsPathValid` timing out, but I think this is a similar but not exactly related issue of that service being very slow.

adivardi, 2025-01-23:
I am running on Jazzy on Gazebo simulation, on a laptop with i7-12700H.

Currently using the higher timeout seems to be working, but it does mean many nodes may run slower than the BT tick frequency. I am not sure how big of an issue it is, but it does not sound correct.

I did read the thread, my intention was to support the idea of either using spin for the action cancellation, or better yet supporting a configurable timeout for cancellation different than the "normal" one.

mamariomiamo, 2025-01-24:
> However isn't this block focused on the goal sending, not the actual goal execution? This is a timeout for getting an ACK that the goal was receieved. I don't think canceling is appropriate if the goal was never received because there's no goal to handle.

So the tricky thing is that, the goal was actually received by the server and being processed but the client "thinks" it was not received because the ACK was not received properly.

In our case, the goal was actually received at the server side and the server starts to execute the action. But somehow when the server sends back the goal acknowledgement the client does not receive the ACK which result in the BT node to return FAILURE. If the BT has a recovery branch which also publishes to cmd_vel then we would have two servers publishing to the cmd_vel topic at the same time and thus cause undesirable behavior.

SteveMacenski, 2025-01-27:
> It does mean many nodes may run slower than the BT tick frequency. I am not sure how big of an issue it is

It is not, if the server timeout is greater than the tick rate, we spin the future for up-to the tick rate and return a `RUNNING` state. The next tick, we keep spinning -- and so forth until the total timeout is met or the server returns a result. 

>  better yet supporting a configurable timeout for cancellation different than the "normal" one.

I think these problems have been fully resolved in modern distributions since the Planner Server now has the cancellation check that is run in the planning algorithms. That however is a API/ABI breakage for folks' custom planner plugin algorithms so this cannot be backported to Humble, but is resolved in Jazzy and newer. If you have a way that we can make sure the planner server responds more immediately without breaking ABI/API in Humble for Humble users, I'm all ears! The problem here is not networking / or server timeout related, its that the planners did not used to be aware of cancel requests in their planning loops, so it would only receive the cancel request to action against at the start of the planning request only. That has since been resolved in the Planner Server and all the planners now respect those cancelation requests very reactively. 

I don't think increasing this timeout is really all that helpful in this case -- **but** I suppose for **Humble only** we could set a cancel timeout field that is users can set proportionately to their planning times (i.e. 100, 200, 500ms, etc) so that they can get responses appropriate. Would that be something you're open to contributing?

https://github.com/ros-navigation/navigation2/blob/main/nav2_planner/src/planner_server.cpp#L538-L543

adivardi, 2025-01-28:
> It is not, if the server timeout is greater than the tick rate, we spin the future for up-to the tick rate and return a `RUNNING` state. The next tick, we keep spinning -- and so forth until the total timeout is met or the server returns a result.

You are right, I have missed that at the time I wrote my last message. In this case I agree with you that increasing the timeout is a sufficient solution.

I am not sure if this is an issue that comes up a lot. If it is you could consider just a different blackboard value for cancellations `server_cancel_action_timeout`, or maybe just expand the error message from the BT.

Thanks a lot for the support 😃