ros-navigation/navigation2 #4671: Bug: Missing `map_load_time` and `update_time` in costmap meta data

gennartan, 2024-09-11:
## Bug report

- Operating System: Ubuntu
- ROS2 Version: All
- Version or commit hash: 7eb47d84012dd9e25988ce5c9c269280e6ee3dd1
- DDS implementation: N/A

#### Expected behavior

The `MapMetaData.msg` contains two time fields: `update_time` and `map_load_time`. I am expecting to have those fields updated when a map update or map load occurs.

#### Actual behavior

There is no mechanism to record the ROS time at which a map was updated or loaded. So those fields are set to the current time as shown in service callback of `get_costmap` [1]. 

#### Additional information

I have already checked a bit the code to investigate how to properly add this feature properly. As I understand, the Costmap class doesn't contain any rclcpp structures, so it should stay like this to keep it abastracted from ROS. So I think that those values should be updated from ROS callbacks when the map is updated (probably somewhere here [3]). But it is not as easy, since not every call to UpdateMap is going to actually update the map. For example, a non rolling costmap that contains a static layer that is set once before the navigation will not require those fields to be updated afterwards. Up to now, I haven't seen any feedback from map updates that could be used to know is a costmap was actually updated or not following a call to updateMap.

[1] https://github.com/ros-navigation/navigation2/blob/7eb47d84012dd9e25988ce5c9c269280e6ee3dd1/nav2_costmap_2d/src/
costmap_2d_publisher.cpp#L270
[2] https://github.com/ros-navigation/navigation2/blob/main/nav2_msgs/msg/CostmapMetaData.msg
[3] https://github.com/ros-navigation/navigation2/blob/7eb47d84012dd9e25988ce5c9c269280e6ee3dd1/nav2_costmap_2d/src/costmap_2d_ros.cpp#L573

gennartan, 2024-09-13:
Ok, maybe I misunderstood the use of map_load_time and map_update_time. Let's focus on `map_load_time` for now.

To me, the `map_load_time` should represent the time the map was received by the costmap server. So, if it was loaded from a map server through an Occupancy grid, it should be the time the occupancy grid was received. As you sait, this part is not missing from the code. However, when pulishing the costmap, it is publishing the current time instead of the actual costmap load time.

----
To give a bit more context to my issue, I am currently developing a robot that sequentially request the map server to load a static map through the '/map' topic (OccupancyGrid), then start a navigation. When I do it, the navigation fails because the costmap is not set yet when starting the navigation. Therefore, I must add a delay to account for the costmap load time. With a simple delay time.sleep(), it works as expected, but it lacks of robustness.

What I am therefore trying to do, is to poll the costmap server and check the `map_load_time` to be sure that the current costmap is the expected one (at least that it is greater than the requested map server load time).

SteveMacenski, 2024-09-13:
> To me, the map_load_time should represent the time the map was received by the costmap server. 

This is part of the cost-map's message, so it is metadata about the cost-**map**. It does not discuss the incoming static map from slam or the map server since it is not required to even use a static layer nor a particular format of 'map' to navigate (or any at all with an external positioning system like GPS). 

I think the biggest thing here is a misunderstanding - the cost-maps are also maps that are published for the same reasons one might want a map. The costmap is also published on an `OccupancyGrid` type which contains that field (as well as an internal Nav2 type called a `Costmap` which is used for non-aliasing the values for other processes to share a compression-free costmap model, but is mimicking the map type with different data meanings).

If you want the `map` time -> subscribe to the `map` topic to obtain it :smile: 

> Therefore, I must add a delay to account for the costmap load time. With a simple delay time.sleep(), it works as expected, but it lacks of robustness.

There is a 'get_costmap' service that should be returning some values when a new map is obtained. You could use the values from the costmap itself to decide if an update is completed. Or, we could do a number of things:
- Have an option for the static layer to take a service call, so you have an ACK that its completed
- Have the static layer publish out the last map's time or a service to obtain it 
- Get the costmap and check its values against your new map
- Probably a few other ways, but those are the most obvious to me at this moment. 

But I can understand what you're trying to do, however that is not what the `map_load_time` is for in the context of another non-static map instance. The same way not all `LaserScan` messages are the same in meaning or content, not all `OccupancyGrid`s are. Or if we have multiple static map layers, which does happen.

closed 2024-09-16 by gennartan