ros-navigation/navigation2 #4672: Potential endless loop if a costmap layer does not update.

tgreier, 2024-09-11:
https://github.com/ros-navigation/navigation2/blob/7eb47d84012dd9e25988ce5c9c269280e6ee3dd1/nav2_planner/src/planner_server.cpp#L282

PlannerServer::waitForCostmap() has no breakout timeout. Therefore, the planner server remains in loop if a costmap layer does not update, and there is no indication of why the planner server does not respond or updates its action request.

This loop should have a breakout condition with failure return and response.

SteveMacenski, 2024-09-11:
Fair point, @tgreier please review https://github.com/ros-navigation/navigation2/pull/4673 and let me know if this solves your concern

SteveMacenski, 2024-09-11:
I also added the controller server as well

closed 2024-09-12 by SteveMacenski