ros-navigation/navigation2 #4681: [Behavior Tree] Robot cancels control when in goal x_y tolerance

chanhhoang99, 2024-09-18:
## Bug report

**Required Info:**

- Operating System:
 - Ubuntu 24.04.1 LTS
- ROS2 Version:
 - Rolling
- Version or commit hash:
  - 771eca41bb7b6ad26856c434dd79e48a399c4601
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
1. Use navigate_to_pose_w_replanning_goal_patience_and_recovery.xml.
2. Navigate to pose.
```

```

#### Expected behavior
 Robot end up at desire goal without any wait and canceling action.
#### Actual behavior
 1. Robot is able to reach its x_y goal tolerance
 2. Robot is rotating to goal heading
 3. Wait node ticks.
 4. Cancel control ticks.
 5. Robot continue to rotate to goal heading after Wait node finishes
#### Additional information
From above actual behavior, I suspect this relates to PathLongerOnApproach.
I tried reverting commit 771eca41bb7b6ad26856c434dd79e48a399c4601, issue still persists
[Screencast from 2024-09-18 11-33-48.webm](https://github.com/user-attachments/assets/37a2f145-7140-4cf1-b965-71d5049e32a6)

Below is my configuration:
1. nav2_params.yaml:
```
amcl:
  ros__parameters:
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
[... 424 lines cut ...]
              </SequenceWithMemory>
            </RetryUntilSuccessful>
          </PathLongerOnApproach>
          <RecoveryNode number_of_retries="1" name="FollowPath">
            <FollowPath path="{path}" controller_id="{selected_controller}" error_code_id="{follow_path_error_code}"/>
            <ClearEntireCostmap name="ClearLocalCostmap-Context" service_name="local_costmap/clear_entirely_local_costmap"/>
          </RecoveryNode>
        </ReactiveSequence>
      </PipelineSequence>
      <ReactiveFallback name="RecoveryFallback">
        <GoalUpdated/>
        <RoundRobin name="RecoveryActions">
          <Sequence name="ClearingActions">
            <ClearEntireCostmap name="ClearLocalCostmap-Subtree" service_name="local_costmap/clear_entirely_local_costmap"/>
            <ClearEntireCostmap name="ClearGlobalCostmap-Subtree" service_name="global_costmap/clear_entirely_global_costmap"/>
          </Sequence>
          <Spin spin_dist="1.57"/>
          <Wait wait_duration="5.0"/>
          <BackUp backup_dist="0.30" backup_speed="0.15"/>
        </RoundRobin>
      </ReactiveFallback>
    </RecoveryNode>
  </BehaviorTree>
</root>
```

padhupradheep, 2024-09-18:
Thanks for reporting the issue, let me take a look at it.

padhupradheep, 2024-09-18:
It's very strange that I cannot reproduce your issue. I just tried with @neobotix robots and as well both the turtlebots that we have here. Seems to work fine, even with the parameters you have given. I'm certainly sure you would have rebuilt the package, but just for completeness, may I ask did you actually do it?

chanhhoang99, 2024-09-18:
Hi @padhupradheep ,
I observe this behavior both on my real robot and turtlebot4 simulation.
The occurrence is not always. But it's not hard to reproduce.
The issue could be reproduced when the goal heading is not straight with the path(In the video attached previously, I tried to give the goal with a different heading at the end.)

padhupradheep, 2024-09-18:
It could actually be an edge case.. 

> The issue could be reproduced when the goal heading is not in the same direction with the path(In the video attached previously, I tried to give the goal with a different heading at the end.)

I did the same but wasn't able to reproduce it. 

I also made a video (sorry for the low quality) with the turtlebot having a different heading at the end:

[nav2bug.webm](https://github.com/user-attachments/assets/de7c7335-dfbb-4aaa-917c-650816bed5a5)

Is there any other scenario in depot world, where you can show us the issue? would be great if I get some more details. Thanks!

chanhhoang99, 2024-09-18:
Okay I have just removed all build artifact (by $rm -r /build /install /log) and rebuild all the nav2 again with latest commit 771eca41bb7b6ad26856c434dd79e48a399c4601.
Issue is reproducible after a few trials. Below is my terminal logs:
```
[component_container_isolated-9] [INFO] [1726659620.186882210] [bt_navigator]: Begin navigating from current location (15.77, 4.73) to (18.01, -2.39)
[component_container_isolated-9] [INFO] [1726659620.207779929] [controller_server]: Received a goal, begin computing control effort.
[component_container_isolated-9] [INFO] [1726659621.307983945] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659622.307969344] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659623.307981063] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659624.407961604] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659625.407989273] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659626.407966414] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659627.508026270] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659628.507995824] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659629.507949135] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659630.507971404] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659631.607968496] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659632.607970118] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659633.607967764] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659634.707977162] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659635.708007138] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659636.707965783] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659637.707964830] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659638.807991231] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659639.807978545] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659640.807992836] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659641.807974343] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659642.907969000] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659643.907965740] [controller_server]: Passing new path to controller.
**[component_container_isolated-9] [INFO] [1726659644.887696426] [behavior_server]: Running wait
[component_container_isolated-9] [ERROR] [1726659644.907908415] [bt_navigator_navigate_to_pose_rclcpp_node]: Failed to get result for follow_path in node halt!
[component_container_isolated-9] [INFO] [1726659644.907951401] [controller_server]: Cancellation was successful. Stopping the robot.
[component_container_isolated-9] [WARN] [1726659644.908178831] [controller_server]: [follow_path] [ActionServer] Client requested to cancel the goal. Cancelling.
[component_container_isolated-9] [INFO] [1726659644.909248916] [controller_server]: Optimizer reset
[component_container_isolated-9] [INFO] [1726659655.087998729] [behavior_server]: wait completed successfully
[component_container_isolated-9] [INFO] [1726659655.099743507] [controller_server]: Received a goal, begin computing control effort.**
[component_container_isolated-9] [INFO] [1726659655.199937299] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659656.199938796] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659657.299927338] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659658.299933281] [controller_server]: Passing new path to controller.
[component_container_isolated-9] [INFO] [1726659658.405593911] [controller_server]: Reached the goal!
[component_container_isolated-9] [INFO] [1726659658.405956410] [controller_server]: Optimizer reset
[component_container_isolated-9] [INFO] [1726659658.429665709] [bt_navigator]: Goal succeeded
```
From above log, there is a gap before Goal succeeded and the gap is around 10s which is the Wait timeout duration that I set in the xml file.

I also capture the rosbag, the last goal in the bag is where the issue happens ( it takes quite many trials to reproduced :) )
Since it's quite a big bag file, would you mind getting it from https://drive.google.com/file/d/1E789QMxflwApboMJpDJiF_WzPlXVfiJM/view?usp=sharing

SteveMacenski, 2024-10-02:
Any thoughts here @padhupradheep ?

padhupradheep, 2024-10-05:
Unfortunately it is hard to reproduce the issue in our setup. It's good that the user has given a bag, I'd try to clear it soon.

chanhhoang99, 2024-10-23:
Hi @padhupradheep ,
Do you have any findings ? Do you need rosbag file on my real set up. It seems to happen quite often on my real robot.

padhupradheep, 2024-10-28:
Unfortunately not yet, I was also away for a conference. Please allow me some more time to get back to this.

padhupradheep, 2024-11-22:
@chanhhoang99 like mentioned in #4729 could you please try it without composition and let me know if you can reproduce the issue?

chanhhoang99, 2024-11-23:
@padhupradheep , Okay I will check it

chanhhoang99, 2024-11-23:
Hi @padhupradheep ,
Seems like it is a bit different when I launch with use_composition:=False.
Logs:
```
[bt_navigator-16] [INFO 2024-11-23 08:15:01.196] [bt_navigator]: Begin navigating from current location (0.56, 4.81) to (4.36, -6.07)
[controller_server-12] [INFO 2024-11-23 08:15:01.218] [controller_server]: Received a goal, begin computing control effort.
[controller_server-12] [INFO 2024-11-23 08:15:02.318] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:03.318] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:04.318] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:05.418] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:06.418] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:07.418] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:08.418] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:09.518] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:10.518] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:11.518] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:12.518] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:13.618] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:14.618] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:15.618] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:16.618] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:17.718] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:18.718] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:19.718] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:20.718] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:21.818] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:22.818] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:23.818] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:24.918] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:25.918] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:26.918] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:27.918] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:29.186] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:30.186] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:31.186] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:32.187] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:33.118] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:34.118] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:35.118] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:36.118] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:37.218] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:38.218] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:39.218] [controller_server]: Passing new path to controller.
[behavior_server-15] [INFO 2024-11-23 08:15:40.197] [behavior_server]: Running wait
[controller_server-12] [INFO 2024-11-23 08:15:40.218] [controller_server]: Cancellation was successful. Stopping the robot.
[controller_server-12] [WARN 2024-11-23 08:15:40.218] [controller_server]: [follow_path] [ActionServer] Client requested to cancel the goal. Cancelling.
[controller_server-12] [INFO 2024-11-23 08:15:40.219] [controller_server]: Optimizer reset
[controller_server-12] [INFO 2024-11-23 08:15:41.207] [controller_server]: Received a goal, begin computing control effort.
[controller_server-12] [INFO 2024-11-23 08:15:42.307] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:43.307] [controller_server]: Passing new path to controller.
[controller_server-12] [INFO 2024-11-23 08:15:43.912] [controller_server]: Reached the goal!
[controller_server-12] [INFO 2024-11-23 08:15:43.913] [controller_server]: Optimizer reset
[bt_navigator-16] [ERROR 2024-11-23 08:15:43.957] [bt_navigator_navigate_to_pose_rclcpp_node]: Failed to get result for wait in node halt!
[... cut ...]
You can see that  `[behavior_server-15] [INFO 2024-11-23 08:15:40.197] [behavior_server]: Running wait`
is stilled triggered although robot is rotating in place. However, this time, it does not take 10s like setting in my xml file `<Wait wait_duration="10.0"/>`. It only takes like 1 or 2 seconds to continue to rotate in place` [controller_server-12] [INFO 2024-11-23 08:15:42.307] [controller_server]: Passing new path to controller..`

Below is my launch command:
ros2 launch nav2_bringup tb4_simulation_launch.py use_gz:=False headless:=False use_rviz:=False use_composition:=False

padhupradheep, 2024-11-26:
Looking more at the issue, I see that BT is not inhibiting and causing the problem. @chanhhoang99 I see there seems to be a localization jump when the robot rotates near the goal location, which apparently causes the `PathLongerOnApproach` to be triggered, followed by the consequent behaviors. Could you please try adjusting the length factor? You have set it to 1.01, meaning, it's sensitive to localization jumps or another option would be to adjust the localization parameters if we are looking for cm level granularity. 

I'm not sure, if your use-case demands, but if that is the case, then we might need to add some `GoalVicinity` behavior to that could check if the robot is within a specific goal tolerance, and probably we don't need path longer on approach after that. 

@SteveMacenski Maybe we can pin this issue to our documentation website. This could be an interesting case for some.

padhupradheep, 2024-11-26:
Analysing the issue, I see that BT is not inhibiting and causing the problem. @chanhhoang99 I see there seems to be a localization jump when the robot rotates near the goal location, which apparently causes the
"PathLongerOnApproach to be triggered, followed by the consequent behaviors. Could you please try adjusting the length factor? You have set it to 1.01, meaning, it's sensitive to localization jumps or another option would be to adjust the localization parameters if we are looking for cm level granularity.

Maybe you might need to add some behavior to the BT, that could check if the robot is within a specific goal tolerance, after which the `PathLongerOnApproach` should not be triggered. I think we have a behavior called, `IsRobotInGoalVicinity` that exactly does that. Something like the one I commented here: https://github.com/ros-navigation/navigation2/issues/4003#issuecomment-1919539960

@SteveMacenski what do you think? Localization jumps are something that we didn't consider for atleast the example BT that we have. Maybe open sourcing the solution that I showed for #4003 atleast might help @chanhhoang99 get through this issue.

SteveMacenski, 2024-11-26:
Agreed!

chanhhoang99, 2024-11-27:
Hi @padhupradheep , Thanks for the finding.
 I will adjust the length factor and do the test.
I also want to have a try with `IsRobotInGoalVicinity`, is it available on nav2 official repository or it's from your fork open source work, could you tell ?

padhupradheep, 2024-11-27:
Not yet, I'd update here, when we open source it (let me try to get it ASAP).

>I also want to have a try with IsRobotInGoalVicinity

Note it is a condition plugin, therefore you would require to tweak around the BT to get your use-case working.

padhupradheep, 2024-11-27:
Ah I see there is already a PR, that adds a similar behavior: https://github.com/ros-navigation/navigation2/pull/4620

@chanhhoang99 Would it be fine if I close this issue now ?

In essence, to solve the problem that you have posted, you need to tweak your BT such that, while reaching the goal proximity, the robot doesn't replan and trigger the `PathLongerOnApproach`.

closed 2024-11-27 by padhupradheep

chanhhoang99, 2024-12-16:
Hi @padhupradheep , would IsPathValid help in this case ?
I have tried tweaking my BT with IsPathValid rather than replanning every 1s, the wierd behavior above is resolved.
Below is my xml file navigate_to_pose_w_replanning_goal_patience_and_recovery.xml

<root BTCPP_format="4" main_tree_to_execute="MainTree">
  <BehaviorTree ID="MainTree">
    <RecoveryNode number_of_retries="6" name="NavigateRecovery">
      <PipelineSequence name="NavigateWithReplanning">
        <ControllerSelector selected_controller="{selected_controller}" default_controller="FollowPath" topic_name="controller_selector"/>
        <PlannerSelector selected_planner="{selected_planner}" default_planner="GridBased" topic_name="planner_selector"/>
        <RateController hz="1.0">
          <Fallback>
            <ReactiveSequence>
              <Inverter>
                <GlobalUpdatedGoal/>
              </Inverter>
              <IsPathValid path="{path}"/>
            </ReactiveSequence>
            <ComputePathToPose goal="{goal}" path="{path}" planner_id="{selected_planner}" error_code_id="{compute_path_error_code}"/>
          </Fallback>
        </RateController>
        <ReactiveSequence name="MonitorAndFollowPath">
          <PathLongerOnApproach path="{path}" prox_len="20" length_factor="1.3">
            <RetryUntilSuccessful num_attempts="1">
              <SequenceWithMemory name="CancelingControlAndWait">
                <CancelControl name="ControlCancel"/>
                <Wait wait_duration="10.0"/>
              </SequenceWithMemory>
            </RetryUntilSuccessful>
          </PathLongerOnApproach>
          <RecoveryNode number_of_retries="1" name="FollowPath">
            <FollowPath path="{path}" controller_id="{selected_controller}" error_code_id="{follow_path_error_code}"/>
            <ClearEntireCostmap name="ClearLocalCostmap-Context" service_name="local_costmap/clear_entirely_local_costmap"/>
          </RecoveryNode>
        </ReactiveSequence>
      </PipelineSequence>
      <ReactiveFallback name="RecoveryFallback">
        <GoalUpdated/>
        <RoundRobin name="RecoveryActions">
          <Sequence name="ClearingActions">
            <ClearEntireCostmap name="ClearLocalCostmap-Subtree" service_name="local_costmap/clear_entirely_local_costmap"/>
            <ClearEntireCostmap name="ClearGlobalCostmap-Subtree" service_name="global_costmap/clear_entirely_global_costmap"/>
          </Sequence>
          <Spin spin_dist="1.57"/>
          <Wait wait_duration="5.0"/>
          <BackUp backup_dist="0.30" backup_speed="0.15"/>
        </RoundRobin>
      </ReactiveFallback>
    </RecoveryNode>
  </BehaviorTree>
</root>

chanhhoang99, 2024-12-16:
With below xml, robot will replan again after waiting completely 10 second
[image]
However, with the original xml, robot do not have to wait 10 seconds.