ros-navigation/navigation2 #4685: Collision between 2 or more robots.

lucasl84, 2024-09-23:
Hi everyone,

I'm opening this issue just to get informed if currently is possible to integrate collision avoidance of 2 or more robots. Using the current Nav2 stack sometimes for example when two robots cross their paths to their goal they will definitely "collide" in the simulator, although they finally manage each one to get to their goals, having the robots colliding is not a good thing.

I've also noticed a similar situation where if I set the same goal for two robots and one arrives first, then the moving robot will definitely push the one that already reached the goal first. I've watched the video presented for the [Collision Monitor](https://docs.nav2.org/configuration/packages/collision_monitor/configuring-collision-monitor-node.html) and by the end of it, the first question asked seems to be related to "dynamic" obstacles and the answer given (at least from what I could understand) is a bit ambiguous.

Initially the author answers that the collision monitor is designed for static objects, but then he explains that the plugin will eventually detect and handle this case in the next iteration/next arrival of a message? This is discussed around minute 50:40.

Is the collision monitor/detector the only current way to handle these situations (provided it does support this)? In case the collision monitor is not appropriate, are there any options available that integrate well with the Nav2 stack?

As usual thanks for everyone's time,
Lucas.

padhupradheep, 2024-09-23:
Hey, have you checked out [Open-RMF](https://www.open-rmf.org/) ? 

Nav2 does not offer any planning mechanism (centralized / decentralised) that communicates the information between the robots and plan accordingly to avoid collisions or deadlocks. 

There are couple of associated issues with regards to this topic:

https://github.com/ros-navigation/navigation2/issues/2600
https://github.com/ros-navigation/navigation2/issues/2758

@SteveMacenski: do you have any additional inputs? 

Closing this issue. This will be a duplicate of #2600, if this is considered as a feature request. However feel free to continue the thread.

closed 2024-09-23 by padhupradheep

SteveMacenski, 2024-09-23:
I have none to provide. Multirobot coordination is not a current or imminently planned feature of Nav2 as that coordination is often highly application specific & in these situations robust single robot navigation is still required.

lucasl84, 2024-09-24:
@padhupradheep and @SteveMacenski thanks for answers and suggestions!