ros-navigation/navigation2 #4693: nav2_costmap_2d node lacks lifecycle manager bond

AnsgarKasselmann, 2024-09-26:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Iron
- DDS implementation:
  - CycloneDDS

#### Steps to reproduce issue

1. Launch nav2_lifecycle_manager
2. Launch nav2_costmap_2d::nav2_costmap_2d node

#### Expected behavior

Standalone costmap node launches with correct configuration without errors

#### Actual behavior

Standalone costmap node launches but lifecycle manager can't create bond and hence aborts bringup

#### Additional information

This issue might be closely related to #1966 and the corresponding fix #1968 since the [Costmap2DROS::on_activate()](https://github.com/ros-navigation/navigation2/blob/5dd7ad106986914cef2f25e3ab39fb976c869456/nav2_costmap_2d/src/costmap_2d_ros.cpp#L270) function is missing the createBond() function aswell.

#### Log

[INFO] [launch]: All log files can be found below /root/.ros/log/2024-09-26-11-18-17-921799-P02053571-488
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [lifecycle_manager-1]: process started with pid [491]
[INFO] [nav2_costmap_2d-2]: process started with pid [493]
[lifecycle_manager-1] [INFO] [1727349498.034495999] [lifecycle_manager_navigation]: Creating
[lifecycle_manager-1] [INFO] [1727349498.041337226] [lifecycle_manager_navigation]: Creating and initializing lifecycle service clients
[lifecycle_manager-1] [INFO] [1727349498.042892800] [lifecycle_manager_navigation]: Starting managed nodes bringup...
[lifecycle_manager-1] [INFO] [1727349498.043032901] [lifecycle_manager_navigation]: Configuring local_costmap
[nav2_costmap_2d-2] [INFO] [1727349498.043382026] [local_costmap]:
[nav2_costmap_2d-2]     local_costmap lifecycle node launched.
[nav2_costmap_2d-2]     Waiting on external lifecycle transitions to activate
[nav2_costmap_2d-2]     See https://design.ros2.org/articles/node_lifecycle.html for more information.
[nav2_costmap_2d-2] [INFO] [1727349498.045854759] [local_costmap]: Creating Costmap
[nav2_costmap_2d-2] [INFO] [1727349498.058982017] [local_costmap]: Configuring
[nav2_costmap_2d-2] [INFO] [1727349498.061877110] [local_costmap]: Using plugin "rgbd_obstacle_layer"
[nav2_costmap_2d-2] [INFO] [1727349498.085944632] [local_costmap]: rgbd_obstacle_layer being initialized as SpatioTemporalVoxelLayer!
[nav2_costmap_2d-2] [INFO] [1727349498.085995437] [local_costmap]: rgbd_obstacle_layer's global frame is odom.
[nav2_costmap_2d-2] [INFO] [1727349498.086379775] [local_costmap]: rgbd_obstacle_layer loaded parameters from parameter server.
[nav2_costmap_2d-2] [INFO] [1727349498.088059443] [local_costmap]: rgbd_obstacle_layer created underlying voxel grid.
[nav2_costmap_2d-2] [INFO] [1727349498.088531153] [local_costmap]: Passthough filter activated.
[nav2_costmap_2d-2] [INFO] [1727349498.089879769] [local_costmap]: Passthough filter activated.
[nav2_costmap_2d-2] [INFO] [1727349498.090381813] [local_costmap]: rgbd_obstacle_layer initialization complete!
[nav2_costmap_2d-2] [INFO] [1727349498.090403497] [local_costmap]: Initialized plugin "rgbd_obstacle_layer"
[nav2_costmap_2d-2] [INFO] [1727349498.090408603] [local_costmap]: Using plugin "inflation_layer"
[nav2_costmap_2d-2] [INFO] [1727349498.092563301] [local_costmap]: Initialized plugin "inflation_layer"
[lifecycle_manager-1] [INFO] [1727349498.097255965] [lifecycle_manager_navigation]: Activating local_costmap
[nav2_costmap_2d-2] [INFO] [1727349498.097324619] [local_costmap]: Activating
[nav2_costmap_2d-2] [INFO] [1727349498.097334785] [local_costmap]: Checking transform
[nav2_costmap_2d-2] [INFO] [1727349498.097391740] [local_costmap]: start
[nav2_costmap_2d-2] [INFO] [1727349498.097399786] [local_costmap]: rgbd_obstacle_layer was activated.
[lifecycle_manager-1] [ERROR] [1727349500.364462253] [lifecycle_manager_navigation]: Server local_costmap was unable to be reached after 4.00s by bond. This server may be misconfigured.
[lifecycle_manager-1] [ERROR] [1727349500.364784493] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.

SteveMacenski, 2024-09-26:
According to [the docs](https://docs.nav2.org/configuration/packages/configuring-lifecycle.html#parameters), you can disable the bond heartbeat using a value of `0` for the `bond_timeout` to allow non-bond connected nodes to use the lifecycle manager.

SteveMacenski, 2024-10-02:
@AnsgarKasselmann Any thoughts or updates?

AnsgarKasselmann, 2024-10-03:
@SteveMacenski Thanks. Yes, suggested solutions works!

closed 2024-10-03 by SteveMacenski