ros-navigation/navigation2 #4696: `/assisted_teleop` action client persists after node has been destroyed

tiwaojo, 2024-09-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04 via Docker on Jetson Xavier NX
- ROS2 Version:
  - Humble Source via [Docker Image](https://hub.docker.com/layers/dustynv/ros/humble-ros-base-l4t-r35.3.1/images/sha256-0a43789d92801cebbc7510fe8fd7acdfc16f8c62e56790435a89a596f9b7c3ba?context=explore)
- Version or commit hash:
  - [b25742ed05b0852b8c116b797b15e64f286f9780](https://github.com/ros2-gbp/navigation2-release/tree/release/humble/nav2_simple_commander)
- DDS implementation:
  - Cyclone DDS

#### Steps to reproduce issue

- Setup your workspace
  ```bash
  mkdir -p ros2_ws/src
  cd ros2_ws/src
  ros2 pkg create --build-type ament_python --license Apache-2.0 --node-name my_node my_package
  ```
- Install dependencies
  <details>
    <summary>Add the following to `dep.repos`</summary>
    
    ```yml
    repositories:
    src/nav2_simple_commander:
        type: git
        url: https://github.com/SteveMacenski/navigation2-release.git
        version: release/humble/nav2_simple_commander
    src/nav2_common:
        type: git
        url: https://github.com/SteveMacenski/navigation2-release.git
        version: release/humble/nav2_common
    src/nav2_msgs:
        type: git
        url: https://github.com/SteveMacenski/navigation2-release.git
        version: release/humble/nav2_msgs
    ```
  </details>

    ```bash
    touch src/dep.repos
    vcs import --skip-existing --input src/dep.repos
    pip3 install fastapi uvicorn
    ```

- Duplicate the [action_client_interfaces](https://github.com/ros2/demos/tree/humble/action_tutorials/action_tutorials_interfaces) package from the [action_tutorials](https://github.com/ros2/demos/tree/humble/action_tutorials) repo into your workspace.

- Configure the `package.xml` file of your package to depend on the `action_client_interfaces` package by adding the following line:
  ```diff
      <?xml version="1.0"?>
      ...
  +   <depend>violation_reporter_interfaces</depend>
      ...
  ```

- Create the action client node by adding the following code to `/ros2_ws/src/my_package/my_package/my_node.py`
  
  <details>
    <summary>/ros2_ws/src/my_package/my_package/my_node.py</summary>

    ```python
    from nav2_simple_commander.robot_navigator import BasicNavigator
    import rclpy
    from rclpy.action import ActionClient

    from action_tutorials_interfaces.action import Fibonacci

    class MinimalActionClient(BasicNavigator):

        def __init__(self):
            super().__init__('fibonacci_action_client')
            self._action_client = ActionClient(self, Fibonacci, 'fibonacci')

        def send_goal(self, order):
            goal_msg = Fibonacci.Goal()
            goal_msg.order = order

            self._action_client.wait_for_server()

            return self._action_client.send_goal_async(goal_msg)

        def destroy_node(self):
            self._action_client.destroy()
            # self.assisted_teleop_client.destroy() # uncomment to fix. parent class(BasicNavigator) should be responsible for deleting this action client
            super().destroy_node()

    def main(args=None):
        rclpy.init(args=args)

        action_client = MinimalActionClient()

        future = action_client.send_goal(10)

        rclpy.spin_until_future_complete(action_client, future)
        
        action_client.destroy_node()

    if __name__ == '__main__':
        main()
      ```

  </details>

- Create your api by adding the following code to `/ros2_ws/src/my_package/api/main.py`
  <details>
    <summary>/ros2_ws/src/my_package/api/main.py</summary>
    
    ```python
    import uvicorn
    from fastapi import FastAPI, Request, Response
    from fastapi.middleware.cors import CORSMiddleware
    import rclpy
    from rclpy.executors import SingleThreadedExecutor
    from rclpy.task import Future
    from my_package.my_node import MinimalActionClient
    from contextlib import asynccontextmanager

    @asynccontextmanager
    async def lifespan(app: FastAPI):
        # Perform startup actions here
        print("Starting up...")
        print("Initializing ROS Context...")
        rclpy.init(args=None)

        yield

        # Perform cleanup actions here
        print("Tearing down...")
        print("Shutting down ROS Context...")
        rclpy.try_shutdown()

    app = FastAPI(
        lifespan=lifespan,
        responses={200: {"description": "Success"}, 404: {"description": "Not found"}},
    )

    origins = ["*"]

    app.add_middleware(
        CORSMiddleware,
        allow_origins=origins,
        allow_credentials=True,
        allow_methods=["*"],
        allow_headers=["*"],
    )

    @app.get(
        path="/",
        name="index",
        tags=["home"],
        summary="Home page",
    )
    async def index(request: Request, response: Response):
        return {"message": "Hello World"}

    @app.get(path="/example_endpoint")
    async def example_endpoint(request: Request, response: Response):
        executor = SingleThreadedExecutor()
        action_client_node = MinimalActionClient()
        try:
            if rclpy.ok():
                # future: Future = action_client_node.send_goal(order=10)

                action_client_node.get_logger().info("Goal sent. Waiting for result...")
                # rclpy.spin_until_future_complete(action_client_node,future=future)
                executor.add_node(action_client_node)
                # executor.spin_once_until_future_complete(future=future,timeout_sec=20)

        except Exception as e:
            action_client_node.get_logger().error(f"An error occurred: {e}")
        finally:
            action_client_node.get_logger().info("Destroying client node")
            action_client_node.destroy_node()
            executor.remove_node(action_client_node)
            executor.shutdown()

    def main(args=None):
        try:
            uvicorn.run(
                app=app,
                host="127.0.0.1",
                port=8000,
            )
        except KeyboardInterrupt:
            print("Shutting down server...")

    if __name__ == "__main__":
        main()
    ```
  </details>

- Build the workspace
  ```bash
  colcon build --symlink-install
  ```

- Run the program
  ```bash
  source install/setup.bash
  python3 src/my_package/api/main.py
  # or 
  uvicorn src.my_package.api.main:app
  ```
- Send a request to the server
    ```bash
    curl http://127.0.0.1:8000/example_endpoint
    ```
- List nodes
    ```bash
    ros2 node list
    ```

#### Expected behavior
All nodes, actions and callbacks should be destroyed when the request is completed.

#### Actual behavior
The action client node is not destroyed when the server is shut down. As the `/assisted_teleop` action client is not destroyed, the node is not destroyed. This can be seen by running `ros2 action list` after the request has been served. Any further calls to the same endpoint recreates the action client node and the action client, hereby creating duplicate nodes and actions.

SteveMacenski, 2024-09-28:
Ask on robotics stack exchange, I'm not sure what this has to do with Nav2 other than using a Nav2 action server as the example. It looks like you're doing some unusual things and I'm not sure what you're expecting the behavior to look like. I don't see why the action server would come down just because you remove a client to it.

tiwaojo, 2024-09-28:
Hello, my apologies if my issue seems out of scope of Nav2 or wasn't communicated properly. I was able to determine the source of the error and was preparing a pull request for it, as a PR requires an issue attached. It was a minor bug with the `assisted_teleop_client` action client where it wasn't destroyed with the rest of the node. 

https://github.com/ros-navigation/navigation2/blob/f79de33c3c192cd66377dc9538bdcbc3597b7f66/nav2_simple_commander/nav2_simple_commander/robot_navigator.py#L97-L107

SteveMacenski, 2024-09-30:
Ah I see, sorry that was unclear. Please open the PR!

tiwaojo, 2024-10-01:
I see what you mean. I apologize for the confusion. I have opened the [PR](https://github.com/ros-navigation/navigation2/pull/4698). I also noticed there were some other action clients not being destroyed with the node. I took the liberty to include those in the pull request.

SteveMacenski, 2024-10-01:
Great, thanks for the contribution and pointing this out!

closed 2024-10-01 by SteveMacenski