ros-navigation/navigation2 #4705: Obstacle with low lidar "signature" not detected

lucasl84, 2024-10-04:
Hi everyone,

I'm trying to tweak the Nav2 stack I'm currently using so obstacles with a low lidar signature are also detected. I'm attaching two images with 3 robots to better explain what's the issue. Robot 1 and 3 are facing oppsite sides each other and Robot 2 on the left. As you can see Robot 1 can see Robot 2 as an obstacle but for Robot 3 only the lidar trace is seen. The other image shows exactly the same from Robot 3. I've noticed that when robots 1 and 3 become sufficiently close they are detected each other as obstacles. The question is if there is any parameter that allows me to tweak the minimum number of lidar readings to detect an obstacle (or a similar settings that allows me to do so).

[image]

[image]

As usual thanks for everyone's time,
Lucas.

SteveMacenski, 2024-10-04:
> As you can see Robot 1 can see Robot 2 as an obstacle but for Robot 3 only the lidar trace is seen

It looks like your local costmap needs to be increased in size so that you can get Robot 3 into your costmap, assuming its in your requirements that you detect and react to other robots that distance away with your trajectory planner. However, if your trajectory planner doesn't actually predict that far away, then its moot since it'll be tracked by the costmap, but not actually utilized by any algorithm to impact behavior until closer. The size of the costmap and the trajectory window should be both adjusted so you can make use of that expanded data access (else, moot).

 This isn't a matter of number of readings, its a matter of local costmap range. If you were visualizing the global costmap **with lidar data being fused in**, that should be fine for the purposes of global planning. It looks to me that your global costmap is only considering the static map + inflation layer. If you add in the same configs you have for the 2D lidar within the global costmap (minding the `range`-based parameters) then you should be fine.

lucasl84, 2024-10-07:
Hi Steve,

Thanks for your reply. The screenshots that I've sent you already have and increased costmap size (10x10, initially were 5x5) and also already using the `_range` settings for each one of the obstacle layers in both costmaps. Note also that I've used the "measure" tool from RViz to see the distance between them and it reports 7,35m, so it seems to be in range. Here are the details of the configurations I'm currently using, perhaps I'm missing something else? When you mention "trajectory window", is that a setting that can be tweaked?

Global costmap & Planner server config:
```
planner_server:
  ros__parameters:
    expected_planner_frequency: 1.0
    planner_plugins: ["GridBased"]
    GridBased:
      plugin: "nav2_navfn_planner::NavfnPlanner"
      tolerance: 0.5
      use_astar: false
      allow_unknown: true

global_costmap/global_costmap:
  ros__parameters:
    update_frequency: 1.0
    publish_frequency: 1.0

    height: 10
    width: 10
    global_frame: map
    robot_base_frame: base_link
    map_topic: /map_server

    robot_radius: 0.22
    track_unknown_space: true
    plugins: ["static_layer", "obstacle_layer", "inflation_layer"]
    filters: ["keepout_filter", "inflation_layer"]
    obstacle_layer:
      plugin: "nav2_costmap_2d::ObstacleLayer"
      enabled: True
      observation_sources: scan
      scan:
        topic: /scan
        max_obstacle_height: 2.0
        clearing: True
        marking: True
        data_type: "LaserScan"
        raytrace_max_range: 29.0
        raytrace_min_range: 0.0
        obstacle_max_range: 29.0
        obstacle_min_range: 0.0
    static_layer:
      plugin: "nav2_costmap_2d::StaticLayer"
      map_subscribe_transient_local: True
    inflation_layer:
      plugin: "nav2_costmap_2d::InflationLayer"
      cost_scaling_factor: 3.0
      inflation_radius: 0.55
    keepout_filter:
      plugin: "nav2_costmap_2d::KeepoutFilter"
      enabled: True
      filter_info_topic: "/costmap_filter_info"
    always_send_full_costmap: True
```

Local costmap & controller server settings:
```
controller_server:
  ros__parameters:
    controller_frequency: 1.0
    min_x_velocity_threshold: 0.001
    min_y_velocity_threshold: 0.5
    min_theta_velocity_threshold: 0.001
    failure_tolerance: 0.3
    progress_checker_plugin: progress_checker

    # Original values as from the nav2_params.yaml example
    goal_checker_plugins: ["general_goal_checker"] # "precise_goal_checker"
    controller_plugins: ["FollowPath"]

    # Progress checker parameters
    progress_checker:
      plugin: "nav2_controller::SimpleProgressChecker"
      required_movement_radius: 0.5
      movement_time_allowance: 10.0
    general_goal_checker:
      stateful: True
      plugin: "nav2_controller::SimpleGoalChecker"
      xy_goal_tolerance: 0.25
      yaw_goal_tolerance: 0.25
    FollowPath:
      plugin: "nav2_regulated_pure_pursuit_controller::RegulatedPurePursuitController"
      desired_linear_vel: 0.5
      lookahead_dist: 0.6
      min_lookahead_dist: 0.3
      max_lookahead_dist: 0.9
      lookahead_time: 1.5
      rotate_to_heading_angular_vel: 1.8
      transform_tolerance: 0.1
      use_velocity_scaled_lookahead_dist: false
      min_approach_linear_velocity: 0.05
      approach_velocity_scaling_dist: 0.6
      use_collision_detection: true
      max_allowed_time_to_collision_up_to_carrot: 1.0
      use_regulated_linear_velocity_scaling: true
      use_fixed_curvature_lookahead: false
      curvature_lookahead_dist: 0.25
      use_cost_regulated_linear_velocity_scaling: false
      regulated_linear_scaling_min_radius: 0.9
      regulated_linear_scaling_min_speed: 0.25
      use_rotate_to_heading: true
      allow_reversing: false
      rotate_to_heading_min_angle: 0.785
      max_angular_accel: 3.2
      max_robot_pose_search_dist: 10.0

local_costmap/local_costmap:
  ros__parameters:
    height: 10
    width: 10
    update_frequency: 1.0
    publish_frequency: 1.0
    global_frame: map
    robot_base_frame: base_link
    map_topic: /map_server
    rolling_window: True
    resolution: 0.04
    robot_radius: 0.22
    plugins: ["voxel_layer", "obstacle_layer", "inflation_layer"]
    filters: ["keepout_filter", "inflation_layer"]
    inflation_layer:
      plugin: "nav2_costmap_2d::InflationLayer"
      cost_scaling_factor: 3.0
      inflation_radius: 0.55
    obstacle_layer:
      enabled: true
      plugin: nav2_costmap_2d::ObstacleLayer
      combination_method: 0
      footprint_clearing_enabled: true
      max_obstacle_height: 2.0
      observation_sources: scan
      scan:
        topic: /scan
        max_obstacle_height: 2.0
        clearing: True
        marking: True
        data_type: "LaserScan"
        raytrace_max_range: 29.0
        raytrace_min_range: 0.0
        obstacle_max_range: 29.0
        obstacle_min_range: 0.0
    voxel_layer:
      enabled: True
      plugin: "nav2_costmap_2d::VoxelLayer"
      combination_method: 0
      publish_voxel_map: True
      origin_z: 0.0
      z_resolution: 0.05
      z_voxels: 16
      max_obstacle_height: 2.0
      mark_threshold: 0
      observation_sources: scan
      scan:
        topic: /scan
        max_obstacle_height: 2.0
        clearing: True
        marking: True
        data_type: "LaserScan"
        raytrace_max_range: 29.0
        raytrace_min_range: 0.0
        obstacle_max_range: 29.0
        obstacle_min_range: 0.0
    static_layer:
      map_subscribe_transient_local: True
    keepout_filter:
      plugin: "nav2_costmap_2d::KeepoutFilter"
      enabled: True
      filter_info_topic: "/costmap_filter_info"
    always_send_full_costmap: True
```

Thanks a lot,
Lucas.

closed 2024-10-07 by SteveMacenski