ros-navigation/navigation2 #4709: Bug report

Forgotten0116, 2024-10-06:
This is my first time asking a question on GitHub. Please forgive any unclear descriptions.
-------------------------------------------------------------------------------------------------------------
Operating System: Ubuntu 22.04
ROS2 Version: Humble
Nav2: installation by git clone https://github.com/ros-planning/navigation2.git
-------------------------------------------------------------------------------------------------------------
Error message after colcon build:

--- stderr: nav2_util
CMake Error at src/CMakeLists.txt:1 (add_library):
  Target "nav2_util_core" links to target "bondcpp::bondcpp" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at src/CMakeLists.txt:35 (add_executable):
  Target "lifecycle_bringup" links to target "bondcpp::bondcpp" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at src/CMakeLists.txt:40 (add_executable):
  Target "base_footprint_publisher" links to target "bondcpp::bondcpp" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_execution_timer" links to target "bondcpp::bondcpp" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  test/CMakeLists.txt:1 (ament_add_gtest)

CMake Error at /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_node_utils" links to target "bondcpp::bondcpp" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  test/CMakeLists.txt:4 (ament_add_gtest)

CMake Error at /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_service_client" links to target "bondcpp::bondcpp" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  test/CMakeLists.txt:10 (ament_add_gtest)

CMake Error at /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_string_utils" links to target "bondcpp::bondcpp" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  test/CMakeLists.txt:13 (ament_add_gtest)

CMake Error at /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_lifecycle_utils" links to target "bondcpp::bondcpp" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  test/CMakeLists.txt:17 (ament_add_gtest)

CMake Error at /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
[... 83 lines cut ...]
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  test/CMakeLists.txt:49 (ament_add_gtest)

CMake Error at /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_validation_messages" links to target "bondcpp::bondcpp" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  test/CMakeLists.txt:52 (ament_add_gtest)

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< nav2_util [1.38s, exited with code 1]

Summary: 8 packages finished [54.7s]
  1 package failed: nav2_util
  2 packages had stderr output: nav2_loopback_sim nav2_util
  34 packages not processed
-------------------------------------------------------------------------------------------------------------
I have I have already installed bondcpp but still error

SteveMacenski, 2024-10-06:
You need to check out the appropriate branch for your distribution, in your case `humble`. The `main` branch targets `rolling`.

closed 2024-10-06 by SteveMacenski

Xlqmu, 2024-11-13:
> 您需要检查适合您的发行版的分支（就您而言）`humble`。`main`分支目标为`rolling`。

My ROS2 version is also Humble.I have same trouble.Could you believe in more detail？

Forgotten0116, 2024-11-13:
you can follow the image below, select the corresponding ROS version and then clone

青青子衿
***@***.***

Xlqmu, 2024-11-14:
> you can follow the image below, select the corresponding ROS version and then clone 青青子衿 ***@***.***  
> […](#)

oh,thank you,I solved the problem.🙇‍