ros-navigation/navigation2 #4718: BUG:tow close right-anlge trigger stop

cliffe10086, 2024-10-10:
## Bug report

**Required Info:**

- Operating System:Ubuntu 22.04
  - 
- ROS2 Version:humble
  - 
- Version or commit hash:f79de33c3c192cd66377dc9538bdcbc3597b7f66
  - 
- DDS implementation:default
  - 

#### Steps to reproduce issue

```use cmd from nav2 website 
export TURTLEBOT3_MODEL=waffle
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/humble/share/turtlebot3_gazebo/models
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False

then in gazebo ignition, put tow walls to filll between tow polls like (.-.-.)
robort is below the wall in the middle , goal is above the wall in the middle.
     G(oal)
.---------.
     R(obort)
```

#### Expected behavior
turn tow right-angle corners and arrive
#### Actual behavior
stop at origin spot
#### Additional information
[录屏 2024年10月10日 10时34分59秒.webm](https://github.com/user-attachments/assets/3ef01ba6-034d-494d-bcac-d2c8128faee3)
none，my english is poor ,sorry if make troubles

SteveMacenski, 2024-10-10:
I don’t understand what you mean. Can you describe this in more detail with some image if necessary?

cliffe10086, 2024-10-11:
i have a video submitted above.here are two more image.in the first ,when global path has two right-angle corner,robort will stop.in second is a provided gazebo map with two small change(added two walls).
[image]
[image]

SteveMacenski, 2024-10-15:
Have you made an attempt to tune the controller? Which are you using and what are its parameters?

SteveMacenski, 2024-12-10:
Closing due to lack of response

closed 2024-12-10 by SteveMacenski