ros-navigation/navigation2 #4722: Online example is not working on ros2 jazzy

davidemannone, 2024-10-15:
## Bug report

Cannot bring to work the "Getting Started" example on Ros2 Jazzy with Ubuntu 24.04, in both binary and source version (even if compilation completes successfully). Logs reports tf frames missing (odom), maps errors and rviz crashs after few seconds, and command terminates execution.
As stated I tried first not to install "sudo apt install ros-jazzy-turtlebot3-gazebo", after trying it apt cannot find it and last, even install all of tb3 and also tb4, will not solve the question. Cannot find this path: export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/jazzy/share/turtlebot3_gazebo/models

What should I do to make your example working? 

- Operating System:
  - Ubuntu 24.04 -->
- ROS2 Version:
  Jazzy, binary
- Version or commit hash:
   Nav2 binaries got as written on your site (using jazzy), same for source 
- DDS implementation:
  Fast-RTPS

#### Steps to reproduce issue
Followed your simple and clear guide and when launched it didn't work

#### Expected behavior
The same as web page

#### Actual behavior
Starts, rviz closes after few seconds followed by console execution stop. Gazebo remains running

#### Additional information

Logs reports tf frames missing and rqt and tftree doesn't shows all expected connections 

[component_container_isolated_88025_1729002265308.log](https://github.com/user-attachments/files/17380067/component_container_isolated_88025_1729002265308.log)

SteveMacenski, 2024-10-15:
So, it looks like our Getting Started (and build/install) guide needs to be updated from the migration to New-Gazebo from Gazebo-Classic. Jazzy / Rolling don't have support for Gazebo Classic anymore since it doesn't ship with those distributions on 24.04. 

We are no longer dependent on the TB3 packages and instead use this repo instead for the URDFs and sim files https://github.com/ros-navigation/nav2_minimal_turtlebot_simulation. That step is not required to `apt install` anything other than packages in the minimal Tb3/4 repo -- which have binaries in Jazzy that you can use. The setting of the Gazebo model paths are also not required anymore (we automated that in the launch files).

If you're open to contributing, I'd love it if you could propose the changes to that instruction set that aligns with the current state of the project using the modern updated gazebo!

davidemannone, 2024-10-16:
I'm a novice in Ros2 and Nav2, so I don't have enough confidence and experience yet to make this. But i did some tries and these are the results.

I've installed Ros2 Jazzy following the guide and it comes already with with Gazebo installed too. I had a look to the version and it was Harmonic. I don't know if it is old (classic) or new one and having a search for something new it seems that a new version of Gazebo should be Ignition but I cannot find it (all brings to Gazebo: https://ignitionrobotics.org/). I've found finally this site: https://github.com/gazebosim/ros_gz/tree/jazzy and followed the instructions for install it. After that, Gazebo is still running (using the gz sim commands, as before: just the nav2 example did not run) , but not with ing gazebo. 
Having a further look I found this https://gazebosim.org/docs/latest/getstarted/ and here https://gazebosim.org/docs/ionic/install_ubuntu/ I found I should use Ionic. Following also this I got all installed correctly and gz is still running but it reports always harmonic. If I do a check with apt for ros-jazzy-gz-ionic and ros-jazzy-gz-harmonic both says they are updated. Maybe there both installed now and gz sim refers just to one? Having a look in opt/ros/jazzy..  /opt, /lib and /share I can find only "gz" (in files or links) and no versions (folders or in names).

I've got the nav2_tb repo you suggested and it compiles. I don't know where to place it correctly: if part of all the other nav2's (inside the suggested nav2_ws/src/navigation2) or elsewhere like the nav2_ws/src. In both cases it compiles well and not doing anything else (not even the exports as you suggested) just sourcing nav2_ws (for the nav2_ws/src/navigation2 or both separate for nav2_ws/src) it still is not running and rviz dies as before.  

I trust in your spare airport hours. :)

SteveMacenski, 2024-10-16:
Jazzy only supports / ships with the new Gazebo. Ignition is an older naming schema which is deprecated due to trademark issues. its all just called 'Gazebo' now - not confusing at all, right? :wink: So you're definitely using new-Gazebo (or as I jokingly call it sometimes: Gazebo2).

You shouldn't need to download + build the code unless you're on Rolling. You should be able to `apt install` those packages instead on Jazzy. Please verify that this works for you. At that point, you should just be able to run all the Nav2 demos in `nav2_bringup` without issue

davidemannone, 2024-10-17:
It sounds good, but it still wont work. 
sudo apt install ros-jazzy-navigations2   --> all up to date
sudo apt install ros-jazzy-nav2-minimal-tb*   --> installed ..tb3-sim, ..tb4-sim, tb4-description
source /opt/ros/jazzy/setup.bash  --> ok
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False    --> as ever: tf errors, rviz starts and closes, gazebo ok with model
ros2 launch nav2_bringup tb4_simulation_launch.py headless:=False    --> as ever: tf errors, rviz starts and closes, gazebo with model (warehouse)
Attached the entire log folder, clean, with just the startup.  
[logs.zip](https://github.com/user-attachments/files/17416774/logs.zip)

SteveMacenski, 2024-10-17:
> available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist

That's usually related to gazebo as the publisher of the odometry data and frames - if you wait for the gazebo window to appear, do you still see these errors afterwards? It can take a moment for gazebo to fully load, especially on the first run while it downloads online mesh assets used in the SDF file

patham9, 2024-10-22:
@davidemannone Same issue here after a week of trying. If you need it to work before online tutorials will catch up with the ongoing code changes in a year or so, I can highly recommend ROS1 in the meanwhile, as it is not a moving target anymore and doesn't segfault randomly. The story with ROS2, despite being sold as something great, is in reality still a sad one full of bugs and instabilities. Just don't waste your time and go for what is proven to work.

davidemannone, 2024-10-23:
Oh yes, I did but it still stops alone and first closes RViz and then stops the console; Gazebo remains open and it should be closed by X.
I've attached this times logs without breaks and i put also some screenshots of the error report that appears first time after RViz crash.
[nav2_log_screenshots.zip](https://github.com/user-attachments/files/17495836/nav2_log_screenshots.zip)

patham9, 2024-10-24:
@davidemannone: Unfortunately ROS2 is realistically speaking Alpha-quality software (that's what is meant with "production-ready" in the OSS community), so expect some manual fiddling for a few days to avoid segfaults, but the good news is that usually there is a way to get stuff to run with enough persistence.

For instance you might need to export `QT_QPA_PLATFORM=xcb` to hamper Gazebo from crashing in case you are running in a VM, since Gazebo (other than simulators like Webots) is not smart enough to select a working graphics mode by itself or to give meaningful error messages for crashes.

Also Ubuntu Mesa drivers can cause issues in case you are using Ubuntu 24 already. This fixed the Mesa issue for me:

```
sudo add-apt-repository ppa:kisak/kisak-mesa
sudo apt update
sudo apt upgrade

```

Additionally you may need to pass some additional parameters. While this contradicts the tutorials, that's what at least seems to be needed for Jazzy to have the relevant nodes spawned up:

```
export QT_QPA_PLATFORM=xcb
ros2 launch nav2_bringup tb3_simulation_launch.py slam:=True nav:=True headless:=False use_sim_time:=True
```

Please let me know how that goes for you, if it doesn't work I am willing to share a VM file with you.

davidemannone, 2024-10-30:
So nice, that's it! Or almost it is... 
I had some doubt at beginning because I was not using a VM but a phy Ubuntu 24.04 and Gazebo has never crashed, but RViz does it every time.
But i follow you instructions and after:
- installing
- sourcing the binary setup
- exporting
I have seen for the first time all starting up and standby without issues. That's so great!

Unfortunately, it worked just for the first start!
I did some other tries but nothing. First time restarting seemed to work again for the first time and finally it seems it works just the first time in the same console. What happens at the second launch: as ever RViz crashes.

patham9, 2024-10-30:
I am glad it worked for you!

Since you mention after restart it works once, I see three different plausible scenarios, you can try separately from top to bottom:

- Processes still running in the background, make sure Gazebo and Rviz processes are fully gone before restarting, as well as potentially stuck ROS2 nodes. To test that easily you could logout and login again instead of full a restart. (though this doesn't exclude the next point, you will need to look at the concrete processes to distinguish)

- It could be an issue with your graphics driver (open source drivers often have strange bugs): `export LIBGL_ALWAYS_SOFTWARE=1` (in addition to export `QT_QPA_PLATFORM=xcb`) will switch to more stable but slow CPU-based rendering. If that works you will know whether that's the case, and if so you will need to find a better graphics driver to not rely on CPU rendering.

- There might be some temporary folders which get reset after the restart, you may want to try clearing the content of /tmp/ at first instead of restarting. There is also the user dir ~/.gz/ where cleaning part of it could help between runs but I am not very familiar with the folder structure.

patham9, 2024-11-05:
@davidemannone: Please let us know what you found to resolve it, as others might run into the issue too.

davidemannone, 2024-11-07:
I did all of your suggests and these are the final results:

1. Processes

Here the process list diffs:

before -> ros
a          4371  0.1  0.0 489312 25836 ?        Sl   07:05   0:00 /usr/bin/upda
ta          4450  0.0  0.0 466428  9216 ?        Sl   07:06   0:00 /usr/libexec/
ta          4463  8.3  0.2 1188744 87804 pts/0   Sl+  07:06   0:01 /usr/bin/pyth
ta          4485 16.0  0.1 1164148 56000 pts/0   Sl+  07:06   0:03 /opt/ros/jazz
ta          4487 42.9  0.9 3759320 296028 pts/0  Rl+  07:06   0:08 gz sim -r -s 
ta          4488  0.0  0.0   2804  1664 pts/0    S+   07:06   0:00 /bin/sh -c ru
ta          4489  9.0  0.1 824280 36136 pts/0    Sl+  07:06   0:01 /opt/ros/jazz
ta          4491 27.6  0.6 2984156 218980 pts/0  Sl+  07:06   0:05 /opt/ros/jazz
ta          4492 93.7  1.2 5045736 406388 pts/0  Sl+  07:06   0:19 gz sim -v4 -g
ta          4493 65.9  0.4 4004660 147968 pts/0  Sl+  07:06   0:13 /opt/ros/jazz
ta          4494  8.0  0.1 879620 43752 pts/0    Sl+  07:06   0:01 /opt/ros/jazz
ta          4495  7.9  0.1 975732 38268 pts/0    Sl+  07:06   0:01 /opt/ros/jazz
ta          4496  8.8  0.2 1168876 67884 pts/0   Sl+  07:06   0:01 /opt/ros/jazz
root        4972  0.0  0.0  30732  6272 ?        S    07:07   0:00 (udev-worker)

ros -> after
root        5056  0.0  0.0  30732  6272 ?        S    07:09   0:00 (udev-worker)

before -> after
ta          4371  0.0  0.0 489312 25836 ?        Sl   07:05   0:00 /usr/bin/upda
root        5056  0.0  0.0  30732  6272 ?        S    07:09   0:00 (udev-worker) 

and just the most significant ros log after CTRL+C: 

[ERROR] [gazebo-5]: process has died [pid 4488, exit code -2, cmd 'ruby $(which gz) sim -v4 -g  --force-version 8'].
[INFO] [launch]: process[gazebo-5] was required: shutting down launched system
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Cannot shutdown a ROS adapter that is not running
[INFO] [gz-4]: process has finished cleanly [pid 4487]
[ERROR] [component_container_isolated-8]: process[component_container_isolated-8] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [component_container_isolated-8]: sending signal 'SIGTERM' to process[component_container_isolated-8]
[component_container_isolated-8] [INFO] [1730959693.705000539] [rclcpp]: signal_handler(signum=15)
[ERROR] [component_container_isolated-8]: process[component_container_isolated-8] failed to terminate '10.0' seconds after receiving 'SIGTERM', escalating to 'SIGKILL'
[INFO] [component_container_isolated-8]: sending signal 'SIGKILL' to process[component_container_isolated-8]
[ERROR] [component_container_isolated-8]: process has died [pid 4493, exit code -9, cmd '/opt/ros/jazzy/lib/rclcpp_components/component_container_isolated --ros-args --log-level info --ros-args -r __node:=nav2_container --params-file /tmp/launch_params_p7l0y_8z --params-file /tmp/launch_params_tw54vdd1 -r /tf:=tf -r /tf_static:=tf_static'].

With logout/in it works too, the first time.

2. Graphics driver

By adding this other suggested export it works finally repeatedly without apparently new issues! As you said it seems to start/works slower but it works. 

So i add that I'm using just a Dell Precision 5550.

3. Temporary folders

By removing first only tmp files and then both tmp and all ,gz files it is still not working.

I attach the full ps aux of the beore->ros->after test where I got the above summary.
[ps aux all.LOG](https://github.com/user-attachments/files/17657502/ps.aux.all.LOG)
I do not attach the tmp and .gz folder after ros start because it seems me they are not involved.

@patham9 I'm verdy sorry for my late reply

patham9, 2024-11-07:
@davidemannone 
Thank you for your detailed output. It seems very similar to what I experienced. It probably it means for now the sensor simulation needs to happen on the CPU. However chances are very good that you can have GPU acceleration with Gazebo if it is fine for you to run sdf files with lower graphics requirements. Here is a command you can try which might allow you to run GPU-accelerated Gazebo for visualization, with CPU-based sensor simulation, which should work repeatedly:

```
export MY_WORLD=/opt/ros/jazzy/share/nav2_minimal_tb4_sim/worlds/warehouse.sdf
export QT_QPA_PLATFORM=xcb
export LIBGL_ALWAYS_SOFTWARE=1
gnome-terminal -- bash -c "sleep 10 && export LIBGL_ALWAYS_SOFTWARE=0 && export QT_QPA_PLATFORM=xcb && gz sim --render-engine ogre \$MY_WORLD; exec bash" &
ros2 launch nav2_bringup tb4_simulation_launch.py slam:=True nav:=True headless:=True autostart:=True use_sim_time:=True rviz_config_file:=nartech_view.rviz world:=$MY_WORLD
```
As you see here the trick is to start rviz with "always software GL" while Gazebo starts with GPU acceleration with Ogre 1 render engine which needs less GL features and hence will likely work for you.

closed 2024-11-18 by SteveMacenski

stevedanomodolor, 2024-12-01:
> * Kiwibot

Will start working on the GPS tutorial

SteveMacenski, 2024-12-02:
Which do you mean? If its just for the tutorial and not general purpose for Nav2 bringup, it should probably be in the https://github.com/ros-navigation/navigation2_tutorials repository. If its code that is for general purpose bringup / testing, then it could go into the minimal sim bringup repo

stevedanomodolor, 2024-12-02:
> Which files do you mean? If its just for the tutorial and not general purpose for Nav2 bringup, it should probably be in the https://github.com/ros-navigation/navigation2_tutorials repository. If its code that is for general purpose bringup / testing, then it could go into the minimal sim bringup repo 

I am referring to the sonora world file

SteveMacenski, 2024-12-02:
I think that probably belongs in the tutorials package unless we use it somewhere in Nav2 (i.e. like we use it in the system test)