ros-navigation/navigation2 #4733: RegulatedPurePursuitController Not Loading in Nav2 Composition, Defaulting to DWBController Instead

pky404, 2024-10-29:
Description: I'm experiencing an issue where the RegulatedPurePursuitController specified in the YAML file for the Nav2 container is not being loaded. Instead, the system defaults to using the DWBController. However, the rest of the parameters are loaded correctly from the YAML file, indicating that the file itself is correctly referenced.

Environment:
ROS2 Version: Humble
Nav2 Installation: Binary
Controller: RegulatedPurePursuitController (RPP)
Expected Behavior: The RegulatedPurePursuitController specified in the YAML file should load and initialize as the active controller in the controller_server.

Observed Behavior: The DWBController loads instead of the RegulatedPurePursuitController, despite RPP being specified in the YAML file.

Steps to Reproduce:

Configure a YAML file to use RegulatedPurePursuitController.
Pass the YAML configuration to nav2_container in the Nav2 launch file.
Launch the Nav2 stack and observe which controller initializes.
Additional Information:

RPP and DWBController are both installed in the workspace.
YAML file path is correctly referenced in the launch file, and other parameters load without issue.
Any insight into why this is happening, or if there are additional debugging steps to verify the controller configuration in the Nav2 composition setup, would be appreciated.

```
[INFO] [1730180033.002086469] [controller_server]: Created progress_checker : progress_checker of type nav2_controller::SimpleProgressChecker
[INFO] [1730180033.006463237] [controller_server]: Created goal checker : goal_checker of type nav2_controller::SimpleGoalChecker
[INFO] [1730180033.009155597] [controller_server]: Controller Server has goal_checker  goal checkers available.
[INFO] [1730180033.020483667] [controller_server]: Created controller : FollowPath of type dwb_core::DWBLocalPlanner
[INFO] [1730180033.032076509] [controller_server]: Setting transform_tolerance to 0.100000
[INFO] [1730180033.083994872] [local_costmap.local_costmap]: StaticLayer: Resizing costmap to 333 X 483 at 0.100000 m/pix
[ERROR] [1730180033.088957888] [controller_server]: Couldn't load critics! Caught exception: No critics defined for FollowPath
[ERROR] [1730180033.110182776] [controller_server]: Caught exception in callback for transition 10
[ERROR] [1730180033.110314234] [controller_server]: Original error: No critics defined for FollowPath
[WARN] [1730180033.110378587] [controller_server]: Error occurred while doing error handling.
[FATAL] [1730180033.110427804] [controller_server]: Lifecycle node controller_server does not have error state implemented
[ERROR] [1730180033.111244520] [lifecycle_manager_navigation]: Failed to change state for node: /amr0/controller_server
[ERROR] [1730180033.111332105] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
```

```

      FollowPath:
            plugin: "nav2_regulated_pure_pursuit_controller::RegulatedPurePursuitController"
            desired_linear_vel: 0.5 #0.5
            lookahead_dist: 0.8 #1.5
            min_lookahead_dist: 0.3
            max_lookahead_dist: 3.0 #3.0
            lookahead_time: 1.5 #1.5
            rotate_to_heading_angular_vel: 0.5 #1.8
            transform_tolerance: 0.1
            use_velocity_scaled_lookahead_dist: false
            min_approach_linear_velocity: 0.05
            approach_velocity_scaling_dist: 0.6
            use_fixed_curvature_lookahead: false
            curvature_lookahead_dist: 0.25
            use_collision_detection: true
            max_allowed_time_to_collision_up_to_carrot: 1.0
            use_regulated_linear_velocity_scaling: true
            use_cost_regulated_linear_velocity_scaling: false
            regulated_linear_scaling_min_radius: 0.9
            regulated_linear_scaling_min_speed: 0.25
            use_rotate_to_heading: false
            allow_reversing: true 
            rotate_to_heading_min_angle: 0.785
            max_angular_accel: 0.5
            max_robot_pose_search_dist: 10.0
            use_interpolation: true
```

SteveMacenski, 2024-10-29:
Please provide your full configuration file. Are you sure your file is being loaded at all?

pky404, 2024-10-30:
Rest of the parameters like costmap parameters are loaded correctly in the same launch sequence except the controller server parameters. 

[nav2_launch.txt](https://github.com/user-attachments/files/17566672/nav2_launch.txt)
[nav2_params.txt](https://github.com/user-attachments/files/17566673/nav2_params.txt)

closed 2024-10-30 by pky404